MarlinFirmware/Marlin #1284: When bed PID is enabled, compile fails.

andrewsil1, 2015-01-01:
#1249 introduced a constant called MAX_BED_PID which doesn't appear to be included anywhere, causing compilation to fail.

JoshyuNL, 2015-01-01:
I can confirm that this fails to compile.

*temperature.cpp: In function 'void manage_heater()':
temperature.cpp:585: error: 'MAX_BED_PID' was not declared in this scope*

MagoKimbra, 2015-01-01:
In temperature.cpp line 585 replace
if (pid_output > MAX_BED_PID)
width
if (pid_output > PID_MAX)

andrewsil1, 2015-01-01:
If that's really the fix then the author should submit a pull request.  I'm not looking for a short term fix, the compile failure needs to be fixed in the code base.

Andy
Tiny keyboard, tiny messages.

alhirzel, 2015-01-02:
resolved by #1286 ?

andrewsil1, 2015-01-02:
Verified that this compilation error is now resolved.

boelle, 2015-01-02:
closing as it is solved

closed 2015-01-02 by boelle