MarlinFirmware/Marlin #1299: Support ino or another build+upload method

alhirzel, 2015-01-03:
I would like to use [ino](http://inotool.org) or another console tool to build Marlin, perhaps this can be achieved with ino by creating another folder under the ArduinoAddons folder and using the `--arduino-dist` option. Mostly I'd like to be separate of the Arduino IDE.

Has anyone gotten this working with ino or another tool?

galexander1, 2015-01-03:
I've never used the Arduino IDE, and just use make with the Makefile that
already ships with Marlin...

alhirzel, 2015-01-03:
Ah, I figured that was autogenerated/not to be used. Is the Makefile the workflow most people use?

nophead, 2015-01-03:
I use ino.

alhirzel, 2015-01-03:
@nophead would you mind sharing info on how you set it up?

monkeydave, 2015-01-03:
To use ino just create a symlink to src folder and remove Marlin.pde like so

ln -s Marlin src
rm Marlin/Marlin.pde

Then build with 
ino build -m (YOUR BOARD TYPE)

alhirzel, 2015-01-03:
Did this require you to install the Arduino platform add-ons system-wide? I have them installed in ~/sketchbook but ino doesn't seem to pick them up

joco-nz, 2015-01-03:
I  had to install a number of dependent libraries.  That possibly could have been my ignorance/learning process in getting it working. You can also use an ino config file to tell it board type and USB port for the ino build and upload commands. Saving some typing. 

Sent from mobile device.

gregrebholz, 2015-01-03:
On OSX the Arduino IDE comes with avrdude and avr-gcc in the tools/ directory.  I use the Makefile from the command line and update AVR_TOOLS_PATH to point to /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/

boelle, 2015-01-15:
i use ino too... i simply just open it in Arduino IDE....

but since there has  been no traffic should we not close this one?

thinkyhead, 2015-01-19:
I'm using UECIDE from now on – and it apparently uses the .ino file.

closed 2015-01-19 by thinkyhead