MarlinFirmware/Marlin #1300: In autolevel mode X and Y axis steppers cannot be disabled when not used

fmalpartida, 2015-01-03:
If autoleveling is activated X and Y steppers cannot be disabled. If X and Y servos are configured to be disabled when not used, autoleveling will not work.

What happens is that when disabling the X and Y motors, the position of the X and Y axis are marked as unknown. Therefore, if you do a G28 to home both axis are set to a know stated. On the other hand, in the idle loop they are marked as unknown as soon as they are deactivated if not used. If you then do a G29, it will not work as the position is marked as unknown. 

The only way to get this working is to mark both X and Y axis to be enabled always.

It looks as if it is determining that the machine is a a particular location if we homed the machine and the steppers are enabled.

boelle, 2015-01-03:
as said on the testers thread once they are disabled the machine has no way of knowing what the pos is...

even if you dont move it how should the machine know that? it does have to assume the worst

so you would have to do a G28 and then G29 like this

G28
G29

fmalpartida, 2015-01-03:
Are you sure?
Currently have a colleague working on this and it is fully operational.
https://www.youtube.com/watch?v=L-_5KsAOVko#t=22

If you do:
G28
G29

it doesn't work if steppers are disabled. The way to fix it is just to decouple enabling and disabling the steppers with a known position. One way around it is to force in the code a home before probe.

boelle, 2015-01-03:
G28 does the home of all axis...
G29 does the autolevel...

since you fire the orders in one go it will do them right after each other... so the steppers should not be disabled

boelle, 2015-01-03:
oh... the video... what he did was pulling the axis while it was printing?

fmalpartida, 2015-01-03:
The whole point is that they are disabled. I couldn't get it to work with Cura generated gcode.

Yes in the video he is pulling on the X axis and recovers the position while printing. Also for the Y axis. Very cool!

boelle, 2015-01-03:
but who would want to disable steppers during print? i would not... not even if the steppers are not used...

fmalpartida, 2015-01-03:
This is a fairly common use case, steppers are automatically disabled (if configured) when no movements are scheduled for a particular axis. This allows steppers to work cooler. If your mechanics are sufficiently robust, there is no need to have additional torque on the axis. It is very common on the Z axis.

boelle, 2015-01-03:
the Z axis i can understand... they will only move on layer change mostly

gregrebholz, 2015-01-03:
DC motor + encoder (as in the video) is the same as servos ... quite different from stepper motors.  In the former the controller *measures* the position and signaling is to *set* the desired position.  Stepper motors work by taking individual steps clockwise and anticlockwise, and keeping track of a presumed position in software based on that.  The moment the drivers are disabled, the presumed position is completely unknown.  I don't think this is a bug at all.

boelle, 2015-01-03:
fmalpartida commented 25 seconds ago
Perhaps the easiest way round it, is to force a G28 with the G29.

fmalpartida, 2015-01-03:
The motor is controlled with a attiny + a encoder like a servo. It takes in step and dir like any normal stepper: step + dir.

If you want to ensure that we have a known position before doing an autoleveling, I think the safest would be to force a G28 with the G29. That way the idle loop doesn't kick in and you exactly know the position.

It is a bit strange how it works now and would need documenting: if you disable X or Y when not used auto bed levelling will not work.

boelle, 2015-01-03:
again this is mostly not a bug....

of course when steppers are disabled you need to first do a home and then right after the autolevel

a normal setup will not disable the steppers before 60 sec of no movement

gregrebholz, 2015-01-03:
I think we're agreeing.  The video doesn't apply to open-loop motion systems which are the majority of 3d printers supported by Marlin.  In fact, it doesn't use stepper motors at all, so I'm confused about why its being discussed with respect to the feature that disables stepper motors when inactive.

I agree that the section where you set DISABLE_X, DISABLE_Y, etc. could use a comment which mentions that whenever this occurs you will lose position and require homing before proceeding.

fmalpartida, 2015-01-03:
@gregrebholz, the video is just to comment on boelle's comment, not the purpose of the "issue" report.

I think we are homing-in to a resolution, I would go about this two ways:
1. force a home on autoleveling
2. add a comment in the DISABLE_X, DISABLE_Y indicating that if they are set to true, autolevelling will not work. At least with Cura, it doesn't work, I have a G28 just before G29 and it doesn't play ball.

gregrebholz, 2015-01-03:
Ok, we can set the video aside, but I have to point out there is no "stepping" at all in that video.  He is using DC motors.  They freewheel in either direction based only on the voltage applied to the single winding... positive voltage for one direction, negative for the other, and torque relative to the amplitude of voltage.  His attiny are probably accepting a "step+dir" command and translating that to a new desired position, but because the attiny has an encoder to "know" the position, it is capable of the recovery depicted... move current position to desired position, continuously.  When you DISABLE_{X,Y,Z} with stepper motors, @boelle is correct, no recovery of known position is possible.  For that reason the code in disable_x() that sets "axis_known_position[X_AXIS] = false;" is correct, and not a bug.

The problem then is that the DISABLE_X feature is an instantaneous disable.  As soon as the motion planner has nothing to do, the axis is disabled.  Since G28 homes, and then goes idle, the motion planner goes empty, position is lost, etc.  My hunch, however, is that G29 will probe the bed, then the motion planner will be empty, position will be lost, and you won't be able to print after that either (even if you force a home at the start of it).

We could build a timer into the DISABLE_X feature so there's a little while for you to get the next command in... but, that's actually the *default* behavior of Marlin, set in Configuration_adv.h as DEFAULT_STEPPER_DEACTIVE_TIME for 60 seconds.  I wonder if we shouldn't just remove the per-axis disable feature, and direct users to the global deactive time which they can dial down further if they'd like.  But, I don't see the benefit to an instantaneous stepper disable... maybe others do?  If so, we should add a comment to it that indicates it breaks auto bed leveling, and any other operation that allows the motion planner queue to empty.  I'm curious if a "pause" on a host would also break because of this.

fmalpartida, 2015-01-03:
This is a video from a colleague who is looking at replacing a stepper by a "servo". The current prototype uses a pro-mini and takes in step and dir and encoder position to plan the motion through as you mention. The PID loop controls the entire system.

Back on track, I agree with you that it is a bit overkill having an immediate disable. To be honest, I would just either eliminate the immediate disable feature per axis and go through the default behaviour. Should users wish to shorten the time, they can. If we keep it, I do think that we should add a comment stating what will happen though.

wgm4321, 2015-01-24:
I am very confused, I've been using ABL on a UMO and have never seen this "instant" disable nor have G28+G29 sequence fail due to disabled axes.

thinkyhead, 2015-01-24:
Allowing motors to sleep during a print can cause lost steps.

boelle, 2015-03-11:
@thinkyhead  i agree... once a stepper or any motor for that matter is disabled we dont know the pos... it can be moved by the user etc etc etc

so: is this still an issue with latest copy and latest non-beta IDE?

boelle, 2015-03-17:
closing as there was no response....

closed 2015-03-17 by boelle

nicksears, 2016-01-21:
I'm fairly certain the root of the conversation is the issue I'm running into, but no one seems to be talking about it the way I would describe it. I've noticed that since I've turned on auto bed leveling, the motors do not idle-off after the prescribed time. Even after homing with G28 and G29 consecutively, if I move x or y afterwards they never time out after the prescribed time. The only reason I can imagine is so that it NEVER loses position so it can safely home (I have safe homing on).

I have xyz set to not auto disable, but all e, including the active one do. yet with a timeout of 60 sec, they never let go:

latest marlin 1.1.0 on RAMPS 1.4 EEB with RRD fan extender

relevant settings in configuration.h:

#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
#define DISABLE_E true
#define DISABLE_INACTIVE_EXTRUDER true

configuration_adv.h:
#define DEFAULT_STEPPER_DEACTIVE_TIME 60

I guess I can see how it needs to always know it's location, but even when it should (i've already homed with g28 and g29) it asks to home with g28 before probing with g29 again (sometimes)