MarlinFirmware/Marlin #1301: Cant compile when REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER defined

monkeydave, 2015-01-03:
I am unable to compile Marlin after commit cb12161191be148906d31642b44af04248c3d6cc with a  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER defined. 
Compiles fine with commit c1df713e4d304643f3b4f713f9b9754786a6aa59.

Errors is :- 
ultralcd.cpp: In function 'void lcd_control_volumetric_menu()':
ultralcd.cpp:865: error: 'lcd_implementation_drawmenu_setting_edit_callback_float43_selected' was not declared in this scope
ultralcd.cpp:865: error: 'lcd_implementation_drawmenu_setting_edit_callback_float43' was not declared in this scope

@wgm4321 this was your commit, can you confirm this for me?

Note:- commit cb12161191be148906d31642b44af04248c3d6cc broke everything but it also introduced this error, the next commit d60efc3b5dcbf4e56afd6c37b8beac6188de90c3 is the first to give the actual error I am reporting here.

wgm4321, 2015-01-03:
yep, it was mine.   I noted that it would NOT compile for other controllers and languages...

boelle, 2015-01-03:
This will be put on the next milestone... for now we will need to get the last issue on milestone 1 sorted and then test out the code... then we set a new milestone with the next 4 errors starting with the oldest one first

monkeydave, 2015-01-03:
I see that now on your pull request, do you have a fix yet?

boelle, 2015-01-03:
me?

monkeydave, 2015-01-03:
No sorry not you @boelle

wgm4321, 2015-01-03:
I'm not sure how to fix.   How many controllers are support?
I certainly can't fix the language stuff as I barely speak English....
I can try to fix the float43 errors by duplicating all copies of float32 buyt how do you want to handle the language translations?

wgm4321, 2015-01-03:
OK so looks like just the HD44780 and something called dogm are supported.

Let me fix the dogm by dup'ing the routines and adjusting.   Someone else will need to test though.

Any suggestions on the language issue?  Should I just put the English phrase in the non-English files?

monkeydave, 2015-01-03:
I can test as I have the LCD
Not sure about the language issue

wgm4321, 2015-01-03:
Hmmm, I fixed the problem I caused but I think  there are other problems that I didn't introduce...

wgm4321, 2015-01-03:
What machine type are you trying to build for?

monkeydave, 2015-01-03:
I have an Azteeg X3 Pro to test with

alexborro, 2015-01-03:
@wgm4321 ,  what is the Define that enables this feature? It should be
undef by default (in configuration.h) and explained that works only with
2004 LCD.

About translations, you should always use English in all languages you
don't know. So it will not break the compilation of someone choose a
language other than English.

Cheers.

Alex.

wgm4321, 2015-01-03:
@alexborro , this isn't wrapped with a define because the feature it implements which is basically the M200 command is always available.

I submitted a pull request (1303 ) and alerted @monkeydave  so he can test.

I will submit another request to add the English defines in the non-English languages.

monkeydave, 2015-01-04:
Fixed with PR #1303 
Thanks for the quick fix @wgm4321

closed 2015-01-04 by monkeydave

wgm4321, 2015-01-04:
You are welcome.  Sorry I borked it in the first place.   I'm glad it is straightened out now.