MarlinFirmware/Marlin #1305: createTemperatureLookupMarlin.py domain error

gaby64, 2015-01-04:
sometimes c3 is negative

adding

if c3 < 0:
    c3 = c3 * -1

seems to work, i get a working table

stv0g, 2015-01-06:
Thanks, now I'm able to reproduce the error.

But I'm still investigating whether your fix is applicable.

Read this for an detailed explanation: http://www.thinksrs.com/downloads/PDFs/ApplicationNotes/LDC%20Note%204%20NTC%20Calculator.pdf

gaby64, 2015-01-06:
thermistor reported temperature seems to agree with thermocouple measured temperature using the resulting table

stv0g, 2015-01-06:
Okay, but the calculation doesn't matches with the Steinhart-Hart equation.
The correct solution is somehow complex:
http://m.wolframalpha.com/input/?i=-1%2Ft+%2B+a+%2B+b*log%28x%29+%2B+c*log%28x%29%5E3+%3D+0&x=0&y=0

gaby64, 2015-01-06:
but are Steinhart-Hart coefficients allowed to be negative?

boelle, 2015-01-06:
should be fixed... if not open new issue

commit df17a6a1 (2015-01-06): fix #1305 by proposed workaround (thanks gaby64)
closed 2015-01-06 by boelle