MarlinFirmware/Marlin #1310: G1 F12000 Exxx (retraction) freeze Marlin for 1-2 second

orangeudav, 2015-01-04:
I played with different Cura's options and tried to enable retraction. Without success but with strange behaviors. This gcodes freeze Marlin for 1-2 seconds:
```
G1 X78.345 Y123.734 E549.65927
G1 X79.020 Y124.063 E549.68549
G1 X115.591 Y95.350 E551.30928
G1 F12000 E545.80928
```
Is it arguments order error? Or 12000 too big?
This code was generated by Cura 14.09

alexborro, 2015-01-04:
It should not freeze, but anyway I will check the code processing this
command.

And I think 12000mm/min is unachievable with regular extruders..  Try
lowering this value in gcode to something about ~2000 and let me know
whether the error happens or not.

Cheers

Alex.

Wurstnase, 2015-01-04:
Marlin has a hardlimit of 65.535 steps/s. When you have rates bigger then this Marlin can freeze.

alexborro, 2015-01-04:
Humm,  it makes sense, otherwise it will result a overflow in the
plan_buffer_line... I think it is worth adding a check to avoid this..

I'm working on that code for a fix in CoreXY support, I will try fixing
such overflow.

orangeudav, 2015-01-05:
Yes, it working ok with smaller limit such as F3800

Wurstnase, 2015-01-05:
How much steps/mm has your extruder?

orangeudav, 2015-01-05:
@Wurstnase 1344.33 steps/mm

Wurstnase, 2015-01-05:
Can't believe that 3800mm/min will work. This would be 85k steps/s. The normal Mega should arrive max. 63k.

boelle, 2015-01-05:
maybe he meant 3800 steps / s

orangeudav, 2015-01-05:
Hmm, from my last working gcodes
```
G1 F1200 X36.906 Y57.554 E218.41600
G1 F1800 E212.91600
````
No, slicer nothing known about steps, only units (mm)

My settings for speed
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   {160.306,160.845,7918.158,1344.33} // steps/mm
#define DEFAULT_MAX_FEEDRATE          {500, 500, 2, 50}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}   
#define DEFAULT_ACCELERATION          3000    //  mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000
```

closed 2015-01-05 by orangeudav