MarlinFirmware/Marlin #1312: M140 S0 not always disable heatbed but remove temperature limit - infinite heat O_o

orangeudav, 2015-01-04:
I dont know how to repeat this bug, but i saw it more than 3 times. It happen after target temperature (110C) was reached.

brainscan, 2015-01-05:
Are you sure you haven't 'cooked' the heatbed mosfet? The standard ones on ramps aren't really up for some set ups and get very hot. My original one fried itself and then wouldn't switch off which produced a similar problem to the one you're experiencing.

orangeudav, 2015-01-05:
@brainscan, very interesting! I know than normal MOSFET's working mode not heat it, only during open-close and close-open cycles (very short). But, my MOSFET is very hot all time. I have custom scheme, cause standard RUMBA's port can not warm my heatbed.

nophead, 2015-01-05:
The MOSFET is heated when it is on because they never have zero resistance. Even 10 milliohms will dissipate 1W with 10A flowing through it.

To have a MOSFET than doesn't get too hot it needs to have a low RDSon when the gate is driven with 5V. I.e. a modern logic drive MOSFET where RDS on quoted at 4.5V and is 3-5 milliohms.

alexborro, 2015-01-05:
The regular Rumba's mosfet has RDSon = 6mOhms. I replace it by one with 1
mOhms so I can drive 30A with no problem at all.

Cheers.

Alex.

brainscan, 2015-01-05:
I swapped mine for a beefier one and it doesn't even get warm, even without a heat sink. I just thought it worth mentioning as my old one would switch on but wouldn't switch off without a hardware reset so the bed would just keep getting hotter and hotter. It's worth checking hardware as there seems to be a lot of 'potential bugs' that are actually hardware faults.

boelle, 2015-01-15:
think we can close this one as OP have not responded and its 10 days... can be reopened as always

closed 2015-01-15 by boelle