MarlinFirmware/Marlin #1315: Compiling errors

spac3man, 2015-01-05:
Hello,

With the most recent version of Marlin,  I now receive the following errors:

ultralcd.cpp: In function 'void lcd_goto_menu(void (*)(), uint32_t, bool)':
ultralcd.cpp:199: error: 'lcd_set_custom_characters' was not declared in this scope
/dogm_lcd_implementation.h: In function 'void lcd_status_screen()':
dogm_lcd_implementation.h:88: error: too many arguments to function 'void lcd_implementation_init()'
ultralcd.cpp:271: error: at this point in file
/dogm_lcd_implementation.h: In function 'void lcd_update()':
dogm_lcd_implementation.h:88: error: too many arguments to function 'void lcd_implementation_init()'
ultralcd.cpp:1192: error: at this point in file
ultralcd.cpp: In function 'void lcd_finishstatus()':
ultralcd.cpp:1296: error: 'progressBarTick' was not declared in this scope

Can anyone tell me what I've done wrong?

Thanks

boelle, 2015-01-05:
most recent.... is that the dev branch or stable?

monkeydave, 2015-01-05:
I have managed to emulate the same errors with RepRap Discount Full Graphic Smart Controller & LCD_PROGRESS_BAR defined in the Development branch. 
@spac3man if you comment out #define LCD_PROGRESS_BAR  on line 301 of Configuration_adv.h this should allow you to compile and get up and running.

spac3man, 2015-01-05:
Sorry I didn't clarify.  It was the Dev Branch.  I will try your suggestion MonkeyDave and report back

Thanks

spac3man, 2015-01-06:
I commented out the Line 301 in Configuration_adv.h as monkeydave suggested and the following error is what occurred.  I am using the Dev branch of Marlin with Ramps 1.4 configured for a Prusa I3 with 2 extruders and heated bed.  I also have a ReprapDiscountFullGraphics display.

In file included from ultralcd.cpp:39:
/dogm_lcd_implementation.h:36:20: warning: U8glib.h: No such file or directory
In file included from /dogm_lcd_implementation.h:38,
                 from ultralcd.cpp:39:
/dogm_font_data_marlin.h:12:25: warning: utility/u8g.h: No such file or directory
In file included from /dogm_lcd_implementation.h:40,
                 from ultralcd.cpp:39:
/ultralcd_st7920_u8glib_rrd.h:9: warning: ignoring #pragma GCC optimize
In file included from /Configuration.h:788,
                 from /Marlin.h:22,
                 from /temperature.h:24,
                 from ultralcd.cpp:1:
/thermistortables.h:10: warning: only initialized variables can be placed into program memory area
/thermistortables.h:173: warning: only initialized variables can be placed into program memory area
In file included from /dogm_lcd_implementation.h:37,
                 from ultralcd.cpp:39:
/DOGMbitmaps.h:4: warning: only initialized variables can be placed into program memory area
/DOGMbitmaps.h:131: warning: only initialized variables can be placed into program memory area
/DOGMbitmaps.h:156: warning: only initialized variables can be placed into program memory area
In file included from /dogm_lcd_implementation.h:38,
                 from ultralcd.cpp:39:
dogm_font_data_marlin.h:13: error: 'u8g_fntpgm_uint8_t' does not name a type
dogm_font_data_marlin.h:172: error: 'u8g_fntpgm_uint8_t' does not name a type
In file included from /dogm_lcd_implementation.h:40,
                 from ultralcd.cpp:39:
ultralcd_st7920_u8glib_rrd.h:43: error: 'u8g_t' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: 'u8g' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: 'u8g_dev_t' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: 'dev' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: expected primary-expression before 'msg'
ultralcd_st7920_u8glib_rrd.h:43: error: expected primary-expression before 'void'
ultralcd_st7920_u8glib_rrd.h:43: error: initializer expression list treated as compound expression
ultralcd_st7920_u8glib_rrd.h:44: error: expected ',' or ';' before '{' token
/ultralcd.h:21: warning: 'blink' defined but not used
ultralcd.cpp:27: warning: 'manual_feedrate' defined but not used
/ultralcd_st7920_u8glib_rrd.h:24: warning: 'void ST7920_SWSPI_SND_8BIT(uint8_t)' defined but not used

Wurstnase, 2015-01-06:
You've copied u8glib to program files > Arduino > libraries?

ErikZalm, 2015-01-06:
This issue was moved to MarlinFirmware/Marlin#2

spac3man, 2015-01-06:
I checked to see if the u8glib library was there and indeed it was.  From https://bintray.com/olikraus/u8glib/Arduino I downloaded and copied the library again and now with the commented line as per monkeydave, it compiles properly.  I don't know why it didn't the first time.  

Thanks again for your help.

monkeydave, 2015-01-06:
This should probably be left open and the title changed, there is a bug with LCD_PROGRESS_BAR  & DogLCD defined together.

JoshyuNL, 2015-01-08:
This issue  #1257 is looking similar to this from my quick reading of this.
Since the babystepping is no longer an broken,  #1257 is only about the LCD_PROGRESS_BAR not compiling when on a RepRap Discount Full Graphic Smart Controller.

Just to clarify, I get the same bug if I enable the progress bar.

Grogyan, 2015-01-09:
RAMPS 1.4
No SD card inserted but enabled
RepRap Discount Full Graphic Display
Max6675 Thermocouple

Display stuck on Marlin startup screen with vertical bar in the middle
Milestone 2

stv0g, 2015-01-09:
@Grogyan which version are you using? Maybe you can try my version vis LogoLCD fixes: http://github.com/stv0g/Marlin/tree/LogoLCD

Grogyan, 2015-01-09:
@stv0g 
I'm using Milestone 2.
Do you have a pull request in your repo that has the changes?

thinkyhead, 2015-01-09:
LCD_PROGRESS_BAR support still needs to be added to the DOGM implementation. So, don't enable it for that LCD yet.

stv0g, 2015-01-09:
No, its in the LogoLCD branch of my fork, still testing...

thinkyhead, 2015-01-09:
@stv0g I wrote the implementation for Hitachi, so I should probably finish the job. But I won't be able to get to it until tomorrow.

thinkyhead, 2015-01-10:
#1343 is in the queue and ready for testing.

stv0g, 2015-01-10:
Oh yeah, I didn't knew that about the Hitachi Implementation.

Nice work!

thinkyhead, 2015-01-10:
Cleanup is always a risk, haha! Hmm, I've gotten pretty good at merging through git, but it sure takes a while. Since it's not a huge file, if I could grab your changes, I will add them onto my pull request. I have a pretty decent diff tool (ok, BBEdit) that speeds the process. I haven't gotten used to this new Collaborator tag yet, but will explore the higher git powers available there soon... Planning to stick to a policy of not pulling my own requests without peer review. (Be kind.)

boelle, 2015-01-10:
should be fixed with PR #1343

closed 2015-01-10 by boelle