MarlinFirmware/Marlin #1317: Set GitHub default branch to 'Stable'

stv0g, 2015-01-06:
I'm still landing on the outdated 'Marlin_v1' branch when visiting https://github.com/ErikZalm/Marlin .

nophead, 2015-01-06:
Also searches of the repository all seem to fail, presumably because the default branch is empty. I don't know how you change the branch it searches in.

stv0g, 2015-01-06:
The default branch can be changed in the repository settings.

boelle, 2015-01-06:
i have asked @ErikZalm for this over his private email

will do once more

boelle, 2015-01-06:
done... he is the only one able to do it, collabs do not have enough rights for it

collabs dont have the settings... only @ErikZalm have those

closed 2015-01-06 by boelle

stv0g, 2015-01-06:
thanks :+1:

boelle, 2015-01-06:
what i did was delete all files in V1 and change the readme so people at
least are shown the right way

@Erikzalm are very busy and it took me ages to get added as collab....

So well yes i did as best i good with the rights i have [?]