MarlinFirmware/Marlin #1318: Commit 9472bbe causing constant "ok" responses when no timeout is set

wgm4321, 2015-01-06:
@i-make-robots

Commit 9472bbe causing constant "ok" responses when no timeout is set

This code causes the firmware to constantly send "ok" when the printer is idle as max_inactive_time is by default 0.   Note that the kill() is protected by testing for a non-zero max_inactive_time.   However, the "ok" response generation is not so when max_inactive_time is zero, millis() - previous_millis_ok) >  max_inactive_time/4  is always true.

```C
   if( (millis() - previous_millis_cmd) >  max_inactive_time )
     if(max_inactive_time)
       kill();

	// If 'OK' is garbled on sending PC won't receive it.  Both machines will wait on each other forever.
	// This resends OK if nothing is heard from PC for a while to avoid this bad case.
        if( (millis() - previous_millis_ok) >  max_inactive_time/4 ) {
		SERIAL_PROTOCOL(MSG_OK);
		previous_millis_ok=millis();
  }

```

ErikZalm, 2015-01-06:
This issue was moved to MarlinFirmware/Marlin#1

avluis, 2015-01-07:
Just saw this. Glad it wasn't just me (after updating my fork to latest dev).

ErikZalm, 2015-01-07:
I reverted the patch.
It is not a good idea to send random oks. (Unless tested it works ok with all host software)
This is a known problem. If the fix was this simple it would be implemented a long time ago.
The reprap protocol is not very robust. But it is difficult to change.

boelle, 2015-01-07:
Q... are you not also behind cura? if so could cura not be the first to use a better protocol?

but if a new protocol is needed how many host softwares need to be behind such a change? pronterface is one i guess...

i-make-robots, 2015-01-07:
Not good because?

Printer recieves line N.
Printer Sends OK twice.
PC receives extra OK.
PC sends next line N+1 and N+2.
Printer says BADLINENUM?

If Marlin designs to make host happy and host designs to make Marlin happy
it's a giant game of chicken.  Someone has to go first.

boelle, 2015-01-07:
host uses a protocol.... marlin has to use the same...

problem here is the protocol is not robust enough... its does not have
anything to do with making anybody happy... the protocol has to be
changed... of course more easy said than done

nophead, 2015-01-08:
If the OK is garbled the host should time out waiting for it and send something to invoke one. It shouldn't wait forever, so I think there needs to be a change to the protocol even if it is weak.

i-make-robots, 2015-01-08:
I agree, Chris.  One change in Marlin is easier than one change in every
host.

alexborro, 2015-01-08:
I'm facing a problem with such "OKs".... I use the serial for debugging the
code - unfortunately I don't have a JTAG...

Check the result of a single "M114" command sent by pronterface:

​

Cheers.

Alex.

ErikZalm, 2015-01-08:
In my opinion think the host should timeout.
The firmware does not know when to expect a command. The host knows it send a command and expects an acknowledge.
The firmware can not acknowledge on something it does not receive.
Don't make a host problem a firmware problem.
At best we can send a "waiting for new command".

bkubicek, 2015-01-08:
one might abuse a Serial info message for that, that can be sent to the
host. A default hosts will just show it on the screen, a clever host might
understand it.

Bernhard

nophead, 2015-01-08:
Octoprint already does time outs and resends so I don't think it hangs on a missing OK.

Note that with correct hardware and software you should never need a resend. The USB protocol has a transport layer than already does CRC checks and resends. The data that arrives should always be correct even in the presence of noise or you get a disconnect. This is what happens for me with Marlin on Melzis. I never ever see corrupt data. 

Boards that do must either have bad track layout for the serial link between the USB chip and the atmega or perhaps baud rate errors or buffer overruns in the firmware. If you fix your hardware there is no longer a need for an error correcting protocol. USB already provides that.

i-make-robots, 2015-01-08:
That's what OK means to me.

lordofhyphens, 2015-01-11:
That  commit breaks pretty much everything, IMHO, current octoprint and Printrun. Printrun is less broken in that it just fills the UI with messages.

That commit is sitting in master branch. 

I've reverted it for my own use.

boelle, 2015-01-11:
we dont use the master branch

lordofhyphens, 2015-01-11:
Then what branch should people be using when trying to download/compile? Development? I didn't see a stable in this repo. -.-

boelle, 2015-01-11:
yep dev.... we are still bug fixing, a stable release will not come before all bugs are sorted and we have tested the code to be stable... thou we lack coders that can help out and that can worked focused on the bugs

lordofhyphens, 2015-01-11:
I'll see what I can do to test stuff, but from looking at the many 
issues on github, I haven't noticed any of these bugs on my config.

boelle, 2015-01-11:
ok... https://github.com/MarlinFirmware/Marlin/milestones

these are the 2 milestones set so far... in them the most important bugs are listed....

if you are good at doing code we can use all the help we can get

boelle, 2015-01-15:
@ErikZalm was this one fixed by your revert?

boelle, 2015-03-11:
i will assume this one is fixed 

closing... reopen if needed

closed 2015-03-11 by boelle