MarlinFirmware/Marlin #1322: reprap discount LCD display not working

hawkan-rules, 2015-01-06:
I've defined  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in configurations.h. When I start the electronic it just shows the backlights and no text or nothing. 

Versions:
Running Marlin commit: 3d9f4ac7576e6f23a70f8d7986a6388d889ef9e4
U8glib:  version 1.14  and  version 1.13
Arduin: 1.0.6

I've tried a few older versions but havent got it working! :(

My printer is an Prusa I3 from replikeo.com http://replikeo.com/en/KIT001

hawkan-rules, 2015-01-06:
Auto bed level is also enabled if that makes any difference.

alexborro, 2015-01-06:
Are you sure your LCD is working properly with your board? Have you ever
got them working together with any other firmware?

hawkan-rules, 2015-01-06:
Nope havent been able to see anything on the display at all! Is there some way to troubleshot the connection with the display??

stv0g, 2015-01-07:
There are cheap logic analyzers out there. Or use an oscilloscope.

JoshyuNL, 2015-01-08:
For me it works fine. Make sure you're wires are all connected correctly like mentioned above.
I think I've had it before that my wires were switched and I got a similar issue.

I've updated my copy of marlin this evening, haven't checked the changes in the last few days to see if they did anything to it but someone would've probably said something by now if they had.

boelle, 2015-01-09:
@hawkan-rules still having trouble?

hawkan-rules, 2015-01-10:
@boelle yes! 

I've also tried to rotate the cables 180 degress as well. Noting seems to be helping, I'm trying to get the display replaced by the seller as i suspect that it is broken.

boelle, 2015-01-10:
also remember we are working on it

any chance you use thermocouples and not thermistors?

Den lørdag den 10. januar 2015 skrev hawkan-rules <notifications@github.com

hawkan-rules, 2015-01-11:
@boelle I'm using a thermistors

Grogyan, 2015-01-11:
I only get this in the current Development branch when using the MAX6675 thermocouple
What Thermistor are you using?

hawkan-rules, 2015-01-13:
@Grogyan: Thermistor NTC100K3950 3950-1%

This is how my configuration.h looks like:
//1 is 100k thermistor - best choice for EPCOS 100k (4.7k pullup)

#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_BED 1

boelle, 2015-01-15:
@hawkan-rules did you get the display replaced?

hawkan-rules, 2015-01-15:
@boelle Nope and I'm guessing it might take two or three weeks before I get it! :S If you want you guys can resolve this issue and I'll repoen if the replacement isnt working.  Thanks for the help though!

boelle, 2015-01-15:
reopen? we have not closed it... but these displays are on the SPI bus and we have several issues where its the SPI bus.... spi devices each need their own CS line so the MCU can tell what device it wants to talk to... as long there is only one device it MIGHT work... but really even if only one device it should use the CS line...

Grogyan, 2015-01-15:
Problem is not related to thermocouple. I can't replicate it unless the cables are switched over.

Grogyan, 2015-01-16:
Interesting to note that I have either blown up my LCD/Arduino, while testing out some other stuff.
Currently experiencing the same as OP, same config - no thermoouple.
Meaning that it is a hardware problem.

hawkan-rules, 2015-01-29:
Got my new display and it worked! :)

closed 2015-01-29 by hawkan-rules