MarlinFirmware/Marlin #1326: QTMarlin

boelle, 2015-01-06:
i read about this tool.... any idea why it has not been packed together with marlin ?

tmilker, 2015-01-06:
It has it's own repository, it doesn't need to be merged with Marlin.

boelle, 2015-01-06:
oki... how would i make a windows app out of it?

tmilker, 2015-01-06:
You would have to get a build environment setup for QT in Windows and then port any Unix specific code/paths/etc.  It looks like it may be partially setup for Windows.

bkubicek, 2015-01-07:
My tools is horribly outdated. The idea was to have a detection of lost
steps in the firmware.
Then have a pc program try different acc/velocity settings, in each axis,
drive the setting, and find out if steps were lost or not.
By I hoped to be able to measure the max-torque curve of the system
stepper+printer, to know under which max velocity which acceleration is
possible.
The lost step detection was thought in the following way.
Home Axis
fast move axis away from endstop
fast move axis towards endstop, but not touching (hopefully). Then slowly
move and measure the distance until endstop is triggered.
If this distance is aprox. what it should be. Good. Otherwise: steps lost.
Honestly, it never worked well.

Also, as a "bonus" it was intendet to have settings viewing and writing.
But why use that if you can use the display.

Bernhard

boelle, 2015-01-07:
I agree on the display part...

just a shame the other stuff did not work

@bkubicek remember to accept your invites as owner etc...

brainscan, 2015-01-07:
I always wanted to try this out but never got around to it. Shame it doesn't work as you had intended. I think it's a brilliant idea though as it would be great if we could run a program that outputs our printers optimum acceleration/speed. Do you think it just needs a little more work or is the whole thing a dead end?

boelle, 2015-01-07:
+1 from here... could maybe even do it automatic if both a min and max switch is mounted....

or temp a block in the middle with x and y switches that the nozzle touches.... or something like it

brainscan, 2015-01-07:
I really wish I already knew how to develop things like this, learning code is so hard!!! You all make it seem so easy. I definitely like the idea of having something automatic that starts off slow and then starts ramping up the speed and acceleration until errors occur. It would simplify and speed up the commissioning of a new printer or upgrades. Would be great if it can help work out appropriate jerk values too so they all work in unison.