MarlinFirmware/Marlin #1329: Logo for Marlin firmware

boelle, 2015-01-07:
It was mentioned on IRC chat the other day, so i thought of marlin the fish and started to look arround

http://imgur.com/gqhNTDS

its just starters.... 

thoughts?

scotty1024, 2015-01-07:
Usage rights eg is the image open sourced at the same license level as Marlin source code? The whole Firefox vs Ice Weasel fork was over that frickin' Firefox logo image open source license.

boelle, 2015-01-07:
it's based on an free drawing i found via google....

but BASED on... i have redrawn parts of it as i was not happy with the org.

ErikZalm, 2015-01-07:
There already is a logo for Marlin.
I will add it soon.

boelle, 2015-01-07:
ok :-D

boelle, 2015-01-07:
can it be found on internet? if so i could add it?

boelle, 2015-01-07:
anyways will close this one as the logo is allready there

closed 2015-01-07 by boelle

boelle, 2015-01-07:
@ErikZalm maybe drop the logo in the repro so we can have it shown on LCD displays also instead of the reprap thing

scotty1024, 2015-01-07:
Not to beat the horse but repos like Marlin are based around the original authors of the contribution directly checking it into the repo. Or someone checking in something created by someone else that has an attached compatible license. Images are problematic to check in from a third party as they generally don't have attached licenses. 

I highly recommend a LogoLicense.txt file be checked in spelling out the license of the logo if it isn't 100% created by the person that checks it in.