MarlinFirmware/Marlin #1337: reprap discount full graphics LCD bootup screen not correctly displayed.

JoshyuNL, 2015-01-08:
So this is really a non-issue and something that has the lowest priority of all things I would think.

But just thought I would document it now that I've noticed it.
While the printer is booting up, it show the new Marlin logo on the left and some notes on the right.
On my display these two columns seem to be separated by a thick vertical line which cuts off part of both columns.

My guess is that the scaling is off or that that vertical line should not be there at all.
It looks like it doesn't really reach the bottom of the sceen either, since the last line of text on the right goes a few pixels lower. It does seem to go all the way to the top.

[image]

boelle, 2015-01-09:
one argument for reducing the info to bare minimum is that this is something that will reduce code size... not by much but everything counts

stv0g, 2015-01-09:
> should we keep the lower part of the right?
>
> or just have marlin and ver. no ?

Bo,

did you removed the padding/whitespace from the logo before converting it to HEX?

Looks like there are the same whitespace like in the GitHub Organization Logo.
I added another version of the Logo without the padding/whitespace.
This one is called „Marlin Logo GitHub.png“ and is used by README.md.

For the firmware splash screen we should use the version without whitespace/padding.

Steffen Vogel
Robensstraße 69
52070 Aachen

Mail: post@steffenvogel.de
Mobil: +49 1575 7180927
Web: http://www.steffenvogel.de
Jabber: steffen.vogel@jabber.rwth-aachen.de

boelle, 2015-01-09:
i just did remove as much padding arround the logo as i could this morning and updated the file

boelle, 2015-01-09:
i was going to commit the logo but you have beat me to it

JoshyuNL, 2015-01-09:
The new file does change it a bit. But it's still not fully fixed.
The logo moved to the top and the thick line is still there where the logo is.

[image]

stv0g, 2015-01-09:
The „fitted“ logo I’ve committed still a 3-5 pixel transparent border.

I’dont know if we should remove this too for the splashscreen.

Steffen Vogel
Robensstraße 69
52070 Aachen

Mail: post@steffenvogel.de
Mobil: +49 1575 7180927
Web: http://www.steffenvogel.de
Jabber: steffen.vogel@jabber.rwth-aachen.de

boelle, 2015-01-09:
try remove that too.. the vertical line is 4 pixels

ErikZalm, 2015-01-09:
If you are changing the Logo then also remove ErikZalm.

boelle, 2015-01-09:
oki doki... my idea:

strike ErikZalm and replace it with Marlin

strike DOFM128 and replace with ver info

strike everything else

boelle, 2015-01-09:
download dogm_lcd_implementation.h also 

how does the ver info side come up?

JoshyuNL, 2015-01-09:
Like this:

[image]

Maybe, if this is all the info we would want on the splash screen, try adding the marlin logo in the center and put the text below it?
I have no more time to test right now, will take another look here tomorrow.

boelle, 2015-01-09:
ok.... thanks anyway.... 

have an idea on how to move down the logo again ... and then we just need to remove the bit of vertical line

stv0g, 2015-01-09:
Hi,

I rewrote this splashscreen. But unfortunatly I'm unable to test it without hardware.

Take a look in my repo: https://github.com/stv0g/Marlin/tree/LogoLCD/

I would be happy aboit some feedbackbefore merging it into the official repo

alexborro, 2015-01-09:
Sorry my ignorance, how do you guys convert the image to hex?

boelle, 2015-01-09:
http://www.digole.com/tools/PicturetoC_Hex_converter.php

stv0g, 2015-01-09:
@JoshyuNL Can you test my version? Please note, that there are 3 new Configuration variables in `Configuration.h`.

https://github.com/stv0g/Marlin/tree/LogoLCD/

Please also try it once with: `#define START_BMPHIGH 1` in `Configuration.h`
This is a high resolution version of the boot screen logo.
I just want to see if not variants are working before pushing this to the main repo.

Thanks :-)

closed 2015-01-10 by boelle

Shrikant888, 2016-09-26:
Hello All!
What is meant by "const unsigned char start_bmp[574] PROGMEM". In this what is 574? I know it may be arrary size. But how come 574 only. Please reply.

thinkyhead, 2016-09-26:
@Shrikant888 

- `const` means the data will not be alterable in code.
- `unsigned char` means each element of the array is an unsigned character.
- `start_bmp` is the name of the array variable.
- `[574]` means the array contains exactly 574 bytes.
- `PROGMEM` means the array will be stored in the program memory.

This particular array stores the bitmap data for the boot screen.

If you look at the newer code you will see that the arrays for the bitmap (and the custom bitmap) is 512 bytes, which corresponds to the dimensions of the screen. I'm not sure what the extra 62 bytes was for in the older code.

Shrikant888, 2016-10-03:
Please let me ask again sir. I have designed my own 128x64 resolution logo and I wanted to display it. So, process will be like first my 128x64 logo will have to come and then regular REPRAP logo should come. I am not changing REPRAP and then rest will be same. What changes I should make in dogm_lcd_implementation.h and in DOGMbits.h.. Please tell.

thinkyhead, 2016-10-04:
See the files in `Marlin/example_configurations/Hephestos_2` for an example of making a custom boot screen, with links to the tools used.

PR #1348 (2015-01-10): Adding new Marlin to splash screen
I merged @thinkyhead 's latest fixed for the progress bar issue to my splash screen rewrite

  - Added new Logo to splash screen
  - two versions (high/low resolution) are available
  - padding and alignment is calculated automatically

fixes #1337