MarlinFirmware/Marlin #1338: Single wall bug

barneycg, 2015-01-09:
I know this was closed in bug https://github.com/alexrj/Slic3r/issues/1837 but I'm still seeing the following behaviour:

I've been using Slic3r 1.1.7 and I've hand checked the gcode and the start corner x and y coordinates are the same for every level. I measured the position of the bed on the y axis compared to the frame on my Prusa i3 at the first later and again on the last layer, and the bed is indeed 5mm closer to the frame at the end (the z axis is at 20mm at this point). The angle of the lean is consistent across prints and too regular within each print to be slippage but I could see or hear no evidence of it while watching several test prints.

The test cube was printed with layer height 0.25 with an extrusion width of 0.5 and 0.55 and 0.3 with extrusion widths of 0.6 and 0.66 ... all exhibited the same behaviour. Changing Slic3r to do 2 shells or to use infill "fixed" the problem and resulted in a very nice cube.

I tried updating Marlin to the stable version in the Repo and still no change, but I've yet to try another Slic3r but will so at the next opportunity.

alexborro, 2015-01-09:
> I know this was closed in bug alexrj/Slic3r#1837

I'm sorry to tell you but this is not a firmware issue.

barneycg, 2015-01-09:
I'm sorry but I'm not sure I agree.  Its clear it is not the gcode and the fact that 3 printers (two from the other issue and they were of different physical designs) and mine exhibit the exact same symptoms kinda suggests that it isn't mechanical either.  That leaves only Firmware or the electronics.  I've swapped motors and stepper drivers between Axis' ... I suppose it could be the RAMPS or the Arduino.

nophead, 2015-01-09:
What stepper drivers are you using, are they opto coupled? Looks like a
timing error on the direction signal.

alexborro, 2015-01-09:
@barneycg , you are saying your parts are tilted about 5mm in ZY plane and you think the firmware can be the culprit. 

Otherwise, there are thousands of Marlin users that daily print high precision parts - me included - without any issue. And believe me, 5mm would be a deviation like a Grand Canyon in those parts..

So we have a dead-lock here.. I think it deserves a little more investigation on your side so you can point us whats is going wrong with your firmware.

Cheers.

Alex.

boelle, 2015-01-09:
yep... how can we reproduce the  error?

boelle, 2015-01-09:
maybe a clue here?

https://github.com/MarlinFirmware/Marlin/issues/853#issuecomment-69374779

barneycg, 2015-01-09:
yeah there has been a load of discussion on #853 where you can see some of the things I've tried :)

boelle, 2015-01-15:
so you have the same problem but created a new issue? 

and as @alexborro said you need to point us to where you think the problem is.... you must point us to where marlin does not follow the Gcode its given.... 

even a non-coder like me could not imagine that the coders have hidden an easter egg that would not follow the gcode its given

barneycg, 2015-01-15:
The other issue was marked as closed and I couldn't re open it.  Since it was still a problem that needed fixed I raised an issue.  As it turned out one of the other devs reopened it and all the discussion and fix happened in the older issue.  I'm going to close this.

closed 2015-01-15 by barneycg