MarlinFirmware/Marlin #1339: PWM on pin 12 (PB6) on an arduino mega (ATMEGA2560) is not possible.

Nerveasy, 2015-01-09:
Hi, 

Is it normal that an analogWrite to PIN 12 (has hardware PWM support) is not working with a virgin marlin install? It also does not revert to standard digital behaviour (0 if < 128 and 1 if >128)... I can however us it as a digital pin.

I have made sure that in pins.h pin 12 is not defined. I tried to search what timers marlin uses but I cannot find any clues and I got a bit lost in fastio.h ... :(

lordofhyphens, 2015-01-11:
Marlin uses pin 12 on several motherboard configurations, but I can't suss out which is which. Are you sure that you are using a motherboard config that is not using that pin? Otherwise, shouldn't be a reason why it doesn't work. 

http://arduino.cc/en/Reference/analogWrite

Nerveasy, 2015-01-12:
Hi,

I changed all the pins in pins.h that where routed to pin 12 just to be sure. And I have still digitalWrite control over pin 12 but no PWM. The strange thing is when you do an analogWrite to a pin that is not PWM capable it will act as a digital pin (lowwhen <128, high when >128) but even this does not work with pin 12 I can only perform a digitalWrite.

thinkyhead, 2015-01-24:
Just in case, check the resistance between pin 12 and neighboring pins, in case there might be a short. It's a long-shot, but best to eliminate shorts early in debugging.

velbug, 2015-01-25:
There is no short.. I checked it on multiple boards and also measured some pins. The fact I can get the pin high when addressing it as an digital pin tells me it is propably not an electrical issue, but more something to do with marlin and how it takes control of timers (have no idea). If I send a simple pwm only sketch to the board instead of the whole marlin firmware the pin spits out a perfectly fine pwm signal...

Nerveasy, 2015-01-25:
There is no short, I checked it on multiple boards and also measured the pin. The fact I can get the pin high when addressing it as an digital pin tells me it is propably not an electrical issue, but more something to do with marlin and how it takes control of timers (have no idea). If I send a simple pwm only sketch to the board instead of the whole marlin firmware the pin spits out a perfectly fine pwm signal...

thinkyhead, 2015-01-26:
@Nerveasy What are the boards you tried? I want to look closer at the pins.h for those. I wonder if there's a limit on the number of PWM pins you can run simultaneously, or if there are other forbidden combinations. According to [Arduino documentation](http://arduino.cc/en/Tutorial/SecretsOfArduinoPWM):

    Some combinations of bits that you might expect to work are reserved,
    which means if you try to use them, they won't work.
    For example, toggle mode doesn't work with fast PWM to 255, or with output B.

boelle, 2015-04-01:
should this be labeled as a bug or?

thinkyhead, 2015-04-02:
We haven't seen any bad effects for most common boards involving this pin, and it's not yet verified as a bug. I will close this for now. It can be re-opened if it is causing trouble with our supported configurations.

closed 2015-04-02 by thinkyhead

lukeweston, 2015-08-26:
You can't output PWM on that pin because timer1 is used in Marlin for stepper control - have a look at stepper.cpp