MarlinFirmware/Marlin #1340: Move printhead aside when print is stopped by user

Drayson80, 2015-01-09:
Additionally to the current implementation of the lcd_sdcard_stop functionality (stop, release steppers), the function shall be extended to lift the printhead a few mm and move it to a configurable position (homing might be a bad idea as the 0/0 could be a position middle of the printbed).

PS: unfortunately I´m not a SW engineer nor have programming skills so I open this as a feature request as I can´t implement it by myself...

alexborro, 2015-01-09:
It is possible and quite easy to do, but there are few points to think
about.

Moving the head to X=0 and Y=0 is ok.. Raising the head is a problem if the
machine doesn't have a Z_Max endstop.

Cheers.

Alex.

thinkyhead, 2015-01-09:
Raising the head ought to be safe since firmware is configured with the MAX Z position, and being near the very top is extremely rare. As long as it's during a print, or at the end, it should be no less a problem than moving Z up during printing.

Drayson80, 2015-01-10:
I thought of raising the head just by 1 or 2 mm - just ehough to move away safely and not knocking on something else.

lordofhyphens, 2015-01-11:
@thinkyhead It's always possible that MAX_Z is incorrectly configured or software endstops are disabled. If you only move X/Y, no need to lift Z. 

Initial reasonable solution may be to just home the head in X or X/Y.

Laura3, 2015-01-12:
@lordofhyphens, rehoming may throw of the positioning slightly: bad idea.

There is a need for z lift when printing thin tall objects. Perhaps having this configurable would be best.

lordofhyphens, 2015-01-12:
What about caching where home should be and going there? Shouldn't affect
positioning more than any other move.

This could also go into the slicer. I know slic3r already will do z lift on
retract if you ask.

Laura3, 2015-01-12:
Maybe we could run some custom gcode from config.h when event 'pause' happens..

bkubicek, 2015-01-12:
there is the macro functionality which might or might not be used for that.

Bernhard

Drayson80, 2015-01-12:
I would suggest to keep things simple and don´t mix too much.
The stop-functionality (maybe rename it to "abort print"...) is one issue, the implementation of the pause-function is another, but let´s make one step after another...

A dedicated stop/abort-implementation is from my point of view key in each system.
A simple implementation from a gcode point of view would be
- G91     relative positioning
- G0 Z1  raise head by 1mm
- G0 X0 Y0 or to a predefinable position - or just G28 X0 Y0
Please keep it simple... I kow there are so many ideas for fancy implementation, but again, please keep it simple... :-)

Drayson80, 2015-01-12:
I would suggest to open an additional feature request for pause implementation - maybe based on the change filament function by berhard ?

Drayson80, 2015-01-12:
Sorry, wrong button...

Drayson80, 2015-01-12:
@ all: I´d like to ask if we should open an additonal feature request for the pause-function implementation (maybe derived from bernhards change filament function?)

boelle, 2015-01-16:
features requests does not belong in milestones.... bugs only

Drayson80, 2015-03-16:
Has anybody news on this issue? 
@ alexborro, I would help but have no programming skills so the only thing I could do is offer assist in testing at an Ultimaker Original...

alexborro, 2015-03-16:
Thanks @Drayson80 for offering help.. I will try to do that this week.

Cheers.

Alex.

boelle, 2015-05-31:
closing this one as to little have shown interest and not activity since march. if alex wants to do this he can always submit a PR

closed 2015-05-31 by boelle

Drayson80, 2015-06-01:
:-( sad day... If I would be able to do it by myself, I would do, but I´m a mechanical engineer...
Same for a correct "pause" implementation.

Nevertheless, understandable...

thinkyhead, 2015-06-08:
@alexborro We miss you buddy. Praying to the syntax gods for your safe return…