MarlinFirmware/Marlin #1341: More CODERS needed

boelle, 2015-01-09:
we could use more coders on marlin....

PxT, 2015-01-13:
A barrier to entry is that the code is quite complicated for a casual contributor.  Maybe if there are some open issues which would be easy for a new person, or don't require a complete understanding of the entire code they could be tagged with a new keyword so that someone would have an idea of where to start.

NarimaanV, 2015-01-13:
@PxT ^^^^ +1 As an intermediate programmer, this is my exact issue with trying to help out with marlin. It seems unless it's a simple issue, it takes days to understand just that part of the code, and then even longer to try and add something while taking into consideration every single other aspect of Marlin that could possibly conflict with adding anything...it's all kind of overwhelming for someone who isn't 100% familiar with how everything in Marlin actually works. A new label isn't a bad idea, but I wish there could be someway of adding comments throughout the entire code to try and make it easier to understand what does what, and what needs to be considered when trying to change anything. I don't know how we would be able to do that (as in exactly how much information is enough without adding paragraphs of comments between lines of code), but I think it's something we definitely need to consider.

galexander1, 2015-01-13:
Embedded development does have a steep learning curve.

This is just my own two cents, but here it is: Just dig in!  Learning
something like how the planner operates, how the G-code parsing interacts
with the interrupt-driven state machine, it is really educational.  Just
the experience of poking your head into the buried mysteries will make
you a better programmer.  No one is familiar with a complicated project
before they dig into it.

It will take some time, but there will be a reward.

boelle, 2015-01-14:
@galexander1 agreed... 

this post was not to mean to start the beginning of a 10+ page long novel... it was just a statement that we are in need of coders that can or will help out... the fact that marlin is a big project kind of explains itself... that is why i dont get much involed other than keeping things organized and to an house cleaning now and then... 

if this begins another of those pages long talks that just end up in going in circles i will close an lock... the intention was again just to state the fact that we could use more coders, not discuss the code as such

PxT, 2015-01-14:
Well if you make things easy to newcomers by pointing out tasks that are somewhat simple (or, hell, even just not crazy complex) then eventually some of those people get interested and/or experienced in the project and contribute more and more.  Just saying "it's a complicated project, deal with it" is probably not an optimal way to find people to help you, especially when they are volunteers.

IMO.

boelle, 2015-01-14:
what do you mean by you ?

PxT, 2015-01-14:
I mean owners & collaborators.  Presumably those two of you who have commented above speak for the project, after all you (@boelle) are the one asking for more coders.  

I'm willing to help do what *I* can, but I'm also willing to point out that as a newcomer it's difficult to just dive in.  I don't have hours upon hours to dedicate to this, but if there are tasks that have a lower learning curve then I will help.  I just need help to find out what those tasks are, thus my suggestion to tag suitable bug reports.  If there are none, or it's not possible / easy to identify them then that's a reasonable response too.

boelle, 2015-01-14:
hehe... yes i did....

and also many times it has been mentioned that the code should be cleaned up... 

but i stronlgy advocate for bugs comes first...  then we can clean up the code... then add new features if they can run on the minimum hardware (644 based boards) or if the feature makes sense on boards they are intended for

and then for owners and collabs... we are 4 owners and 10 collabs... some can do code... some can do documentation... others can organize

the main problem is that most have a normal life and a job and marlin is a spare time proejct for the,... so to keep the ball rolling at a decent paste i think we need more people that can do code..

agreed that the code as it is now is big, you might even call it a mess.... but that is why i say bugs out first... then do a clean up... but with the limited number of people it will take a long time... the more people the faster we can move

closed 2015-03-17 by boelle