MarlinFirmware/Marlin #1342: Individual steps per mm setting for second extruder

Jo3yDaPh1sH, 2015-01-09:
Hello I would like the ability to change the extruder steps per mm independently when using a dual extruder setup. I have tried to change the define axis to 5 and just adding a fifth value to the steps per mm line with no success(And any other lines that have defines in the (0,0,0,0) format. I have added a bowden setup to my existing direct drive and are using different exturders so I am having an issue extruding the correct amount from each with only the one value. Thanks.

whosawhatsis, 2015-01-09:
This has been needed for some time, but the way multiple extrusion is implemented makes it difficult. I'd like to see each extruder have its own axis letter, and E be a virtual axis that is mapped to each of those by the toolchange codes, which would solve this problem, but there's probably an easier way to just modify the planner to look for different steps/unit values.

MagoKimbra, 2015-01-15:
I Jo3yDaPh1sH, if you want can you try my marlinkimbra. I have put 4 extruder and single step, feedrate, accelleration for extruder...

boelle, 2015-01-15:
why not E0..E1 etc etc... ?

whosawhatsis, 2015-01-15:
@boelle So you would extrude 1.2mm from the second extruder with G1 E11.2?

boelle, 2015-01-15:
i would put a space after E1 but othervise yes... but then if you cant put a space in there i see the problem

whosawhatsis, 2015-01-15:
The usual solution to this is to Map extruders starting from A, B, etc. A toolchange code should then change which of these an E code redirects to, but it would take some deep changes to Marlin to make this work.

barneycg, 2015-01-15:
if you did the kind of changes you propose @whosawhatsis would you then be able to have independent PID's for the different Hotends ?

grob6000, 2015-01-16:
@barneycg I just submitted a fix for the independent PID's the other day, adds an E parameter in M301 to do it. There's still only one E axis, so steps-per-mm is shared - see #1344).

Is there any reason I couldn't have a dig and do the same thing for axis-steps-per-mm? Where the parameter is called up it selects it from the array based on the active extruder?

whosawhatsis, 2015-01-16:
@barneycg separating the motor axes wouldn't separate the settings for the heaters. You'd have to do that separately.

MagoKimbra, 2015-01-16:
I have done so, in configuration.h:
#define DEFAULT_AXIS_STEPS_PER_UNIT 80,80,3200,625,625,625,625 {} // X, Y, Z, E0, E1, E2, E3 default steps per unit
Of course then I store in eeprom these values...
In planner.ccp when using the variable axis_steps_per_unit the use with the extruder active.
target [E_AXIS] = lround (e*axis_steps_per_unit [active_extruder + 3]);
The + 3 to increase the value of active extruder of the three previous axis.

barneycg, 2015-01-16:
@whosawhatsis I meant more that would adding the changes you proposed be a requirement for independent PID's but I @grob6000 has already done it :)

Jo3yDaPh1sH, 2015-01-18:
Thanks for looking into this issue, I have found a workaround for now, I have been using the new version of repiter hoist with the cura engine and it has a custom pre and post tool change g-code where I have inserted a M92 EXXX (XXX being Correct Step Per Min) before and after to switch it back and forth and it seems to be doing the trick. It would be nice to be able to set it in the firmware but this seems to be working for now, once again thanks for all the hard work.

Dim3nsioneer, 2015-01-18:
The number of axes before the extruders should not be hardcoded to 3 IMHO. Just to make Marlin future-ready...

@Jo3yDaPh1sH : Another workaround would be to use the extruder specific volume multiplier with M221 Tx Sy.

Jo3yDaPh1sH, 2015-01-18:
Thanks, I think I might give kimbra a go, thanks for all the help

thinkyhead, 2015-01-19:
To make it easier to add more axes, we can try moving the `X_AXIS`, `Y_AXIS`, etc. defines into an enum, rename them like `AXIS_X` (etc.), and add `AXIS_4`, `AXIS_5`, `AXIS_6`, followed by `AXIS_E0`, `AXIS_E1`, `AXIS_E2` (and `AXIS_E3` and beyond...). The last item in the enum would be `AXIS_COUNT`. By placing conditionals around these axes based on configuration options, they would all acquire the proper values. For radical configurations, savvy users could edit the enums directly.

Thoughts?

daid, 2015-01-21:
When you goto more axis, you might want to look into more standardized stuff. For example, when using Mach3 or LinuxCNC for 3D printing (yes, some people do this), the extruder is fitted to axis A and B.
So instead of:
T0
G1 X0 Y0 E10
T1
G92 E0
G1 X10 Y10 E10

You get:

G1 X0 Y0 A10
G1 X10 Y10 B10

However, increasing the number of axis does increase the load in the planner routines.

KriLL3, 2015-08-12:
Could someone detail MagoKimbra's work around for me? I don't understand his instructions.

Roxy-3DPrintBoard, 2015-11-04:
> Really odd that marlin still doesn't support multiple e step values. MarlinKrimba works nicely for me but doesn't work with octoprint so I can't use it.

I suspect the reason is most people use identical stepper motors and gear ratios for all of their extruders.  (That is just a guess)   But if you want them to be different, it shouldn't be hard to add.   Just try to do it in a clean manner so we can fold it back into the code base for others to use.

KriLL3, 2015-11-04:
Beyond my abilities, I need it because I have an i3 with a direct wade with gearing and a bowden extruder with no gearing, (bowden is faster, direct more reliable and can print flex better) my solution for now is a bit weird, I change the e-steps using the starting g-code in slic3r and have a profile for direct and one for bowden

Roxy-3DPrintBoard, 2015-11-04:
Yeah...  That sounds like a good work around.

closed 2016-05-04 by jbrazio