MarlinFirmware/Marlin #1346: Error to select two extruders (RAMBO, ULTIMAKER)

danielprint, 2015-01-10:
Hi, when I select two extruders get the following error:

temperature.cpp: In function ‘void manage_heater()’:
temperature.cpp:665:1: error: unable to find a register to spill in class ‘POINTER_REGS’
 }
 ^
temperature.cpp:665:1: error: this is the insn:
(insn 95 94 97 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 146 [ ivtmp.211 ])) [4 MEM[base: _110, offset: 0B]+0 S4 A8])) temperature.cpp:449 99 {*movsf}
     (expr_list:REG_INC (reg:HI 146 [ ivtmp.211 ])
        (nil)))
temperature.cpp:665: confused by earlier errors, bailing out

this is my fault somehow?
how could it be solved?

nophead, 2015-01-10:
It's actually a compiler bug.

galexander1, 2015-01-10:
Compiler bug, indeed!  The question is, is it a bug in a version of
avr-gcc that we feel we need to support?

@danielprint Where did you get your build environment, are you using the
arduino IDE?  What version?  Can you try a different version?

danielprint, 2015-01-10:
used the following version of arduino 1.0.5 +dfsg2-2 
in ubuntu 14.04 

in version 1.0.6 there was no error, compiled perfectly.

boelle, 2015-01-10:
1.0.6 is also the ones we would like people to use as its the latest non-beta

danielprint, 2015-01-10:
I think I can close this question, thank you all for your help, the problem was even the compiler.

closed 2015-01-10 by danielprint

boelle, 2015-01-10:
np

JamesT42, 2015-06-10:
So I got this error with the Arduino 1.6.4 IDE with my Ramps Board and 3 Extruders. Any ideas how to fix this except downgrade to a very old and crappy Arduino IDE?

KiteLab, 2015-06-10:
@JamesT42 
Would you please be so kind to open a new issue with a matching title.
As there is no BOARD_RAMPS_13_EEE until now, this will require a bit of work.
Arduino 1.6.4 is crapy on Windows 7 64 at least - but 1.6.3 will do the job better.
The messoge from @boelle is from January - at this time there was no 1.6.x

JamesT42, 2015-06-10:
Ok, it works with the current development version. So it should be fixed with the next release version. Its working now with 3 Extruders. I build a custom board and modified the pins.h.

Wackerbarth, 2015-06-26:
I see that the Arduino folks have made some changes to the Windows toolset. Can someone try the latest nightly build of the IDE and report on the nature of "crappy on Windows 7 64"?