MarlinFirmware/Marlin #1352: Makefile vs Arduino IDE build size differs.

thawkins, 2015-01-13:
Setup 

1. Clone repo into fresh directory 

2. set MOTHERBOARD to BOARD_RAMPS_13_EFB (33) 

compile with arduino IDE version 1.0.5 and get program size of 49390 

then edit Makefile for standalone operation 

1. Set HARDWARE_MOTHERBOARD to 33
2. Set ARDUINO_INSTALL_DIR to /usr/share/arduino (for ubuntu linux or fedora linux)
3. set BUILD_DIR  to ~/tmp  which is a world writable tmp directory in my home directory 
4. run make 

thawkins@thawkins-N550JK:~/Public/Firmware/Marlin/Marlin$ make
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_analog.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_digital.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_pulse.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_shift.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/WInterrupts.c
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/WMath.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/WString.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/Print.cpp
  CXX   Marlin_main.cpp
  CXX   MarlinSerial.cpp
  CXX   Sd2Card.cpp
  CXX   SdBaseFile.cpp
  CXX   SdFatUtil.cpp
  CXX   SdFile.cpp
  CXX   SdVolume.cpp
  CXX   motion_control.cpp
  CXX   planner.cpp
  CXX   stepper.cpp
  CXX   temperature.cpp
  CXX   cardreader.cpp
  CXX   ConfigurationStore.cpp
  CXX   watchdog.cpp
  CXX   /usr/share/arduino/libraries/SPI/SPI.cpp
  CXX   Servo.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/Tone.cpp
  CXX   ultralcd.cpp
  CXX   digipot_mcp4451.cpp
  CXX   vector_3.cpp
  CXX   qr_solve.cpp
  CXX   /usr/share/arduino/libraries/LiquidCrystal/LiquidCrystal.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/main.cpp
  CXX   /home/thawkins/tmp/Marlin.elf
  COPY  /home/thawkins/tmp/Marlin.hex

AVR Memory Usage
----------------
Device: atmega2560

Program:   50012 bytes (19.1% Full)
(.text + .data + .bootloader)

Data:       2761 bytes (33.7% Full)
(.data + .bss + .noinit)

   text	   data	    bss	    dec	    hex	filename
  49476	    536	   2225	  52237	   cc0d	/home/thawkins/tmp/Marlin.elf

thawkins@thawkins-N550JK:~/Public/Firmware/Marlin/Marlin$ 

Note that the program size is different, 49476

OS Ubuntu 14.10 64bit.

nothinman, 2015-01-20:
@thawkins you forgot to mention avr-gcc versions for both cases. I'm also pretty sure arduino doesn't use the Makefile, so with different compiler and/or different optimisations this is exactly what you would see.
But why is this a problem?

thawkins, 2015-01-20:
Consistency, i will check on the versions, i suspect they are the same as
the makefile is suposed to (and is in this case) configured to use the avr
toolchain embedded in the arduino 1.0.5 ide. So both methods (makefile and
ide) are both using the same gcc etc, hence they should produce the same
results.

I tend to use the makefile as i use the jetbrains cLion ide to edit and
maintain the code and have a tool setup that allows calling of the makefile
when i need to compile.  I have arduino ide 1.0.5 installed, but i dont use
it as the ide is too limited.

nothinman, 2015-01-20:
I've noticed that a while ago; the difference between Makefile and Arduino at the time was approximately 9kB. And I've noticed, because my avr had 64kB of flash and the code didn't fit.
After playing with linker optimisation flags my hex fit, but I can't remember whether the sizes were exactly the same after compilation. Probably not, given that I've used two separate 'installations' of avr-gcc, so the versions could differ.
But this is where I would look... check Arduino console during compilation and you'll see all the flags in the command line; then compare against output generated via make.

chertykov, 2015-01-24:
1. Makefile have extra options: `-fno-use-cxa-atexit -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums`
  *  *IMHO: better to remove because resulting code may have differences*
2. Arduino have extra option `-fno-exceptions` 
  * *Better to add it to Makefile*
* Arduino have extra linker option --relax (optimize long jumps at link time) 
  * *Better to add it to Makefile*
* Makefile compile and link LiquidCrystal.cpp. Arduino compile and link LiquidCrystalRus.cpp
  * *I don't know. Probably `LiquidCrystalRus.cpp` is a right thing.*
* Arduino build a library `core.a` before linking.
  * *Don't care*
* Arduino compile and link more files:
  * Default Configuration.h for arduino
  * HARDWARE_MOTHERBOARD=7 for Makefile
  * applet/* - Makefile compiled
  * /tmp/build175383027878887211.tmp/* - Arduino
  * compiled with equal avr-gcc options
  * different files marked by `*`
```
   text	   data	    bss	    dec	    hex	filename
     30	      0	      0	     30	     1e	applet/main.o
     30	      0	      0	     30	     1e	/tmp/build175383027878887211.tmp/main.cpp.o
*     56	      0	      0	     56	     38	/tmp/build175383027878887211.tmp/new.cpp.o
*    116	      0	      1	    117	     75	applet/SPI.o
    260	      0	      0	    260	    104	applet/wiring_shift.o
    260	      0	      0	    260	    104	/tmp/build175383027878887211.tmp/wiring_shift.c.o
    306	      0	      0	    306	    132	applet/WMath.o
    306	      0	      0	    306	    132	/tmp/build175383027878887211.tmp/WMath.cpp.o
    312	      0	      0	    312	    138	applet/wiring_pulse.o
    312	      0	      0	    312	    138	/tmp/build175383027878887211.tmp/wiring_pulse.c.o
*    450	      0	      0	    450	    1c2	/tmp/build175383027878887211.tmp/realloc.c.o
    512	      0	      6	    518	    206	/tmp/build175383027878887211.tmp/IPAddress.cpp.o
    558	      1	      0	    559	    22f	applet/wiring_analog.o
    558	      1	      0	    559	    22f	/tmp/build175383027878887211.tmp/wiring_analog.c.o
*    618	      6	      0	    624	    270	/tmp/build175383027878887211.tmp/malloc.c.o
    642	      0	      9	    651	    28b	applet/wiring.o
    642	      0	      9	    651	    28b	/tmp/build175383027878887211.tmp/wiring.c.o
    844	      0	      0	    844	    34c	applet/wiring_digital.o
    844	      0	      0	    844	    34c	/tmp/build175383027878887211.tmp/wiring_digital.c.o
   1120	      0	     16	   1136	    470	applet/WInterrupts.o
   1120	      0	     16	   1136	    470	/tmp/build175383027878887211.tmp/WInterrupts.c.o
*   1414	      1	      0	   1415	    587	/tmp/build175383027878887211.tmp/Stream.cpp.o
   1704	      2	    133	   1839	    72f	applet/MarlinSerial.o
   1704	      2	    133	   1839	    72f	/tmp/build175383027878887211.tmp/MarlinSerial.cpp.o
   1865	      1	     42	   1908	    774	applet/Tone.o
   1865	      1	     42	   1908	    774	/tmp/build175383027878887211.tmp/Tone.cpp.o
   1872	     14	      0	   1886	    75e	applet/Print.o
   1872	     14	      0	   1886	    75e	/tmp/build175383027878887211.tmp/Print.cpp.o
   1986	      0	      0	   1986	    7c2	applet/motion_control.o
   1986	      0	      0	   1986	    7c2	/tmp/build175383027878887211.tmp/motion_control.cpp.o
   2630	      0	      0	   2630	    a46	applet/LiquidCrystalRus.o
   2630	      0	      0	   2630	    a46	/tmp/build175383027878887211.tmp/LiquidCrystalRus.cpp.o
   3094	      1	      0	   3095	    c17	applet/ConfigurationStore.o
   3094	      1	      0	   3095	    c17	/tmp/build175383027878887211.tmp/ConfigurationStore.cpp.o
*   3186	      0	    680	   3866	    f1a	/tmp/build175383027878887211.tmp/HardwareSerial.cpp.o
   4800	      1	      1	   4802	   12c2	applet/WString.o
   4800	      1	      1	   4802	   12c2	/tmp/build175383027878887211.tmp/WString.cpp.o
   5397	     83	     79	   5559	   15b7	applet/temperature.o
   5397	     83	     79	   5559	   15b7	/tmp/build175383027878887211.tmp/temperature.cpp.o
   7140	     26	     83	   7249	   1c51	applet/stepper.o
   7140	     26	     83	   7249	   1c51	/tmp/build175383027878887211.tmp/stepper.cpp.o
   9494	     20	   1374	  10888	   2a88	applet/planner.o
   9494	     20	   1374	  10888	   2a88	/tmp/build175383027878887211.tmp/planner.cpp.o
  16683	    294	    526	  17503	   445f	applet/Marlin_main.o
  16683	    294	    526	  17503	   445f	/tmp/build175383027878887211.tmp/Marlin_main.cpp.o
```

thawkins, 2015-01-24:
>     3. Arduino have extra linker option --relax (optimize long jumps at
>     4. Makefile compile and link LiquidCrystal.cpp. Arduino compile and
>     5. Arduino build a library core.a before linking.
>     6. Arduino compile and link more files:

I will try these changes out tomorrow, i have been a bit busy trying to get
my printer build completed and tested. Thanks for the efforts of looking at
this. There is a small change i need to commit anyway tomorrow, an
unessacary extra option in my original commit.

thinkyhead, 2015-02-24:
I'd be happy if I could get the Makefile to work at all on OS X today. I keep getting the dreaded

    make: *** No rule to make target `applet/wiring.o', needed by `applet/Marlin.elf'.  Stop.

I'm messing around with moving elements of Marlin into a `libraries` subfolder, a hackish way to clean up the sketch folder, but… I'm a hacker.

boelle, 2015-03-11:
@thawkins is this still an issue?

thawkins, 2015-03-11:
Create a seperate directory in your home somewhere and set the path in the
makefile that is currently set to applet, to the full path, then it works

thawkins, 2015-03-11:
I got it very close, but not 100%, its probaly down to things like
timestamps etc now.

thawkins, 2015-03-11:
Close it

boelle, 2015-03-11:
ok, closing :-D

closed 2015-03-11 by boelle