MarlinFirmware/Marlin #1355: Build fails if BOARD_MELZI_1284 and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

roglio, 2015-01-13:
Hi, if I both set the board and additional gfx display the compiler fails to resolve some pin definitions.

I fixed the problem replacing in pins.h at row 1276 the whole #ifdef ULTRA_LCD definition with this code:

#ifdef ULTRA_LCD
   #ifdef NEWPANEL
     //we have no buzzer installed
     #define BEEPER -1
     //LCD Pins
     #ifdef DOGLCD
	   #ifdef U8GLIB_ST7920 //SPI GLCD 12864 ST7920 ( like [www.digole.com] ) For Melzi V2.0
	  	 #define LCD_PINS_RS 30 //CS chip select /SS chip slave select
	  	 #define LCD_PINS_ENABLE 29 //SID (MOSI)
	  	 #define LCD_PINS_D4 17 //SCK (CLK) clock
	  	 #define BEEPER 27 // Pin 27 is take by LED_Pin, but Melzi LED do nothing with Marlin and I take this pin for BEEPER.... See here > [github.com] , If you want use this pin with Gcode M42 instead BEEPER
	   #else
	  	 // Pins for DOGM SPI LCD Support
	  	 #define DOGLCD_A0 30
	  	 #define DOGLCD_CS 29
	  	 #define LCD_CONTRAST 1
	   #endif
       // Uncomment screen orientation
       #define LCD_SCREEN_ROT_0
         // #define LCD_SCREEN_ROT_90
         // #define LCD_SCREEN_ROT_180
         // #define LCD_SCREEN_ROT_270
       #else // standard Hitachi LCD controller
       #define LCD_PINS_RS        4
       #define LCD_PINS_ENABLE    17
       #define LCD_PINS_D4        30
       #define LCD_PINS_D5        29
       #define LCD_PINS_D6        28
       #define LCD_PINS_D7        27
     #endif
     //The encoder and click button
     #define BTN_EN1 11
     #define BTN_EN2 10
     #ifdef LCD_I2C_PANELOLU2
       #ifdef MELZI
         #define BTN_ENC 29 //the click switch
         #define LCD_SDSS 30 //to use the SD card reader on the Panelolu2 rather than the melzi board
       #else
         #define BTN_ENC 30 //the click switch
       #endif
     #else
       #define BTN_ENC 16  //the click switch
       #define LCD_SDSS 28 //to use the SD card reader on the smart controller rather than the melzi board
     #endif //Panelolu2
     //not connected to a pin
     #define SDCARDDETECT -1

   #endif //Newpanel
 #endif //Ultipanel

Please advice.

boelle, 2015-01-13:
advice is that we are doing bugfixing... you saw that at the front page right?

and these displays do use the i2c bus if i'm correct and we have some issues there...

roglio, 2015-01-13:
Hi, thank you for your prompt reply! Yes these displays do use i2c bus, but when I simply restore some #define as shown in my snippet, it simply works like a charm.

boelle, 2015-01-13:
ok... did not note that... what defines excatly makes it work?

roglio, 2015-01-13:
those define under #ifdef U8GLIB_ST7920

boelle, 2015-01-13:
i could be wrong here, but i think that there is a genral issue... mostly
I2C is not used, and when its used there are only one device connected most
of the time

in those cases i think you dont need the chip select...

trouble starts when you have more devices... another device could be a
thermocouple instead of thermistors...

now the trouble is if the different part of the codes does not take this in
to account....

if i'm right then every module that uses i2c should always use chip select
no matter how many devices....

https://github.com/MarlinFirmware/Marlin/milestones/Bug%20Fixing%20Round%203

in that one you can see we have some similar issues....

boelle, 2015-01-13:
hmmm... I2C does not have chip select.... but SPI bus does have...

are you realy sure its not SPI bus?

roglio, 2015-01-13:
Yes, I'm sure: I2C is used to connect the gfx display (sdl and sda). SPI is used to connect an external SD card.

This is the build's output:

/ultralcd_st7920_u8glib_rrd.h: In function 'void ST7920_SWSPI_SND_8BIT(uint8_t)':
/ultralcd_st7920_u8glib_rrd.h: In function 'uint8_t u8g_dev_rrd_st7920_128x64_fn(u8g_t*, u8g_dev_t*, uint8_t, void*)':

ultralcd_st7920_u8glib_rrd.h:29,30,32: error: 'DIOLCD_PINS_D4_RPORT' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:29,30,32: error: 'DIOLCD_PINS_D4_WPORT' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:29,30,32: error: 'DIOLCD_PINS_D4_PIN' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:29,30,32: error: 'DIOLCD_PINS_ENABLE_RPORT' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:29,30,32: error: 'DIOLCD_PINS_ENABLE_WPORT' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:29,30,32: error: 'DIOLCD_PINS_ENABLE_PIN' was not declared in this scope

If I add these declarations
#define LCD_PINS_RS 30 //CS chip select /SS chip slave select
#define LCD_PINS_ENABLE 29 //SID (MOSI)
#define LCD_PINS_D4 17 //SCK (CLK) clock

Then, the issue seems related to the SDSUPPORT flag. BTW if I enable it, the situation is still the same: build stops with the same errors.

roglio, 2015-01-13:
NOPE! I was wrong: the lcd panel is SPI. BTW, enabling DOGLCD, SDSUPPORT and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER the compilation fails...

boelle, 2015-01-15:
hmmm... i dont think you nead to enable DOGLCD if you have the Full graph display.... or said in other words check that you dont enable 2 display and only have one.... and again we have some issues for SPI bus....

roglio, 2015-01-15:
Hi! Yes I do not have two lcd display enabled at once. My configuration.h foresee only the full gfx controller enabled and if I do not fix the pins.h, the compilation fails.
With my patch, I was able to load the firmware in my melzi v2 and print using the gfx display.
Then, if there are SPI issues I'm not affected. Also the SD card reader in my geeetech display works!
I did a successful printing from my sd card.

Take into consideration the possibility to include this fix in your release: it works! And is related only to MELZI boards.

This is the pinout used to connect a melzi to the broadly available geeetech reprap full gfx display:
AUX. 1 -> EXP1. 5
AUX. 2 -> EXP1. 4
AUX. 3 -> EXP1. 2
AUX. 4 -> EXP1. 3
AUX. 5 -> EXP2. 5
AUX. 6 -> EXP2. 4
AUX. 7 -> EXP2. 3
AUX. 8 -> EXP1. 1
AUX. 9 -> EXP1.10
AUX.10 -> EXP1. 9

ISP. 1 -> EXP2. 1
ISP. 2
ISP. 3 -> EXP2. 2
ISP. 4 -> EXP2. 6
ISP. 5 -> EXP2. 8
ISP. 6
[image]

boelle, 2015-01-15:
if you have a fix why not submit a pull request and have the honour of fixing an issue?

roglio, 2015-01-15:
Because I never did and I don't know how to do it...

boelle, 2015-01-15:
you wrote: "With my patch, I was able to load the firmware in my melzi v2 and print using the gfx display." and "Take into consideration the possibility to include this fix in your release"

?

roglio, 2015-01-15:
Yes, I did. But I never used a versioning system like github to submit patches! This is my problem.

PxT, 2015-01-16:
@roglio One way to help would be to paste your patch into a gist (https://gist.github.com) and then provide the link.  Then someone with more Github experience can create a pull request if appropriate.

thinkyhead, 2015-01-19:
Surely, paste your code as a gist and one of us will roll a patch.

roglio, 2015-01-19:
Hi! I just created my first pull request!!! :D

https://github.com/MarlinFirmware/Marlin/pull/1386

roglio, 2015-01-19:
I was fighting with some configuration management concept I do not master yet. Btw, The pull request is clean and build fine.

roglio, 2015-01-19:
I suppose that this issue could be closed because deprecated.

thinkyhead, 2015-01-20:
Just looking over your changes to make sure they're clean and don't affect other configurations. I will post comments on #1386 if I have any concerns.

closed 2015-01-20 by thinkyhead

roglio, 2015-01-20:
Yes! Of course! I'm a noob and still learning how to collaborate. I did a mistake pushing a configuration.h change with my own settings. I reverted it back to the main repository defaults.
I've already replied to your question on pin.h changes!