MarlinFirmware/Marlin #1360: Offsets for multiple hotend

clefranc, 2015-01-13:
Hi all,
I apologize in advance if this is not the place to ask question about configurations.

I'm using a delta with 3 hotends on the effector, positionned at the apexes of a centered equilateral triangle. I copied the delta configuration files from the example_configurations folder.

I need to center the first nozzle on the building plate (round shaped), so I've set the MANUAL_Y_HOME_POS to -10. The homing proceeds normally, but if I move the nozzle towards the bed, the Z tower's carriage bump on the endstop, i.e. goes up, violating the logic of the endstop.

Did I miss a basic configuration?

BTW, without offsets, the firmware work great, very impressed by its retraction and volumetric extrusion.

P.-S. Sorry for my bad English, I'm French!

thinkyhead, 2015-01-19:
I'd be curious to see this behavior in action, and see whether this is the appropriate option for multiple extruders. A Delta should always try to home to the center of the bed. It sounds like what is needed is an additional offset for the first extruder. Currently Marlin assumes that Extruder 0 will always be the reference point, so it always has [0,0,0] as its offset. But for your machine you need to be able to give Extruder 0 an offset too.

boelle, 2015-03-11:
is this still an issue with latest copy and latest non-beta IDE?

boelle, 2015-03-11:
that was an half-arsed attempt to wake this one up again

please let us know if you are still having issues

an idea for doing offset on a delta... if 3 extruders... arrange them in a triangle pattern and let the point that go towards say stepper X be extruder one... and then work offset from there

boelle, 2015-03-17:
closing as there have been no response from OP

clefranc, 2015-03-20:
Just uploaded the latest dev, tried the same config

#define MANUAL_Y_HOME_POS -10

After succesful homing, moved the nozzle toward the bed (Z in host) and the Z carriage (or C or 3 or whatever it is called in a delta) goes up. Big problem here, this can potentially damage the printer.

For reference, I was using Repetier firmware before switching to Marlin. What Repetier does is move the other carriage down (not up), so the E0 is moved to the center of the bed. This is done at the G28 level.

thinkyhead, 2015-03-21:
Interesting. But I'm not entirely sure that the MANUAL_Y_HOME_POS is the right parameter for a delta with its nozzle offset. I think we need to look at this closer. But, were you getting some kind of better behavior in an earlier version of Marlin? If so, we should look at the older code and compare it to the new.

clefranc, 2015-04-08:
Somewhat related https://github.com/MarlinFirmware/Marlin/issues/1856

boelle, 2015-04-25:
just on the hangout now.... we talked about arrange the nozzle's in a triangle and define one of them as number 0 and then offset the other from that one.... would that be ideal?

clefranc, 2015-04-27:
@boelle 
Would it not be simpler to define an offset for each nozzle, a set of XY arrays based on the number of EXTRUDERS? There is already existing EXTRUDER_OFFSET_X EXTRUDER_OFFSET_Y, why not apply them correctly to delta printer.

If the result is the same, I accept the choice of implementation chosen at the discretion of the coder. It's just that I don't like my carriages trying to destroy my endstops. If an endstop is triggered, then the motor shoud stop an only be allowed to go in the other direction.

ProjectEnder, 2015-04-27:
I Agree i think it would be better to apply an off set to each nozzle. 
I figure it would just take a look at the code (only basic knowledge) and i noticed something that might be wrong or i might be wrong. 
the offset is stored and created here @ line 291 in Marlin_main.cpp

  float extruder_offset[][EXTRUDERS] = {
    EXTRUDER_OFFSET_X,
    EXTRUDER_OFFSET_Y
    #ifdef DUAL_X_CARRIAGE
      , { 0 } // supports offsets in XYZ plane
    #endif
  };

and then used and set in place like Lines 962, 986, 993 in this format
extruder_offset[X_AXIS][target_extruder]

i might be wrong but should it not be setup like this

  float extruder_offset[EXTRUDERS] = {
    EXTRUDER_OFFSET_X,
    EXTRUDER_OFFSET_Y
    #ifdef DUAL_X_CARRIAGE
      , { 0 } // supports offsets in XYZ plane
    #endif
  };

and then called for by 

extruder_offset[target_extruder][X_AXIS]

to get the offset of the extruder, this should then make it easy to have have offset for all extruders as it should make possible to skip the need to check if your not on Extruder 0 and just add the offset depending on the extruder your currently using.

thinkyhead, 2015-04-28:
Side note: we'll be using "nozzle" instead of "extruder" for naming things that pertain to the nozzle going forward.

With the current code, for deltas, if you have no nozzle in the center, then there's no way to set up offsets that are compatible with homing, because if the first nozzle is not at (0,0) the printer will try to home off-center, obviously not possible with deltas. So, we need to de-couple homing from nozzle offsets and allow the reference points for nozzle offsets to be the center of the effector on deltas. This also means de-coupling the reference point from the tool-change, or add a "no tool" option that will cause deltas with no centered nozzles to nevertheless move on-center.

boelle, 2015-05-31:
any news on this one?

thinkyhead, 2015-06-08:
The code is moving towards more generalization, so the formal "initiative" for all nozzles —including 0— to have their own offsets from the "carriage/effector position" will perhaps need to wait until after the part where we generalize nozzles more. This is something that a coder who has more than one nozzle —and preferably a delta to experiment with— should be working on, and then if the change isn't very radical to support it, we might get it in earlier. Unfortunately my delta is not yet constructed.

a4jp-com, 2015-09-02:
Did you finish building your delta printer @thinkyhead ?

a4jp-com, 2015-09-03:
@clefranc have any of the updates fixed this problem yet?

clefranc, 2015-09-03:
@a4jp-com No, not that I am aware.

a4jp-com, 2015-09-03:
Okay. Sorry to hear that.We will probably just have to wait for @thinkyhead to reply then.

thinkyhead, 2015-09-04:
@a4jp-com I haven't finished my Deltabot yet :( but I hope to get that done asap! I have it at a friend's workshop, so we just have to grab some scraps, fire up some tools, and build away for a week or so... Meanwhile I'm just asking delta owners to keep pushing Marlin to find those boundaries. As for this issue, it still has to be implemented, perhaps in a simper way by adding a flag just for this situation. I can look into it myself, see what will be required, but it will have to wait until after we do the 1.1.0-RC1 release.

a4jp-com, 2015-09-05:
Can't wait for the release candidate ^^

clefranc, 2015-09-05:
@thinkyhead Correct me if I'm wrong, there is an offset for nozzle in Configuration.h: `EXTRUDER_OFFSET_X` and `EXTRUDER_OFFSET_Y`. There is also a note: `The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).`

There is mention of the `EXTRUDER_OFFSET_X` for `DUAL_X_CARRIAGE`. You also find another offset for the duplication mode: `DEFAULT_DUPLICATION_X_OFFSET`.

About the 0,0 note: I think it was originally introduced for cartesian printer, there is no such offset for hotend 0. The offset for hotend 0 is mechanically adjusted with endstops, ie you just make sure the nozzle is at location 0,0 (relative to bed) when homing.

For deltas, if you have more than 1 extruder, it is very unlikely you have hotend 0 at 0,0 (it's possible). But the second or third nozzle are definitively not at the same location (unless you're using a Diamond like one).

I propose the `EXTRUDER_OFFSET_X` and `EXTRUDER_OFFSET_Y` used for offsetting delta nozzles, cartesian user can always follow the note with hotend 0 at 0,0.

clefranc, 2015-09-05:
Here's the current behavior for `EXTRUDER_OFFSET_X` and `EXTRUDER_OFFSET_Y` on my delta.

Nozzle configuration:

```
        Tower Z
           O

       _________
      /         \
     |  2     1  |
     |           |
     |     0     |
      \_________/

   O               O
Tower X         Tower Y
```
***Note: I'll use hotend 0 and 1 in the following procedure.***

The setting for hotend offset is:
```cpp
#define EXTRUDER_OFFSET_X {0.0, 9.0933}
#define EXTRUDER_OFFSET_Y {-10.50, 5.25}
```
I home the carriages with `G28`, then issue a `M114` (Get Current Position):
`X:0.00 Y:0.00 Z:387.00 E:0.00 Count X: 742.12 Y:738.67 Z:739.80`

Then I issue a `T0` (Select Tool), no movement, followed by `M114`:
`X:0.00 Y:0.00 Z:387.00 E:0.00 Count X: 742.12 Y:738.67 Z:739.80`

I now send a `T1`, no movement, `M114`:
`X:9.09 Y:15.75 Z:387.00 E:0.00 Count X: 729.96 Y:738.19 Z:750.73`

The `T1` have now the new offset coordinates, but it added (or substracted) the `T0` offset as well.

I'm still at the top of the delta (387), if I send the current nozzle `T1` to 0,0 with `G1 X0 Y0`, the Z carriage go down a little, the Y almost not move, but the X carriage go up pass the endstop and collide with the top of the printer. This is the problem with offsets on a delta, you must calculate the new position of the carriages, so they doesn't collide or in other term go beyond the endstops.

Is this helping or is a useless demonstration?

thinkyhead, 2015-09-07:
@clefranc You demonstrate the precise problem with the current code and the concept of multiple hotends on a delta. The first hotend is assumed to be the reference point for all multiple hotend arrangements. It must be 0,0 for the current code, insofar as it isn't just ignored. Code that deals with moving hotend 0 is currently free to ignore `extruder_offset[X_AXIS][0]` and `extruder_offset[Y_AXIS][0]`. That includes homing code, and any other code that is moving "the carriage" as opposed to "the tool." So, one of the first things would be to figure out where non-zero `extruder_offset[*][0]` might have an adverse effect...

For Deltas or other robots to have a carriage center independent from a tool with the current code-base, the "simplest" thing might be to make a dummy tool (at "index" -1) which corresponds to the center of the effector. This tool would have to be selected before homing would be allowed. (Or, homing would cause it to become selected.) And the delta carriages would have to be lowered enough to give clearance before you could select any of the offset-from-center extruders.

closed 2016-04-23 by jbrazio