MarlinFirmware/Marlin #1362: Steppers gets stuck running continuously

clefranc, 2015-01-13:
This is well explained and documented here:

https://github.com/RichCattell/Marlin/issues/7in

I've the same issue with the 1.0.2 release.

boelle, 2015-01-15:
Fixed by PR #1357

clefranc, 2015-03-20:
No, it is not.

thinkyhead, 2015-03-21:
@clefranc  If moving into a negative coordinate is causing the stepper motor to keep running continuously, then to me that indicates there's possibly a comparison between signed and unsigned numbers someplace and it's getting the result backwards, or possibly we're getting some kind of threshold problem where suddenly values go to 0 or infinity.

Would you mind uncommenting the code that prints the delta coordinates and capture that output, and either paste it here or summarize what you're seeing in the numbers?

clefranc, 2015-03-26:
@thinkyhead Where is the code I must uncomment?

In the mean time, I've added some code in planner.cpp to print the stepper position, here is the result: 

23:42:37.015 : X=40493 Y=12906 Z=35302
23:42:37.016 : X=40493 Y=12906 Z=35398
23:42:37.018 : X=40493 Y=12906 Z=35398
23:42:37.744 : X=40442 Y=12906 Z=35398
23:42:37.745 : X=40442 Y=12295 Z=35398
23:42:37.746 : X=40442 Y=12295 Z=35494
23:42:37.748 : X=40442 Y=12295 Z=35494
23:42:38.559 : X=40390 Y=12295 Z=35494
23:42:38.560 : X=40390 Y=11429 Z=35494
23:42:38.561 : X=40390 Y=11429 Z=35588
23:42:38.563 : X=40390 Y=11429 Z=35588
23:42:39.400 : X=40338 Y=11429 Z=35588
23:42:39.402 : X=40338 Y=-2147483648 Z=35588
23:42:39.404 : X=40338 Y=-2147483648 Z=35682
23:42:39.406 : X=40338 Y=-2147483648 Z=35682

I've just uploaded the latest dev, and the behavior is not the same as previous, when moving the nozzle to a corner, instead of going down crazy, the Y carriage is now going up, and don't stop at the endstop, I must hit the panic button.

thinkyhead, 2015-03-26:
@clefranc Oops, I was referring to the serial output lines at the end of `prepare_move`.

Those are some weird numbers you've printed. What does your print code look like?

ProjectEnder, 2015-04-02:
@clefranc I was getting the same problems at first with my Printer, but it turned out that my configuration was wrong. I will download the latest version a make sure my system is still moving fine.

thinkyhead, 2015-04-04:
It's possible with some configurations to move past the limits of the delta transformation. The idea is that if numbers go negative at all (usually where one of the carriages tries to go far higher than the others) then you just limit the carriage to the highest reasonable point (not too far from the nozzle Z position). The only part I'm stuck on is figuring out what that limit should be, based on the length of the delta arms.

boelle, 2015-04-25:
@clefranc have you tried to download the lastest copy and checked your config?

clefranc, 2015-04-29:
@boelle No, my delta is currently on a massive upgrade, perhaps next week, I hope...

thinkyhead, 2015-04-30:
Related to #928 where I posted this concept code – certainly not correct, but the idea is to limit the physical position of the arms so they can't be far from the current Z position. The calculations need more work.

    void calculate_delta(float cartesian[3]) {
      // Proposed in #928 to limit these values to
      // Z plus the vertical effector offset
      // Impossible positions will just be in the wrong place
      float cx = cartesian[X_AXIS], cy = cartesian[Y_AXIS], cz = cartesian[Z_AXIS],
            dx = sqrt(delta_diagonal_rod_2 - sq(delta_tower1_x - cx) - sq(delta_tower1_y - cy)),
            dy = sqrt(delta_diagonal_rod_2 - sq(delta_tower2_x - cx) - sq(delta_tower2_y - cy)),
            dz = sqrt(delta_diagonal_rod_2 - sq(delta_tower3_x - cx) - sq(delta_tower3_y - cy));

      dx = constrain(dx, 0, DELTA_DIAGONAL_ROD / 1.414);
      dy = constrain(dy, 0, DELTA_DIAGONAL_ROD / 1.414);
      dz = constrain(dz, 0, DELTA_DIAGONAL_ROD / 1.414);

      delta[X_AXIS] = dx + cz;
      delta[Y_AXIS] = dy + cz;
      delta[Z_AXIS] = dz + cz;
    }

clefranc, 2015-04-30:
I'll test it right now...

clefranc, 2015-04-30:
@thinkyhead  Same result, do you want a video to see the behavior?

thinkyhead, 2015-05-01:
@clefranc Yes, I would like to see a video, that will be helpful. It would also help to have a log of the effector and tower position values as they approach these limits, with comparisons to the delta dimensions, current Z, etc. so that it will be more "obvious" where the limit is.

boelle, 2015-05-31:
any news here?

clefranc, 2015-06-10:
@boelle @thinkyhead I'm away from my printer for a couple weeks.

thinkyhead, 2015-06-12:
As you can… I have a delta under construction that I could perhaps test with, leaving off all the hardware except the belts, carriages, and endstops. But I may not have that put together very soon either…

boelle, 2015-06-27:
Will close this as its hard to repoduce, no one in tonights hangout have had the issue

we can always reopen

closed 2015-06-27 by boelle