MarlinFirmware/Marlin #1371: X axe homing - mirror printing -

starzleo, 2015-01-16:
hi, 
I do not find the solution , I have a RepRap rework I reversed the endstop (left) but I mirrored print?
my config with mirror printing (endstop on ramps MIN)

#define INVERT_X_DIR false   
  #define INVERT_Y_DIR true   
  #define INVERT_Z_DIR false     
  #define INVERT_E0_DIR true  
  #define INVERT_E1_DIR false  
  #define INVERT_E2_DIR false   
  
  // ENDSTOP SETTINGS:
  // Sets direction of endstops when homing; 1=MAX, -1=MIN
  #define X_HOME_DIR -1
  #define Y_HOME_DIR -1
  #define Z_HOME_DIR -1

I tried this with and without endstop RAMPS MIN and MAX but when I click the x HOME Nothing happens ?

#define INVERT_X_DIR false   
  #define INVERT_Y_DIR true   
  #define INVERT_Z_DIR false     
  #define INVERT_E0_DIR true  
  #define INVERT_E1_DIR false  
  #define INVERT_E2_DIR false   
  
  // ENDSTOP SETTINGS:
  // Sets direction of endstops when homing; 1=MAX, -1=MIN
  #define X_HOME_DIR 1
  #define Y_HOME_DIR -1
  #define Z_HOME_DIR -1

nicolas-rambaud, 2015-01-16:
Hello
I had thé sme behvior once.
The solution i found was tout keep thé défaut direction settings ans only
invert the
motor connecter when it was turning the opposite way

It seems that if you start playing with direction setting you will likely
configuré the thing with some mirror effect

starzleo, 2015-01-17:
Hi, just inverse the motor connector ?

nicolas-rambaud, 2015-01-17:
Try reverting jour configuration ans only revert jour motor connecter fort
thé directions

nophead, 2015-01-18:
The important thing is lower X values are on the left of the bed and lower Y values are at the front of the bed (note that if it is moving bed machine the bed moves the opposite way, i.e. it moves to the back to get the nozzle at the front).

If you have one axis working in the wrong direction you get mirrored prints. If both are the opposite to above you get a print rotated 180 degrees, which is just the same as looking at the machine from the back.

If your end stop is triggered at the higher end of the axis you need to configure it as a MAX end stop (i.e. use the MAX pin) and change the homing direction to 1. Otherwise it is a MIN endstop and the homing direction is -1.

If the motor turns the wrong way, i.e. moves away from the endstop when homing then, assuming the homing direction is correct, you can invert the direction signal to the driver by setting INVERT_X_DIR to true. That does exactly the same thing as reversing connector.

closed 2015-02-20 by thinkyhead