MarlinFirmware/Marlin #1372: Error compiling.

linjinc, 2015-01-16:
ConfigurationStore.cpp: In function 'void Config_ResetDefault()':
ConfigurationStore.cpp:426:31: error: expected initializer before 'MARLIN_PROGMEM'
     const static float tmp1[] MARLIN_PROGMEM =DEFAULT_AXIS_STEPS_PER_UNIT;
                               ^
ConfigurationStore.cpp:427:31: error: expected initializer before 'MARLIN_PROGMEM'
     const static float tmp2[] MARLIN_PROGMEM =DEFAULT_MAX_FEEDRATE;
                               ^
ConfigurationStore.cpp:428:30: error: expected initializer before 'MARLIN_PROGMEM'
     const static long tmp3[] MARLIN_PROGMEM =DEFAULT_MAX_ACCELERATION;
                              ^
In file included from Marlin.h:16:0,
                 from ConfigurationStore.cpp:1:
ConfigurationStore.cpp:431:48: error: 'tmp1' was not declared in this scope
         axis_steps_per_unit[i]=pgm_read_float(&tmp1[i]);  
                                                ^
ConfigurationStore.cpp:432:41: error: 'tmp2' was not declared in this scope
         max_feedrate[i]=pgm_read_float(&tmp2[i]);  
                                         ^
ConfigurationStore.cpp:433:65: error: 'tmp3' was not declared in this scope
         max_acceleration_units_per_sq_second[i]=pgm_read_float(&tmp3[i]);
                                                                 ^
Error compiling.

linjinc, 2015-01-16:
arduino-1.5.8

Marlin	Update printrboard pin assignments	12 hours ago

nophead, 2015-01-16:
Broken by this commit: https://github.com/MarlinFirmware/Marlin/commit/538859669d40b83a1100cb391c1be729db6b3fb9

Seems like  MARLIN_PROGMEM is only defined if GCC_VERSION2 < 40602. All other cases will not compile.

boelle, 2015-01-16:
use ide 1.0.6.... 1.5.8 is only in beta

nophead, 2015-01-16:
Or revert the commit that is obviously not correct. Presumably there should be a #else that defines it as something else, not leave it undefined which will always cause an error.

daid, 2015-01-16:
Reverted it, left a message at the original commit why the whole pull request is a bad idea anyhow.

boelle, 2015-01-16:
so this one could be closed?

lordofhyphens, 2015-01-17:
Looks like it to me.

closed 2015-01-18 by linjinc

Mike-Menci, 2016-08-11:
C:\arduinoHEX\sketch\ConfigurationStore.cpp: In function 'void Config_ResetDefault()':

C:\arduinoHEX\sketch\ConfigurationStore.cpp:215:9: error: 'axis_steps_per_unit' was not declared in this scope

         axis_steps_per_unit[i]=tmp1[i];  

         ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:216:9: error: 'max_feedrate' was not declared in this scope

         max_feedrate[i]=tmp2[i];  

         ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:217:9: error: 'max_acceleration_units_per_sq_second' was not declared in this scope

         max_acceleration_units_per_sq_second[i]=tmp3[i];

         ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:219:5: error: 'acceleration' was not declared in this scope

     acceleration=DEFAULT_ACCELERATION;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:220:5: error: 'retract_acceleration' was not declared in this scope

     retract_acceleration=DEFAULT_RETRACT_ACCELERATION;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:221:5: error: 'minimumfeedrate' was not declared in this scope

     minimumfeedrate=DEFAULT_MINIMUMFEEDRATE;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:222:5: error: 'minsegmenttime' was not declared in this scope

     minsegmenttime=DEFAULT_MINSEGMENTTIME;       

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:223:5: error: 'mintravelfeedrate' was not declared in this scope

     mintravelfeedrate=DEFAULT_MINTRAVELFEEDRATE;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:224:5: error: 'max_xy_jerk' was not declared in this scope

     max_xy_jerk=DEFAULT_XYJERK;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:225:5: error: 'max_z_jerk' was not declared in this scope

     max_z_jerk=DEFAULT_ZJERK;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:226:5: error: 'max_e_jerk' was not declared in this scope

     max_e_jerk=DEFAULT_EJERK;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:227:5: error: 'add_homeing' was not declared in this scope

     add_homeing[0] = add_homeing[1] = add_homeing[2] = 0;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:237:5: error: 'Kp' was not declared in this scope

     Kp = DEFAULT_Kp;

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:238:5: error: 'Ki' was not declared in this scope

     Ki = (DEFAULT_Ki*PID_dT);

     ^

C:\arduinoHEX\sketch\ConfigurationStore.cpp:239:5: error: 'Kd' was not declared in this scope

     Kd = (DEFAULT_Kd/PID_dT);

     ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Mike-Menci, 2016-08-11:
Arduino 1.6.10 - Marlin latest version June 30th 2016  ?? any help out there?

thinkyhead, 2016-08-11:
@Mike-Menci The latest tagged version is `1.1.0-RC7` (July 31). Are you seeing this with that version or in the current `RCBugFix` branch?

Mike-Menci, 2016-08-11:
Yes July 31st sorry! - latest version

Mike-Menci, 2016-08-11:
Hello - Seems working now - was my computer library problem !! Sorry!! My mistake!

Mike-Menci, 2016-08-11:
Which version of GRBL to use _ ? grbl_v0_9i_atmega328p_16mhz_115200

thinkyhead, 2016-08-11:
I don't know. GRBL and Marlin are totally separate things.

Mike-Menci, 2016-08-11:
OK I ask Mrs. Google > Thanks

FelisucoVFR, 2018-02-15:
Hi Mike-Menci,
I have the exact same problem. Can you please tell me which was the library giving you problems?

Thanks

Bob-the-Kuhn, 2018-02-16:
@FelisucoVFR - please open a new issue for your problem.

Please include the following in the new issue:
- What controller you're using.
- What version of Marlin that you're using.
- What compiler you're using (Arduino 1.8.5, PlatformIO, ...)
- What you've tried. 
- A copy of the **section** of the compiler log showing the error.
- A ZIP file containing the Configuration.h, Configuration_adv.h and pins_YOUR_BOARD.h files.

FelisucoVFR, 2018-02-16:
@Bob-the-Kuhn
Thanks Bob. It has been self solved.
It seems an Arduino INO 1.8.5 issue as I just selected another board on it to force an error and going back to the Mega with Ramps 1.4
I am with the last bugfix 1.1.x just dowloaded yesterday.
Many thanks for your work and support!