MarlinFirmware/Marlin #1374: Max feed rate not being applied

Grogyan, 2015-01-16:
TangleRap 
RAMPS 1.2 derivative with onboard Max6675 
EEPROM enabled

Maximum feed rate not being passed from Configuration.h to EEPROM. 

When initially started values in EEPROM should be matched to the constants in the configuration file. 
Should have expected an error message. 

As it was, I had to set in EEPROM manually via 
M203
Then store that with M500

My Z max feed rate is meant to be 1.25mm/min 
M503 reported it to initially be 5mm/min
Had set in configuration.h Z max to be 1.25mm/min

nophead, 2015-01-17:
>When initially started values in EEPROM should be matched to the constants in the configuration file. 
Why do you expect the firmware to override the EEPROM? It is the other way round, EEPROM overrides the defaults in the firmware. You need to run 502: revert to the default "factory settings."

Grogyan, 2015-01-17:
That's right, however I was expecting a warning message when the constants and EEPROM don't match. 

Spent a whole week trying to figure out why the max feed rate on the Z axis wasn't taking in the new values in the constants. 
Would have helped to have a warning message pop up at power on to say that they don't match, would have reduced the wasted time figuring it  out.

bluesign2k, 2015-01-17:
It's often the case that one or more (or maybe all) such default values won't match the EEPROM values if you've tuned your printer. If they did all match then it probably means you're changing the defaults in the firmware to match your setup and re-flashing it... in which case the EEPROM becomes somewhat redundant anyway.

I will admit that I'be fallen foul of the same thing you have when I've been rushing, but I wouldn't call it a bug and I don't think a message is required. At most it might be worth mentioning in the config documentation. Indeed, in a way you are already getting a message but probably ignoring it - Marlin spits out all of the current federates etc when you connect to the board.

Grogyan, 2015-01-17:
I agree, it isn't a bug,  more of a feature request. 
Will change the title of this to be more accurate.

daid, 2015-01-18:
The firmware already reports "Stored settings retrieved" or "Hardcoded Default Settings Loaded" depending on if data is stored to EEPROM.

thinkyhead, 2015-01-28:
#1428 fixes some EEPROM issues and changes the Store / Retrieve messages to include the EEPROM version and the storage size.

closed 2015-01-28 by thinkyhead