MarlinFirmware/Marlin #1375: Stable Branch needed desperately.

lordofhyphens, 2015-01-17:
Due to all of the activity with the Dev branch (what with commits and pulls accidentally breaking stuff here and there), I think that we should come up with a Stable branch.

I think a decent candidate is the old Marlin_v1 branch (which I have a copy of). While it has its issues, they're pretty well-known at this point and we won't have to worry about breaking stuff with regressions. 

Proposed workflow:
1) Issue gets raised.
2) If it's a real bug and not a feature request, then it gets higher priority (or at least as far as "priority" applies to people doing this on their own time).
3) Clone Stable (probably called stable-issuexxxx-fix), and apply fix to that. 
4) Get people to test stable-issuexxx-fix to be extra careful about not breaking anything else.
5) Merge stable-issuesxxx-fix into main Stable.

I'd suggest any feature improvements from Development should probably be incorporated in the same manner -- only stuff that has been tested gets to go into Stable from Development.

boelle, 2015-01-17:
we can get a stable branch VERY easy..... Get more coders

lordofhyphens, 2015-01-17:
That's not a solution. And frankly, I'm trying very hard to not be insulted by that answer.

I've uploaded a copy of the old v1 branch to my staging fork. 
https://github.com/lordofhyphens/Marlin/tree/Marlin_v1

galexander1, 2015-01-17:
The current status quo is not perfect, but we are working towards stable
tested releases.  If development branch doesn't work for you, the new
problems are not very deep and your input is desired.

But the old Marlin_v1 branch was also not stable, if it happened to work
for you when you checked it out, you were simply lucky.  We did not throw
away the mythical stable branch of yore -- we simply changed the name of
what has always been a dev branch.

boelle, 2015-01-17:
my answer was simply just a vent of frustrations over the number of issues raised and the urgent need for coders....  sorry for that one

lordofhyphens, 2015-01-18:
Then I suggest we take the dev branch and call it stable *now*, with what we know doesn't work listed, and then apply the workflow I mentioned (or something similar) to put fixes into it. Stable doesn't mean that all the bugs are fixed, it means it's not under heavy development.

alexborro, 2015-01-18:
In all software development project I have worked in my life we had a
branch just for bug fixing and other for new features/development. The
first one had the state of Release Candidate (RC) and after some time of
testing and bug fixing it was released as Stable. The second one was the
Development Branch.. after hitting a few milestones (usually new features
implemented and bug fixed) it become the RC branch and so one..

Cheers.

Alex.

bkubicek, 2015-01-18:
we do not need more coders, IMHO. What is a nice to have, but not really
necessary, is more educated coders, and this is a slow process. The more
fresh coders there are, the more instable marlin naturally becomes. Its
quite troublesome if people, who do not know why something was done i a
given way,  change concepts is a direction that is problematic.

Bernhard

alexborro, 2015-01-18:
I agree with you @bkubicek . How many problematic PR we had these days? Like you said, this is a slow and natural process. We do not expect having only "super" coders developing to Marlin. That's the reason we need some caution on integrating PRs. Those PRs need the assessment of more Marlin experienced coders. 

And that's the reason @boelle usually ask us before integrating PRs. 

We have had many PRs these days for bug fixing and new features. And we are doing our job carefully analyzing them. The testers are doing great as well.

But I think we could use this idea of Release Candidate:

Development: Unstable by nature. New features and Major bug fixes enter here.
Release Candidate: We freeze Dev branch and start only bug fixing. 
Stable: After some weeks using and testing RC branch, we freeze it to Stable. And it become unchanged. 

Cheers.

Alex.

Sunz3r, 2015-01-18:
+1 @alexborro

daid, 2015-01-19:
@boelle I need your opinion on something. I'm soon in the need for a very basic and clean Marlin. Pretty much nothing but the planner, stepper, temperature and GCode parser routines. Would this be something you would like to base a new "stable" marlin branch off?
(Yes, I'm not even needing SD support, or EEPROM)
I'm going to build this "bare minimal" Marlin anyhow, so might as well talk this over before starting it.

lordofhyphens, 2015-01-19:
@daid that sounds pretty good as terms of a base as anything. Might even
give us the ability to do a better job w/r/t cleanup and refactoring and
testing.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Booli, 2015-01-20:
@daid That sounds awesome. Current Marlin is so bloated, with servos, LCDs, all these different boards. I've been wanting to make one for quiet some time now but just haven't come round to it yet.

thinkyhead, 2015-02-20:
An automated tool to strip out code between inactive #ifdef blocks would be pretty cool. I'm proposing a "code freeze" on new features pending a new tagged version. Only bug fixes during a testing period... the whole shebang. Wotsay?

closed 2015-02-20 by thinkyhead

gege2b, 2015-02-20:
@daid its a good idea, it's actually what my marlin config gives me : an almost "basic" Marlin (with the EEPROM enabled anyway), but no SD, LCD, Auto bed leveling etc...