MarlinFirmware/Marlin #1376: Make a list of supported configurations or try everything?

lordofhyphens, 2015-01-17:
To help us figure out what combinations of hardware we expect to at least compile (and enumerate stuff for testing), can we set up some sort of poll so people can select a motherboard and then check off a stack of HW (going from the defines in configuration.h)? 

We can then prune that list and make a script that compiles Marlin for each of those combinations.

Alternatively, we could browse the available options in Configuration.h and script out something to check every combination (and see which fail and which pass).

Either way, it'd be a simple way for other developers to do a quick sanity check on anything they want to commit so that their work doesn't break the builds. 

The second option is probably less work but has more noise (there will be some combinations that will just not compile).

monkeydave, 2015-01-17:
Travis CI does this already for some of the configuration.h file options see the .travis.yml file in the repo for the script that runs the checks. I haven't had time this last couple of weeks to add more boards and options to the checks but it is on my todo list. 

Developers can also add it to their fork so they can check the changes they have made, it also checks every pull request and has caught a few bad pull requests but it is not fool proof.

See https://travis-ci.org/MarlinFirmware/Marlin for more information

lordofhyphens, 2015-01-18:
Thanks for that, I've been looking into seeing what else can be caught with an automated test system, either with the Unity (https://github.com/ThrowTheSwitch/Unity) or CppUTest (https://cpputest.github.io/) frameworks.

I'll see about working up some sort of script against boards.h and configuration.h to extract everything. 

I think with some careful use of mock objects we can get better automated testing working (because nobody likes manual testing).

lordofhyphens, 2015-01-18:
Currently working on a very basic testing shell script that tries to compile every board type (pulled from the defines in boards.h). I haven't looked at the travis stuff yet--I figure that being able to generate the list of possible combinations would be sufficient for now.

Unfortunately, the makefile seems to be at least partially broken (I think)--compiling with 
several board types gives 

make: *** No rule to make target 'applet/wiring.o', needed by 'applet/Marlin.elf'.  Stop.

as a make error. I had invoked everything with the stock Configuration.h (commit 993bcd370f7f599d03cbf4f4bd34853ae0c9017f) and the following variables set. 
make HARDWARE_MOTHERBOARD=$board DEFINES="MOTHERBOARD=$board" 

I think it might be because the libs for the processors and those boards are missing, but can't be sure.

thinkyhead, 2015-01-28:
The .travis.yml test has been invaluable in helping me to catch errors. It would be great to have it test more configurations, but Pull Request tests will take 40 times longer if we include all the boards. I wonder if there's some way to have Travis CI do automated testing of some representative configurations when we do Pull Requests, but allow us to do more thorough testing when we want it.

boelle, 2015-04-26:
we could always have 2 copies of YML and just rename them and force travis to do a check?

boelle, 2015-05-31:
this one should be simple.... we support everything that ultimaker puts out, then Ramps boards with the standard configs. that should cover enough. if people want more boards checked they could always submit a PR

closed 2015-05-31 by boelle