MarlinFirmware/Marlin #1378: EPROM cause extruder problem.

HackerTon, 2015-01-18:
When EPROM is enable, extruder stepper is not able to turn at any condition.

arno-millenaar, 2015-01-18:
Confirmed this maybe a problem.  See my comment at https://github.com/MarlinFirmware/Marlin/issues/1209

nophead, 2015-01-18:
If you download new firmware with a different EEPROM layout you will get garbage values until you do  M502: revert to the default "factory settings" to re initialise it. There is a version number in the EEPROM that can be changed if the layout changes but nobody does and having conditional variables will cause layout changes anyway when options are changed.

arno-millenaar, 2015-01-18:
@nophead 
thanks a million as indeed this was my bad. I should have known as i typically always issue a M502 followed by a M500 when loading new firmware.  For whatever reason I was convinced I had done this, however I just tested again, after a M502/M500 combo and now its working fine. With EEPROM and CHITCHAT enabled, extruder etc is working.

Thanks again for your feedback as this problem is solved and confirmed after testing.

HackerTon, 2015-01-18:
I did use M502 but the extruder is still not working.
My setup:Megatronics 2.0, DRV8825 in 16 microstepping, ATX power supply, RRD Full Graphics LCD.

Thanks for you collaboration.

arno-millenaar, 2015-01-18:
After using M502, you should also issue the M500 command. Did you try that as well ?

At least I now seem to have working firmware, including working extruders, having pulled the dev branch this morning.

HackerTon, 2015-01-18:
Yes, I did what you told me to.

arno-millenaar, 2015-01-18:
Just to exclude everything else have you re-loaded your previous (working) firmware, just to make sure its caused by this specific version ?  As far as I can tell from the responses from the other Testers they all seem to have working machines with this specific dev version.  Also try disabling your LCD function, just to make sure their is no relationship as I don't have your type of LCD module.

HackerTon, 2015-01-18:
Finally, I got it working only by using M502 alone. After restart, problem occur again and also M502/M500 combo is not working.

arno-millenaar, 2015-01-18:
After issuing M502/M500 and give it a hard reset do you read back the expected values issuing M503 ?
At least it seems you are heading in a similar direction that 'solved' my initial problem.

HackerTon, 2015-01-18:
This is what i got after issuiing M503
(01:41:19.435 : echo:Steps per unit:
01:41:19.435 : echo:  M92 X78.74 Y78.74 Z533.33 E836.00
01:41:19.435 : echo:Maximum feedrates (mm/s):
01:41:19.435 : echo:  M203 X500.00 Y500.00 Z5.00 E25.00
01:41:19.435 : echo:Maximum Acceleration (mm/s2):
01:41:19.435 : echo:  M201 X9000 Y9000 Z100 E10000
01:41:19.435 : echo:Acceleration: S=acceleration, T=retract acceleration
01:41:19.435 : echo:  M204 S3000.00 T3000.00
01:41:19.436 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
01:41:19.436 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
01:41:19.436 : echo:Home offset (mm):
01:41:19.436 : echo:  M206 X0.00 Y0.00 Z0.00
01:41:19.436 : echo:PID settings:
01:41:19.436 : echo:   M301 P22.20 I1.08 D114.00
)

After 
(01:42:29.296 : echo:Stored settings retrieved
01:42:29.296 : echo:Steps per unit:
01:42:29.296 : echo:  M92 X78.74 Y78.74 Z533.33 E836.00
01:42:29.296 : echo:Maximum feedrates (mm/s):
01:42:29.296 : echo:  M203 X500.00 Y500.00 Z5.00 E25.00
01:42:29.296 : echo:Maximum Acceleration (mm/s2):
01:42:29.296 : echo:  M201 X9000 Y9000 Z100 E10000
01:42:29.296 : echo:Acceleration: S=acceleration, T=retract acceleration
01:42:29.296 : echo:  M204 S3000.00 T3000.00
01:42:29.296 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
01:42:29.296 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
01:42:29.296 : echo:Home offset (mm):
01:42:29.296 : echo:  M206 X0.00 Y0.00 Z0.00
01:42:29.296 : echo:PID settings:
01:42:29.296 : echo:   M301 P22.20 I1.08 D114.00
)

After all the testing, not working. Curious.

thinkyhead, 2015-01-19:
@nophead @arno-millenaar I'm seeing a few messages indicating confusion about this. It seems like a good idea to add code that checks the version number in the EEPROM and resets the EEPROM when it's not the current version – perhaps with some kind of feedback to the user. If an issue hasn't been added for this, I will toss it into the mix now.

closed 2015-01-19 by thinkyhead

nophead, 2015-01-19:
As far as I can see it already does that but nobody changes the version number.

thinkyhead, 2015-01-21:
Hmm... Maybe there could be a failsafe that checks against the _size_ of the EEPROM data...? Since users might enable or disable features that change the EEPROM storage also. Any other ideas bout how it could automatically catch this?

nophead, 2015-01-21:
daid's suggestion of a magic value at the end should catch size changes.

At the moment the size isn't easy to calculate as each variable is written separately. A better scheme is to make a structure with the layout in it. I think that would save a lot of code and you can then take its size.

thinkyhead, 2015-01-23:
@nophead +1 on the struct idea.

thinkyhead, 2015-01-28:
#1428 should makes the EEPROM consistent regardless of enabled features.

tadawson, 2015-07-23:
Try M200 D0 . . . I have found a problem where once volumetric extrusion gets set, the code won't allow you do disable it - the values hang in prom, and it comes back on every reset . . . if this does the trick, likely we are having the same problem.

thinkyhead, 2015-07-24:
@tadawson So far I don't see the behavior, at least not reflected in the current settings.

Here's the procedure I used to test:

- `M200 D1.69` to set `volumetric_enabled`, the `filament_size` for the `active_extruder`, and update all `volumetric_multiplier` values.
- `M503` to see the current state. It replies `M200 D1.69`.
- `M500` to save the settings.
- Power down and reboot. EEPROM restored, says `M200 D1.69` (correct).
- `M200 D0` to disable `volumetric_enabled` and set all `volumetric_multiplier` values to `1.0`.
- `M503` to see the current state. It replies: `Filament settings: Disabled`.
- `M500` to save current state (disabled).
- Power down and reboot. EEPROM restored, says `Filament settings: Disabled` (correct).

Do you see the same output as I do?
Is the volumetric behavior still happening when it claims (via `M503`) to be turned off?

tadawson, 2015-07-24:
The  main artifact I see is that my extruder volume goes insane unless I correct as described. I also feel a little less nuts, since I had the file name written both ways as I updated my bug, from  two different systems, with two separate git images . . . the quickee I didn't get switched to Development was last, hence the filename . . . . I thought I was going nuts!

I don't think M503 base output is accurate, since one of the first things I did before I found this was to capture the output of M503 when good and when bad, and noted no difference. I honestly think that while volumetric_enabled is being set correctly, and which determines the output of M503, the values are still being applied.

As I recall, M503 S0, to show just the gcode did show the issue . . .

I'll retest tomorrow. . . my  memory is that I see what you see except the last, I would still see M200 D1.69

My codebase might be a week or so behind dev - this is a port to Due, but the owner brings in updates often, and on comparison, I did not see any diffs in Marlin_main.cpp or configuration_store.cpp.

- Tim

thinkyhead, 2015-07-24:
> I don't think M503 base output is accurate

It's accurate concerning the values that it displays. But it doesn't reveal the values that are used by the planner, specifically the `volumetric_multiplier[]` array. If these values are set to anything other than 1.0 then you will get volumetric behavior in the planner. The `volumetric_multiplier` values are recalculated only on these occasions:

- Restore with `M501`. All set to 1.0 if `volumetric_enabled`. Otherwise all recalculated.
- Reset with `M502`. Always set to 1.0 by this.
- Any `M200 Dn` command. All set to 1.0 if `volumetric_enabled`. Otherwise all recalculated.
- Edit with the LCD Menu. All set to 1.0 if `volumetric_enabled`. Otherwise all recalculated.

tadawson, 2015-07-24:
I can tell you for certain, that the only way I can get extrusion back to normal is to set filament size in the EEPROM to zero . . . otherwise, I get radically short extrusions . . .

And the M503 base output echos exactly the llast value that was entered by M200 Dn, so in that regard, it's accurate, since nothing in Marlin will zero those values. Even if I set to my actual filament size(M200 D1.75), I still get wrong volume.

Frankly, were there a conditional to disable this feature, that would have been my solution . . .

tadawson, 2015-07-24:
OK, testing . . . I have added trace echo messages in Marlin_main.cpp as well as configuration_store.cpp to give status and values of volumetric_enabled as well as diameter when the setting function in Marlin_main is hit, as well as either when read or write of the eeprom works with volumetric_enabled or filament_size . . . 

Started with clean prom, M503 gives "Filament settings disabled" as expected. 

M200 D1 entered. Hit trace in Marlin_main that volumetric_enabled is set, and diameter is 1.0

M500 entered. Got trace from write that volumetric_enabled is set, diameter written as 1.0, as expected. 

M503 entered. Echos M200 D1.00 as expected. 

M200 D0 entered. Hit trace of volumetric_disabled in Marlin_main, no diameter change made - variables still at 1.0

M500 entered. Trace shows volumetric_disabled in EEPROM, wrote value 1.0

M503 entered, get back "Filament settings disabled" 

Printer reset

M503 entered. Get back Filament settings: M200 D1.00

Entered M200 D0, trace proper as above.

Entered M500, trace proper as above

Entered M503, shows Filament settings disabled

Entered M501, get back read trace of volumetric_enabled as *true* in trace in read routing in configuration_store.cpp

Entered M503, get back M200 D1.00

OK, now for potential fixes:  

I changed Marlin_main.cpp to:

if (code_seen('D')) {
    float diameter = code_value();
    // setting any extruder filament size disables volumetric on the assumption that
    // slicers either generate in extruder values as cubic mm or as as filament feeds
    // for all extruders
    volumetric_enabled = (diameter != 0.0);
    if (volumetric_enabled) {
      SERIAL_ECHO_START; //Dawson - debug
      SERIAL_ECHOPGM("Volumetric Enabled in Marlin_main.cpp");
      SERIAL_ECHOPAIR("Diameter set to ",diameter);
      SERIAL_EOL;
      filament_size[tmp_extruder] = diameter;
      // make sure all extruders have some sane value for the filament size
      for (int i=0; i<EXTRUDERS; i++)
        if (! filament_size[i]) filament_size[i] = DEFAULT_NOMINAL_FILAMENT_DIA;
    }
    else {
      SERIAL_ECHO_START;
      SERIAL_ECHOPGM("Volumetric disabled in Marlin_main.cpp");
      SERIAL_EOL;
      filament_size[tmp_extruder] = 0;
      // make sure all extruders have some sane value for the filament size
      for (int i=0; i<EXTRUDERS; i++)
        if (! filament_size[i]) filament_size[i] = DEFAULT_NOMINAL_FILAMENT_DIA;
    }
  }
  else {
    //reserved for setting filament diameter via UFID or filament measuring device
    return;
  }
  calculate_volumetric_multipliers();
}

Pushing the zero value to the filament size on disable resolves the problem.  The zero values are seen to write to prom, and M503/M501 report back "Filament settings disabled"

I backed that out, and set configuration_store.cpp to always return zero for diameter if it sees volumetric_enabled as false:

   for (int q = 0; q < 4; q++) {
      EEPROM_READ_VAR(i, dummy);
      SERIAL_ECHO_START; //Dawson
      SERIAL_ECHOPGM("Hit volumetric read");
      SERIAL_EOL;
      if(volumetric_enabled) {
        SERIAL_ECHO_START;
        SERIAL_ECHOPGM("Volumetric enabled is true in read");
        SERIAL_EOL;
      }    
      // Dawson - conditional to return 0 when disabled . . .
      if(!volumetric_enabled) {
         dummy = 0;
         SERIAL_ECHO_START;
         SERIAL_ECHOPGM("Volumetric Disabled in read");
         SERIAL_EOL;
      }
      if (q < EXTRUDERS) filament_size[q] = dummy;
    }

This, interestingly, does *NOT* work - M501 right after M500 comes back with a status of volumetric_read is enabled, right after watching it set to false in the write.  It can't be a bad prom, since when the value written is zero, it *does* return disabled . . . 

So, I added another trace point, to output the value of i, the EEPROM offset at the time of an operation, and the result was interesting, to say the least . . . 

      SERIAL_ECHO_START;
      SERIAL_ECHOPGM("volumetric_enable offset is: ");
      SERIAL_ECHO(i);
      SERIAL_EOL;
  EEPROM_WRITE_VAR(i, volumetric_enabled);

      SERIAL_ECHO_START;
      SERIAL_ECHOPGM("volumetric_enable offset is: ");
      SERIAL_ECHO(i);
      SERIAL_EOL;
    EEPROM_READ_VAR(i, volumetric_enabled);

On write, I see a single write to location offset 377 . . . 

On read, I see a single read from location 389 . . . 

Well, this seems to explain a lot . . . Putting the same trace on the diameter read/write, I see a similar offset . . . 378, 382, 386, 390 for write, and 390, 394, 398, 402 for read.  Since the code write all four values, and I have 1 extruder, that explains how I get the value back - 390 is common to both, but on read, volumetric_enabled is reading a latter byte of the third value, so when values are zero, it unsets.   

<* Insert much sleepless debugging here . . . *> 

I found it!  Simple, yet incredibly annoying, and yes, this has been fixed in later builds . . . 

        for (int q=4; q--;) EEPROM_READ_VAR(i, dummy); // Ki, Kd, Kc, Km

Was the culprit!  The write routing (as well as later code) corrected this to 3 . . . which gets rid of the 4 byte offset, and now it *WORKS*!  

Thanks for your time (and sorry for wasting it . . . a *one* character change in a section of configuration_store.cpp that wasn't even related to volumetric extrusion . . . grrrr . . . 

*But* I know the Marlin source a heck of a lot better now!

Feel free to close this as already fixed/NPF . . .