MarlinFirmware/Marlin #1379: Panulolu2 Lcd  compile problems

tanmanguy, 2015-01-18:
When enabling Panulolu2 
And defining btn_enc =-1 to say that the encoder is i2c a bunch of compile errors happen. 
FYI I'm using a printrboard Rev e

T3P3, 2015-01-18:
While the hardware supports using the Panelolu2 encoder via the i2c bus Marlin does not support this and to do so is no easy because the way i2c is implemented in arduino does not play nicely with interrupts (and anyway the interrupt code needs to be optimised for stepping, not for reading encoder clicks). I strongly recommend using 3 digital pins for the two encoder pins and the click pin.
Tony

tanmanguy, 2015-01-18:
This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
Arduino: 1.0.5-r2 (Windows NT (unknown)), Board: "Teensy++ 2.0"
ultralcd.cpp: In function 'void lcd_init()':
ultralcd.cpp:1158: error: 'DIOBTN_EN1_DDR' was not declared in this scope
ultralcd.cpp:1158: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1159: error: 'DIOBTN_EN2_DDR' was not declared in this scope
ultralcd.cpp:1159: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_RPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_RPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp: In function 'void lcd_buttons_update()':
ultralcd.cpp:1378: error: 'DIOBTN_EN1_RPORT' was not declared in this scope
ultralcd.cpp:1378: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1379: error: 'DIOBTN_EN2_RPORT' was not declared in this scope
ultralcd.cpp:1379: error: 'DIOBTN_EN2_PIN' was not declared in this scope

this is the error for either setting up #define BTN_ENC 1 or #define BTN_ENC -1

tanmanguy, 2015-01-18:
i think i fixed those prier issues by coping the #define for lcd from melzi.

now im getting this In file included from C:\Users\taner\Documents\Arduino\libraries\LiquidCrystal\LiquidCrystal.cpp:1:
C:\Users\taner\Documents\Arduino\libraries\LiquidCrystal\/LiquidCrystal.h:82: error: conflicting return type specified for 'virtual size_t LiquidCrystal::write(uint8_t)'
C:\Users\taner\Downloads\arduino0022\arduino-0022\hardware\at90usb1286\cores\at90usb1286/Print.h:98: error:   overriding 'virtual void Print::write(uint8_t)'

thinkyhead, 2015-01-19:
As for that last error, could it be that your `at90usb1286/Print.h` file is too old? The clue comes from `LiquidCrystalRus.h` which also overrides `Print::write`. The code in that file contains this definition:

    #if defined(ARDUINO) && ARDUINO >= 100
      virtual size_t write(uint8_t);
      using Print::write;
    #else
      virtual void write(uint8_t);
    #endif

tanmanguy, 2015-01-20:
so by compiling it in arduino 0022 i am able to compile it with no errors but i get no controls

thinkyhead, 2015-01-21:
Hey someone else saw this before... https://groups.google.com/forum/#!msg/deltabot/FmYLhQRAUYI/exJ5I1BumjcJ

tanmanguy, 2015-01-21:
with the type of boot loader i have i think im stuck with arduino 0022 unless there is a way to save  as hex file in arduino 1.05?

tanmanguy, 2015-01-21:
just tried it 1.05 and 1.06 i still get this
  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
Arduino: 1.0.6 + Td: 1.20 (Windows NT (unknown)), Board: "Teensy++ 2.0"
ultralcd.cpp: In function 'void lcd_init()':
ultralcd.cpp:1158: error: 'DIOBTN_EN1_DDR' was not declared in this scope
ultralcd.cpp:1158: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1159: error: 'DIOBTN_EN2_DDR' was not declared in this scope
ultralcd.cpp:1159: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_RPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_WPORT' was not declared in this scope
ultralcd.cpp:1160: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_RPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_WPORT' was not declared in this scope
ultralcd.cpp:1161: error: 'DIOBTN_EN2_PIN' was not declared in this scope
ultralcd.cpp: In function 'void lcd_buttons_update()':
ultralcd.cpp:1378: error: 'DIOBTN_EN1_RPORT' was not declared in this scope
ultralcd.cpp:1378: error: 'DIOBTN_EN1_PIN' was not declared in this scope
ultralcd.cpp:1379: error: 'DIOBTN_EN2_RPORT' was not declared in this scope
ultralcd.cpp:1379: error: 'DIOBTN_EN2_PIN' was not declared in this scope

i copied over the correct libraries from marlin, this is also most resent marlin from 1/20/2015,

thinkyhead, 2015-01-22:
Did you try this one...?
https://github.com/lincomatic/arduino-1.0.5-r2-at90usb1286

tanmanguy, 2015-01-22:
found the fix comnpared it to marlin_v1 dont remember where i got that but the i2c controls worked,
fix
in pins.h
in #if MB(TEENSYLU) || MB(PRINTRBOARD)
the following was missing
#ifdef ULTRA_LCD
   #ifdef NEWPANEL
     //we have no buzzer installed
     #define BEEPER -1
     //LCD Pins
     #ifdef LCD_I2C_PANELOLU2     
	     #define BTN_EN1 27  //RX1 - fastio.h pin mapping 27
	     #define BTN_EN2 26  //TX1 - fastio.h pin mapping 26
	     #define BTN_ENC 43 //A3 - fastio.h pin mapping 43
	     #define SDSS   40 //use SD card on Panelolu2 (Teensyduino pin mapping)
	 #endif //LCD_I2C_PANELOLU2
     //not connected to a pin
     #define SDCARDDETECT -1    
   #endif //Newpanel
 #endif //Ultipanel

i tried using one a section from the SANGUINOLOLU, realized those pins were not the correct ones.

thanks

thinkyhead, 2015-01-23:
@T3P3 I've rolled a PR with changes proposed by @tanmanguy. Are the issues that you cite concerning I2C serious enough to merit passing on this change? It seems to work okay for @tanmanguy but perhaps there are other configurations where problems will show up more...?

tanmanguy, 2015-01-24:
so just tested my changes i had to set 
pins.h

#if MB(TEENSYLU) || MB(PRINTRBOARD)

#define TEMP_1_PIN         -1
#define TEMP_2_PIN         -1

to -1 they were referencing to the same pin that was being used for the i2c control

thinkyhead, 2015-01-24:
__pins.h__ is one of those cleanups we all dread, but it needs to be done – for consistency and clarity, if nothing else. I would like to see it split up by board, in fact. A project for another day.

So, those two lines should just go anywhere in the TEENSYLU || PRINTRBOARD block (to override values set earlier in the file)?

tanmanguy, 2015-01-24:
Sorry wasn't clear those are already there set to 2 and 3 or what ever. They were they just need to be set to -1

thinkyhead, 2015-01-24:
I'm setting them to -1 but wondering why they were set to 2 and 3 in the first place. Are you sure these pins are unused on all TEENSYLU and PRINTRBOARD (AT90USB1286) setups? (I see that FILWIDTH_PIN is _also_ set to 2, so it looks like pin 2 is general-use on these boards, making it more likely that pin 3 is too.)

tanmanguy, 2015-01-24:
If I new more about the arduino reference to the pin outs on the printrboard. I would be able to answer that. I think there is a way of using expansion 2 on any printrboard from rev e and below with the adapter for the panelolu 2 screen, which breaks out 4 un used pins. For the including the extrudrboard. To get tri extrusion in the future. But I'm waiting for my cyclops from e3d to figure the rest out. Hopefully I'll make a tutorial for that because the pins will be split up a lot. When it comes to adding the extrudr board. Ramble ramble ramble. Llol sorry this is long and point less

thinkyhead, 2015-01-25:
No worries. Your answer is actually very helpful.

thinkyhead, 2015-02-20:
Closing this to make room for more issues, and perhaps a code freeze. Next round!

closed 2015-02-20 by thinkyhead