MarlinFirmware/Marlin #1385: Unprotected interrupt-sensitive code and timing bug in babystep() function

scottlawsonbc, 2015-01-19:
There are two problems with the code in the <code>babystep()</code> function in <code>stepper.cpp</code>:

#### Unprotected interrupt-sensitive code ####
**Line 1093:**
```cpp
    //MUST ONLY BE CALLED BY A ISR, it depends on that no other ISR interrupts this
```
Scary code! The <code>babystep()</code> function executes code that should never be interrupted by the stepper driver interrupt since it modifies the state of stepper driver pins. Instead of commenting this, the function should properly disable and re-enable the interrupts with the <code>cli()</code> and <code>sei()</code> methods.

#### Timing bug ####
**Lines 1114, 1146, 1179, 1215**
```cpp
    
    //perform step 
    WRITE(X_STEP_PIN, !INVERT_X_STEP_PIN); 
    //wait a tiny bit
    {
    float x=1./float(axis+1); //absolutely useless
    }
    WRITE(X_STEP_PIN, INVERT_X_STEP_PIN);
```
The datasheet for the (presumably) Polulo stepper driver calls for a 1 microsecond delay between pin state changes. From what I can tell, this code is supposed to create a delay by creating a <code>float</code> and computing a very expensive floating point division.

Issues with using math operations instead of NOP instructions aside, the big issue here is that the **compiler completely optimizes that code away, leaving no delay at all!** Even if it wasn't optimized away, it would consume hundreds and hundreds of CPU cycles, creating an insane 30-80 microsecond delay **in an interrupt routine**! This may be causing unintended behaviour and missed steps, especially with 20MHz chips.
 
This code should be replaced with:

```cpp
    /* Precise 1 microsecond delay */
    __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");  /* 250 ns delay with 16MHz chips */
    __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");
    __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");
    __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");
```

CONSULitAS, 2015-01-19:
@scottlawsonbc 

What you are writing makes sense to me. If x is not used outside of the loop, the calculation will be optimized away and no delay will happen.

It seems to me that issue  "Marlin and THB6128 Driver" #1041 could be caused by this. What do you think?

It tells that specific stepper drivers loose steps - the experts think about timing problems. 

I would suggest to make your change and let the users who reported test if its gone then.

Just my 2ct :-)

daid, 2015-01-19:
First, sei() and cli() should be used with great care. It's much better to use the ATOMIC_BLOCK macro. Example:
https://github.com/Ultimaker/Ultimaker2Marlin/blob/i2c_refactor/Marlin/i2c_driver.cpp#L59

Next. Tiny delays should be done with the <util/delay.h> macros:
http://www.nongnu.org/avr-libc/user-manual/group__util__delay.html

Or, any math done below the write of the step pin should be moved inside (as it's done in the normal stepper routine, where it updates some variables that need changes anyhow)

galexander1, 2015-01-19:
I could be wrong, but it seems like babystep() is only called from the
TIMER0 ISR in temperature.cpp.  That's a little bit of an awkward place
to put it, but isn't it "working as designed" already?

It seems like for a function that should only be called in an ISR, an
appropriate protocol would be to put a comment saying as much, and then
to only call it from ISRs.  And if that is what is happening, I don't
understand why we would change it?

daid, 2015-01-19:
(Also, disabling the ISRs for a longer period of time should not be done, as this will mess with the temperature controller)

@bkubicek you made the babystepping right, guess you best know what's wrong and right here.

scottlawsonbc, 2015-01-19:
I was not aware of the ATOMIC_BLOCK, that is a useful macro! From what I can tell it does the same thing as cli() and sei() except that the atomic block does not re-enable interrupts if they weren't enabled in the first place. I'm thinking that the babystep() function should be wrapped in the atomic block as @daid suggested.

@galexander1 you are correct in saying that right now the <code>babystep()</code> function is only being called from an interrupt routine. It is low priority but still a good idea to protect the function with sei()/cli() or the atomic block to prevent hard-to-debug interrupt glitches if the code is changed in the future.

daid, 2015-01-19:
@scottlawsonbc the avr-libc documentation isn't that big. So I think it's recommended readying for anyone who does a bit of AVR code:
http://www.nongnu.org/avr-libc/user-manual/modules.html

I wished avr-gcc had a method to annotate functions so they are only allowed to be called from interrupt context or not-interrupt context. Anyhow, the babystep function is clearly marked as "ONLY CALL FROM INTERRUPT!"

(Also, the whole of Marlin is "scary code", pretty much the nature of hardcode 8-bit embedded code doing a lot of things at once)

galexander1, 2015-01-19:
On Mon, Jan 19, 2015 at 11:33:01AM -0800, Scott Lawson wrote:
> It is low priority but still a good idea to protect the function with
> sei()/cli() or the atomic block to prevent hard-to-debug interrupt
> glitches if the code is changed in the future.

No, it isn't a good idea.

If babystep() is called in another context other than the context it was
designed for, then whoever makes that change will need to truly
understand the interactions between babystep() and the rest of stepper
control.  There is no quick hack today we can make that will save the
future programmer from the burden of understanding the code he is changing.

In point of fact, if you put sei() in an ISR, that *will* introduce
hard-to-debug interrupt glitches.  Anyone who changes stepper control
really does have to understand what they're doing.

Never introduce a poorly-thought-out change today in anticipation of
another poorly-thought-out change tomorrow.  Think it through well in
terms of the needs of today.  That's good software engineering practice
in general, and in embedded software it is absolutely essential.

scottlawsonbc, 2015-01-19:
<code>stepper.cpp</code> disassembly is showing that the <code>float x</code> delay is being fully optimized out.

With the current disassembly no instructions are being generated: 
```
    {
    float x=1./float(axis+1)/float(axis+2); //wait a tiny bit
    }
    WRITE(X_STEP_PIN, INVERT_X_STEP_PIN);
```

However, if one of the<code>float x</code> instances is changed to <code>volatile float x</code> the sketch size becomes larger by 40 bytes and instructions are generated:
```
    {
    volatile float x=1./float(axis+1)/float(axis+2); //wait a tiny bit
   12862:	80 e0       	ldi	r24, 0x00	; 0
   12864:	90 e0       	ldi	r25, 0x00	; 0
   12866:	a0 e0       	ldi	r26, 0x00	; 0
   12868:	bf e3       	ldi	r27, 0x3F	; 63
   1286a:	89 83       	std	Y+1, r24	; 0x01
   1286c:	9a 83       	std	Y+2, r25	; 0x02
   1286e:	ab 83       	std	Y+3, r26	; 0x03
   12870:	bc 83       	std	Y+4, r27	; 0x04
    }
    WRITE(X_STEP_PIN, INVERT_X_STEP_PIN);
```

Replacing <code>float x</code> with <code>_delay_us(1U)</code> generates instructions for the intended delay:
```
12858:	8a 95       	dec	r24
1285a:	f1 f7       	brne	.-4      	; 0x12858 <_Z8babystephb+0x58>
#endif
{
  //volatile float x=1./float(axis+1)/float(axis+2); //wait a tiny bit
  _delay_us(1U);
}
```

bkubicek, 2015-01-21:
@scottlawsonbc is the disassembly from the default arduino compiler flags, or from makefile settings, e.g. an 01 or higher?
I was not aware that under the Os optimization this expression would be optimized away.

scottlawsonbc, 2015-01-21:
@bkubicek the disassembly was generated using default Arduino compiler optimization flags (`-Os`)

The `-Os` [optimization flag](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html) uses all optimizations that won't increase code size (such as [loop unrolling](http://en.wikipedia.org/wiki/Loop_unrolling)).

>Optimize for size. -Os enables all -O2 optimizations that do not typically increase code size. It also performs further optimizations designed to reduce code size.

The compiler can guarantee that the `float x` variable is never referenced anywhere else, which is especially easy to prove since the variable has its own scope. Keeping `float x` would generate unnecessary instructions that contribute to the code size, and since the compiler is optimizing for code space it completely optimizes the `float x` variable away.

thinkyhead, 2015-01-24:
___In point of fact, if you put sei() in an ISR, that *will* introduce hard-to-debug interrupt glitches.  Anyone who changes stepper control really does have to understand what they're doing.___

It seems like we can kick this one down the road. The optimized-away code has been replaced with a proper delay, and there won't be any changes to babystep for now.

closed 2015-01-24 by thinkyhead