MarlinFirmware/Marlin #1387: Should include instructions for how to add libraries in marlin

NarimaanV, 2015-01-19:
After some debate in issue https://github.com/MarlinFirmware/Marlin/issues/1366, it seems something we should do now at least, regardless of whether or not its better to include libraries in Marlin or just add them to the IDE is to include some instructions in Marlin about how to do that for users who aren't familiar with adding libraries to Marlin (since they still have to run certain displays). I've already got a draft of the file to add to the documentation done and plan on adding some comments wherever it mentions to add the library to refer to that documentation, but I noticed something regarding the layout of the "ArduinoAddons" folder that I wanted to clarify before presenting the Pull Request. Within that "ArduinoAddons" folder, only the "Arduino_0.0.x" and "Arduino_1.0.x" folders have all the libraries you could need for Marlin ("Arduino_1.5.x" has one of the libraries buried deep in one of the sub folders but I don't think that entirely counts). So I'm wondering, is this intentional? Is there something about the 1.5.x versions of Arduino that doesn't need the libraries? Or should I just direct everyone to the "Arduino_1.0.x" folder regardless of what version they're using?

justmyopinion, 2015-01-19:
Basically this information is already present in Marlin. here is an example from configuration.h where specific display is selected:
"#define LCD_I2C_VIKI
#ifdef LCD_I2C_VIKI
  // This uses the LiquidTWI2 library v1.2.3 or later ( https://github.com/lincomatic/LiquidTWI2 )
  // Make sure the LiquidTWI2 directory is placed in the Arduino or Sketchbook libraries subdirectory.
.........
What more would you need?
"

daid, 2015-01-19:
I think someone made the mistake of moving core functionality into a library. That simply won't cut it for 95% of the users. As suddenly almost everyone his hit with installing libraries, instead of a few odd cases.

For the rest, the above mentioned by "justmyopinion" will do.

NarimaanV, 2015-01-19:
Thing is though, you can be an expert on 3D printing and know how to get 99% of the time perfect prints and not know a thing about code. To someone like that (I used to know everything about 3D printing, and but nothing about code 6 months ago before learning C++), the terms "library", "directory", and "sketchbook" can be complete gibberish. In that case, I don't think the instructions are in any way near being clear enough what to do. What I'm proposing is a pseudo-dummy's guide to addinf libraries, without any complicated terminology. Some simple, roughly 6 steps that just say what they need to do without any complex terminology. And then there would just be a note with the current instructions that for further information refer to the new file in documentation. True, anyone who already knows it wouldn't need to use it. I mean, it doesn't hurt to have too much instructions for the apparently minority that might not know how to do it, now does it?

thinkyhead, 2015-01-21:
There should certainly be documentation about the current library dependencies (and potential dependencies) which gives a good layman's description of the process. If more library dependencies are added then those can be mentioned (or not) also in the documentation. `u8glib` is the most common dependency, for graphical display users.

NarimaanV, 2015-01-21:
So then, to clarify, is it best to direct the users to the arduino addons folder (currently only the Arduino_1.6.x since thats the most recent one that actually has libraries) or the download links shown in their respective places throughout configuration.h?

thinkyhead, 2015-01-23:
The download links will be useful for all those add-ons not included with Arduino (or UECIDE, etc.). Libraries like __u8g__ need to be downloaded and added to the Arduino Sketchbook __libraries__ folder through the `Sketch > Import Library… > Add Library…` command (UECIDE: `Sketch > Libraries > Install library archive…`) for example. The `Marlin.ino` file seems to be set up already to include external libraries that may be needed, so once libraries are installed they should just work.

NarimaanV, 2015-01-23:
I mean configuration.h already has all the download links wherever it says to install the respective library, so I don't think they all need to be listed in the proposed "AddingLibraries" documentation file. My only concern was whether to direct users to use the actual links in configuration.h or direct them to go to the ArduinoAddons folder instead, though after some thought, it seems the links are probably the best bet to get the libraries from.

thinkyhead, 2015-01-23:
Yeah, just refer to the Configuration files generally. They can change over time and we won't have to keep updating the docs.

nophead, 2015-01-24:
If people download libraries from links they will get the latest version,
which may not be compatible and might not have been tested. That is why the
libraries are shipped with Marlin in the ArduinoAddons folder. Since they
are all small (you can't fit a large library in an Arduino) I don't see a
problem with bundling them in this way.

thinkyhead, 2015-01-24:
Somehow I missed (or forgot) that the libraries are included in the ArduinoAddons folder. I've actually never used those copies, but just downloaded them from the web. Truly, I am the _raison-d'être_ –the poster-child– for this documentation project.

So, yeah…. I think it's a nice courtesy to include links for the libraries, but we should definitely point users at `ArduinoAddons` for their immediate library needs (in the Configuration comments as well).

NarimaanV, 2015-01-24:
Here's all the libraries mentioned in Configuration.h with their links:

Line 561 - U8glib (Full Graphic Controller): http://code.google.com/p/u8glib/wiki/u8glib
Line 571 - LiquidCrystal_I2C.h (RA Control Panel): https://github.com/kiyoshigawa/LiquidCrystal_I2C
Line 614 - LiquidCrystal_I2C (Sainsmart Ywrobot LCD): https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home
Lines 625 and 655 - LiquidTWI2 (Panelolu2 LCD and VIKI LCD): https://github.com/lincomatic/LiquidTWI2

The main issue I have with the ArduinoAddons folder, and someone please correct me if I'm missing something here, is that it currently only has the first two libraries listed above in it. So if you use any of the last three panels listed (Sainsmart, Panelolu2, or VIKI), then the instructions won't work because the libraries just won't be there. So a proposed solution would be to say if they're trying to use the Full Graphic Controller or the RA Control Panel, to use the ArduinoAddons folder, otherwise, download the libraries from the links given and select the .zip and add that. Anyone think that's a bad compromise or think there's a better way to go about it?

nophead, 2015-01-24:
The missing libraries should be added, however is looks like all the liquid
ones do the same thing, so the best one should be used for all I2C LCDs.

NarimaanV, 2015-01-24:
Well I can confirm the included one is definitely not the best one because it only works for the RA Control Panel. Going to try the rest if the libraries to see if any single one would work for all of them.

NarimaanV, 2015-01-24:
Well bad news, each library only works with its respective display(s), so if we're adding all the required ones to the ArduinoAddons folder, we're adding all of them :/

nophead, 2015-01-24:
They should probably have better names then. I.e. the Liquid bit needs to
be replaced with the display type, or the controller type.

NarimaanV, 2015-01-24:
Yeah I was thinking something like "LiquidCrystal_I2C_RA_Conrol_Panel" for example. Does it affect compatibility if the folder name is changed at all?

whosawhatsis, 2015-01-25:
I haven't looked, are they really that different? The libraries may be incompatible, but I would still expect to see significant redundancy between them. I would think the best course of action would be to combine the functionality and add configuration options so that one library can serve in place of both.

NarimaanV, 2015-01-25:
Well that test was a preliminary one, just pointed out that no one library could be used for the others in terms of compilation, but I can take a look at them more closely tonight or tomorrow and see if it may just be a simple task of adding a few classes from one to the other or something like that. If it seems completely different I may leave it open for someone else to look if my programming knowledge doesn't seem to be enough.

NarimaanV, 2015-01-25:
Did a little comparison of the the three "Liquid" libraries, and the one for the RA Control Panel and the one for the Panelolu2/VIKI seem the most similar in structure and since they don't seem to repeat variables or functions, the simplest solution could be to just merge the two together (please correct me if that's a terrible idea). The library for Sainsmart...oh man, that's a whole other beast. Somehow uses about 10 files instead of the basic two that the others do, so something tells me that's going to be the least compatible with the rest. Also I'm not too familiar with and haven't had a chance to play around with I2C so does someone else want to take at least a preliminary look at the three of them and give your impressions too?

eboston, 2015-03-11:
It has been awhile since anyone has discussed this in this thread so hopefully someone will see this.  I have a question about the libraries and part of the discussion touched upon what I wanted to ask.  Latest versions of libraries, u8glib in particular.  I am playing around with a little project using the Adafruit ST7565 display and was planning on taking advantage of the fact U8glib is included/used.  However, the version used is older.  That version does work with the display, but is a lot slower than the current version.  If I was to add support for this display in, would it be likely it would ever get merged?  Or should I just do it for myself and not do a pull request for it?

thinkyhead, 2015-03-12:
@eboston I would like to see that. Whatever version of u8glib that Marlin works with needs to be bundled along with it. If there are changes that need to be made to the Marlin source, include that also in the same pull request.  I don't know how important it is for Marlin to be buildable using older versions of Arduino. I personally think it's fine for Marlin to require the latest version of Arduino, but there may be hardware (e.g., Brainwave?) that isn't yet supported in 1.5.x and later. If that's a big concern, we can possibly include both, and pick up the version number from something that is #defined inside the u8glib to distinguish which one is being used.

All that is to say, please give it a shot and make a PR. We like performance improvements!

eboston, 2015-03-12:
It's not the newer version of Arduino, but of U8glib.  I am using 1.7 but I think it would work with the version currently bundled with it.  I'll try it with that one when I have it all working.  Just dealing with an issue with the SD card right now.  Other than that, display and encoder is working.

[image]

thinkyhead, 2015-03-13:
I'd be curious to see some benchmarks, if possible, comparing speed between the old and new code. This is going to be especially exciting for users of deltas!

eboston, 2015-03-13:
I can make a video of the example program that comes with U8glib for graphics.  There was a mark difference is speed with that.  But because the display is not changing that much in this application, it might not be as noticeable.

Got the encoder and sd card working last night.  I'm wondering if this should get moved to it's own thread or is continuing it here fine?

eboston, 2015-03-14:
Was setting up to do the benchmark to show the speed difference but what I found was it wasn't the new U8glib that speed things up, but the new example program.  The new program uses some calls the old library doesn't support.  So, I'll go ahead and link to the old library since it doesn't look like there is a real performance difference to be gained at this time.

thinkyhead, 2015-03-14:
@eboston If it was going to go forward I would say start a new thread. Is there anything in the new example that sheds light on how we might optimize our use of u8glib, or is it just faster because it's better written?

eboston, 2015-03-14:
I would have to dig into the library to see what was changed added. But 
given the display on the printer does not change that much, it might not 
be of much benefit.  If I get some time this weekend, I will see if I 
can see what was added to speed things up.  I haven't looked at how the 
Marlin code uses the library so not sure how any changes to the library 
would help the firmware.

boelle, 2015-04-26:
just read the highlights and must say that lib's needed should be included... links may die... newest version might not work with marlin etc etc....

boelle, 2015-05-31:
closing this one... no activity and lib's needed to compile marlin should be included...

and say a maker adds support for a device or board, if there are libs required those should be included in the PR so that users should not have to worrie

closed 2015-05-31 by boelle