MarlinFirmware/Marlin #1389: How to EXTRUDER_OFFSET added EEPROM

linjinc, 2015-01-20:
#define EXTRUDER_OFFSET_X
#define EXTRUDER_OFFSET_Y

Added to the EEPROM

You can set the offset value on the lcd menu

Forgive my poor English, is translated by Google

I tried to modify Marlin_main.cpp, EXTRUDER_OFFSET_X ,EXTRUDER_OFFSET_Y. can find a place, into a variable, then you can set the LCD MENU, but to no avail

grob6000, 2015-01-21:
Do you change your extruder offset often?
At present, you should edit Configuration.h to define the extruder offsets for your printer, compile the firmware and upload to your board. The example in Configuration.h is for dual extruders, where extruder 1 is offset by +20mm in X and +5mm in Y:
```c++
// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
#define EXTRUDER_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
#define EXTRUDER_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis
```
Alternatively, you can have the slicer manage the extruder offset, instead of firmware: leave `#define EXTRUDER_OFFSET_X` and `#define EXTRUDER_OFFSET_Y` commented out (//), and set the offsets in your slicer (e.g. slic3r: http://manual.slic3r.org/expert-mode/multiple-extruders).

thinkyhead, 2015-01-23:
"The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder)."

There's some conversation (mostly me) about changing this, specifically to deal with the case of a delta with 3 extruders, but none of them in the center.

thinkyhead, 2015-01-23:
@grob6000 is correct. You can set this as a firmware option (for a second extruder) or set offsets in your slicer software (for any extruder), so no new feature is needed.

closed 2015-01-23 by thinkyhead