MarlinFirmware/Marlin #1392: Kraken still not supported with 4 extruders on an Azteeg X3 Pro?

Natealus, 2015-01-21:
I recently build a Rostock MAX v2 and I'm upgrading it with a 4 extruder Kraken and Azteeg X3 Pro. I'm actually kind of surprised that Marlin, being such a well known firmware, is still only setup for 3 extruders.

I read an issue from about 6 months ago where a guy went through and added in all the lines needed in temperature.cpp or .h to do 6 extruders. Can you guys merge that work into the current build? As it stands I'll have to set to 3 extruders just to compile.

MagoKimbra, 2015-01-22:
Hi, if you want try my build MarlinKimbra with 4th extruder and more feature...

Natealus, 2015-01-22:
Oh that's interesting I'll check it out. Are you working with this project and merging stuff here?

MagoKimbra, 2015-01-22:
I'm working on this project for a long time, I tried to merge with Marlin before it was moved to the current one, but have not been taken into account ... Unfortunately I can not speak English well and sometimes I struggle to explain. If they tell me to merge with the current I do it with pleasure. Currently there are several printers that use my firmware and go great, we are developing the use of the 4 colors for hours without fuse them together ... These are the results: http://www.thingiverse.com/thing:546387

thinkyhead, 2015-01-23:
@MagoKimbra I think Marlin would enjoy having the option of more extruders. There may be some debate among collaborators about how to deal with more axes in the future, but as long as your update follows the current methodology I don't see why it shouldn't be merged, especially with the Kraken and other multi-extruders gaining popularity.

MagoKimbra, 2015-01-23:
I am happy to join us, I do not know how to do it. If you tell me what to do I do it.

thinkyhead, 2015-01-23:
@MagoKimbra

1. Update the Development branch in your Marlin fork using `git fetch upstream`, `git checkout Development`, and `git merge upstream/Development` then sync back to your fork.
1. Create a new branch in your fork (named "more_extruders") from the Development branch.
1. Get the new branch using `git fetch origin`, `git checkout more_extruders`.
1. Make the changes you need to make, then __commit__ and __sync__ them back to your fork.
1. You should then be able to create a Pull Request from your changes.

MagoKimbra, 2015-01-23:
Ok thanks..
One question.. When i merge, i  must merge all fw or only modified for 4th extruder?
Sorry for my bad english!!!

thinkyhead, 2015-01-23:
If I understand your question correctly, when you create your pull request, it will only include differences between the current MarlinFirmware Development branch and the feature branch (i.e., more_extruders) that you cloned from Development (since it is the basis for your Pull Request).

This is a pretty big feature addition, and I look forward to seeing your results!

Natealus, 2015-01-31:
By the by all 4 extruders are heatable and temp reading well. I'm having a slight other issue of temp being offset by -25c so I set 260C to achieve 235C real temp but I was having that on release as well. Supposedly the screw in thermistors I got were supposed to be standard EPCOS...thought temp table 1 was the correct setup. Still working on getting the other 3 extruders.

thinkyhead, 2015-02-20:
Virtually everything for 4 extruders is in place now with the latest PRs. Closing this specific issue.

closed 2015-02-20 by thinkyhead

Natealus, 2015-02-24:
Yup I believe this issue is solved. Just more tuning to my machine is needed before I print something with 4 different colors.

thinkyhead, 2015-02-24:
@Natealus You are brave. I have enough issues with just one extruder...!

Natealus, 2015-03-04:
Yeah I'm still having some oddities with the delta movement. It cropped up when I printed a 150mmx150mm square with a 115mm cylinder on top of it. Getting a bowl curve movement on the spaces back close to Z and X...Z to Y. Seems like the calculations are off slightly if I'm getting almost1/8 inch lift in those areas. I've been tuning more and its getting better but still not perfect.

thinkyhead, 2015-03-04:
@Natealus Do you get any noticeable improvement if you print at a slower speed - like half? That sounds like it might be an artifact of not being able to get enough calculations per second.

Natealus, 2015-03-17:
Oh yeah that stuttering issue is mostly to do with circular movements cause they generate exponentially more G code inputs and delta movements than a straight line. Taking the segments down helps. But I fear movement resolution issues if you take it down too far. I haven't done any experiments on that particular question. 

I currently run at 165 delta segments per sec, 60 mm/s infill most often sometimes 80, 40 mm/s inner perimeters, and 20 mm/s outer perimeters. I seem to have pretty good results with these but on certain prints some further down tuning on prints with constant circular surfaces. I think if infills go too fast you can cause some rough edges on circles too from the nozzle mechanically going slightly farther than intended from speed lash.

This other issue has to do with how the firmware sees my print bed and I'm somewhat mystified how it can be so far off between X to Z and Z to Y. We're talking almost a quarter inch off the plate in those corners. I've done the 4 point calibration to death rechecking offsets and measurements and such. Its pretty good  on smaller parts in the middle. When I get closer to the edge it goes way off. I'm wondering if it has something to do with how I setup the extruder offsets. I did it in MatterControl maybe I have to change more in the firmware...but this bed leveling stuff might help.

thinkyhead, 2015-03-19:
@Natealus That delta math is a little mysterious to me. But it definitely matters to get the arm lengths and tower distances and other measurements spot-on.

Natealus, 2015-03-19:
We resquared the towers. X and Y were slightly off and as we read into delta kinematics it is very important for those to be completely parallel. I still have more retuning to do since it took the radius off. I updated Arduino IDE to 1.6.1 and all of a sudden I couldn't complile my old config. I decided to download a fresh copy of development to gather all the recent updates. Perhaps its something there? 

Recent download of Development verified and checked fine before I just started configuring. I'm going to try to do my pin adjustments and see if I can get it to see down the chain in the specific board pin file for variables already in the generic #include board.