MarlinFirmware/Marlin #1394: step_wait() in stepper.cpp needless

AnHardt, 2015-01-21:
When reading stepper.cpp i stumbled across the code beginning at line 241:

void step_wait(){
    for(int8_t i=0; i < 6; i++){
    }
}

A candidate what could be optimized away by the compiler.

Then i greped in Malin directory:

$ grep -r step_wait *
stepper.cpp:void step_wait(){

is not used anywhwere but the definition.

To be sure i commented out the function and compiled  with standard configuration.h and later with my configuration.h without errors.

I'm confident this can be removed.

monkeydave, 2015-01-21:
Looks like it was orphaned in this commit ff6fa09ecf188272761f8d4a08df9749a70c1d7e

@AnHardt Do you want to make a pull request to have it removed?

ErikZalm, 2015-01-21:
This delay was needed for slow stepper drivers.
We need to check if it is still needed. (I think we need some delay for the step pulse)
Maybe a smarter step pulse off position in the code is needed.
It is related to #1041

AnHardt, 2015-01-21:
@monkeydave Right now i tried my first Pull Request.

@EricZalm Maybe Tommy-LSA had no success because this simple useless loop was optimized out by the compiler. Similar to #1393 and #1386

arno-millenaar, 2015-01-21:
For the past year I have added a delay to extend the pulse for my external DM422C stepper drivers as follows without any problems.  The DM422C needs 10mS minimum pulse width for the Step signal and after checking with a Digital Storage Scope (DSO) I only needed to add about 7us to make it a 10us pulse.

So in the configuration.h file I define the following variable:

#define DELAY_WRITE_MICROSECONDS 7   // needed for DM422C, add code in stepper.cpp

And in the stepper.cpp I add a "dealyMicroseconds" command for X,Y,Z and E as follows:

Line 630 (X step):
""
counter_x -= current_block->step_event_count;
          count_position[X_AXIS]+=count_direction[X_AXIS];   
		  delayMicroseconds(DELAY_WRITE_MICROSECONDS);
#ifdef DUAL_X_CARRIAGE
""

Line 657 (Y step)"

""
counter_y -= current_block->step_event_count;
          count_position[Y_AXIS]+=count_direction[Y_AXIS];
          delayMicroseconds(DELAY_WRITE_MICROSECONDS);
          WRITE(Y_STEP_PIN, INVERT_Y_STEP_PIN);

Line 676 (Z step):

""
counter_z -= current_block->step_event_count;
        count_position[Z_AXIS]+=count_direction[Z_AXIS];
        delayMicroseconds(DELAY_WRITE_MICROSECONDS);
        WRITE(Z_STEP_PIN, INVERT_Z_STEP_PIN);
""

Line 690 (E step):

""
 if (counter_e > 0) {
          WRITE_E_STEP(!INVERT_E_STEP_PIN);
          counter_e -= current_block->step_event_count;
          count_position[E_AXIS]+=count_direction[E_AXIS];
          delayMicroseconds(DELAY_WRITE_MICROSECONDS);
          WRITE_E_STEP(INVERT_E_STEP_PIN);
        }
""

I can only testify that for the past 2 years this has worked without a problem.  What this also shows is that WITHOUT the above added delay the actual pulse is already between 2 and 3 us. Hence there may be no need fort adding any delay if your stepper driver will work with a one (1) us pulse width.

thinkyhead, 2015-01-23:
delayMicroseconds is _supposed_ to be accurate above 3µs, so presumably (?) it compensates for the call overhead. Anyway, I think we might want to try this, but make it optional:

    #if defined(DELAY_WRITE_MICROSECONDS) && DELAY_WRITE_MICROSECONDS > 0
      delayMicroseconds(DELAY_WRITE_MICROSECONDS);
    #endif

nophead, 2015-01-24:
You don't need to check if it is defined because undefined symbols in preprocessor expressions evaluate as 0.

#if DELAY_WRITE_MICROSECONDS

is sufficient.

thinkyhead, 2015-01-24:
Interesting feature of the compiler that it doesn't complain on undefined macros. I tend to favor the "defined" prefix in these preprocessor directives, if only because it reminds us that this #define can simply be commented out, and is not just set to 0 someplace. It also keeps open the possibility of using more strict compilers, which may help as we slide towards ARM support.

galexander1, 2015-01-24:
It doesn't really matter, there is a good style argument for #if defined,
either way. But just throwing out that it is not a gcc extension, it is
part of the standard.

ANSI C99 6.10.1 3
  ... After all replacements due to macro expansion and the defined unary
  operator have been performed, all remaining identifiers are replaced
  with the pp-number 0, ...

basically, all identifiers that don't correspond to a #define evaluate as
0 in #if.

thinkyhead, 2015-01-24:
I can sort of grok why ANSI decided to do it that way, but sometimes it would be nice if it would bark (as an option) so coders can clean up any unused identifiers.

nophead, 2015-01-24:
That is what Lint <http://en.wikipedia.org/wiki/Lint_%28software%29> is for.

I guess the reason ANSI defined it that way is because it is ugly having to
test every preprocessor value is defined before using it. Putting that in
the code just gives a false impression that it is needed, when in fact it
does nothing at all.

Also allowing it to be undefined makes code backwards compatibility easier.
You can add new options which default to off without everybody getting an
error and needing to explicitly set it to off.

thinkyhead, 2015-01-24:
Dangerous topic, because I'm the sort of obsessive geek who will go through all the code and remove all the extraneous defined() clauses… but… later… later… it can wait…

boelle, 2015-03-18:
any news on this one? just wanted to know if it can be closed or should be upgraded or?

AnHardt, 2015-03-18:
The original problem was cleared by #1395.

closed 2015-03-18 by AnHardt