MarlinFirmware/Marlin #1397: Bed "Topography" Correction

alexborro, 2015-01-22:
Hi guys,

I'm a little bit "tired" of BAL. It is nice, correct the plane of the bed but it is not enough.
Many folks are having issues related to non-flat bed and/or strange behavior on printers with moving bed (for Y axis) - if the Y plane is not perfectly orthogonal to X and Z, the BAL will not work properly.

So I'm thinking on mapping the bed topography and correct during printing - I guess Johann do something similar in his marlin branch for kossel. 

The Idea is probing like usual, let's say, a 3x3 grid and then split the bed in small rectangles with the probing points on corner. Then we split all movements in small segments fitting the rectangles and correct the Z axis in the rectangles corner. 

For example:  A 200x200mm bed probed on:
(0,0)    , (100,0)    , (200,0), 
(0,100), (100,100), (200,100), 
(0,200), (100,200), (200,200). 

(see attached image bellow)

[image]

We will have 100x100mm rectangles. If we have a move from (0,0) to (200,0) (yellow line on picture), we split in 2 moves (0,0)->(100,0) and (100,0)->(200,0) and correcting the Z at (0,0) , (100,0) , (200,0)... and so on..

I hope I was clear..

And I need suggestion and help doing that, specially for movements not passing for corners. 
Like, If we have a move (0,0)->(200,100) (green line on picture), it will become (0,0)->(100,50) and (100,50)->(200,100).
What height we are gonna apply on (100,50), since it is between (100,0) and (100,100) ?? Will interpolations on rectangle's borders work fine??

Thanks.

Alex.

nophead, 2015-01-22:
Please don't call it "correction" as it doesn't correct anything other than getting the first layer to stick. It makes bent objects.

alexborro, 2015-01-23:
The name is the last thing I will care.  And this feature is meant to correct small non-flatness bed. If one needs good FDM parts,  get a flat bed. If one needs perfect FDM parts, order a Stratasys.

I forgot to mention the crucial issue. It happens when we have one point inside the rectangle, like (50,50) in the give  example. How to interpolate the height?

galexander1, 2015-01-23:
The jcrocholl branch has a grid leveling like this, which uses linear
interpolation between sampled points.  It is for delta printers where a
small amount of spherical distortion is nearly inevitable.  It does
introduce some slight distortion in the other axes, but I calculated it
out and it is dismissable so long as the total error is less than about
1mm.

Using it on a Cartesian may be a bad idea, even if sometimes it's useful.
But on a delta, something like it is basically essential.

ntoff, 2015-01-24:
@nophead better than "bed auto levelling" which doesn't level anything at all.

thinkyhead, 2015-01-24:
Instead of moving the Z axis around to compensate, it would be interesting if (on the first layer) the extrusion could vary, putting down extra material in the regions where the bed is lower, and putting down slightly less material where the bed is higher. That would compensate for an uneven bed, but not for one that is askew.

whosawhatsis, 2015-01-24:
@thinkyhead I've been considering the same thing. It would not compensate for a platform that was significantly skewed, but you wouldn't want to do it in that case anyway because it would result in a surface at the wrong angle. This option would be mostly useful when using a raft, and would almost never be acceptable when printing a raftless part flat on the build plate. When you're printing an object that's mostly sitting on top of supports (like when I print stuff oriented diagonally to improve the strength and/or surface finish), it would be less of a problem than tilting the print to match the angle of the platform.

nophead, 2015-01-24:
It shouldn't be called "correction" because it doesn't fix a problem. A
curved bed will produce a curved object. Maybe bed topography adaptation.

On the other had bed auto levelling completely eliminates the need to level
the bed relative to the axes and produces objects exactly the same as if
you had. It doesn't fix any other problem, such as axes not being
orthogonal. I designed Mendel90 to address that.

Perhaps BAL should be called bed inclination correction but I think it is
too late to change it now as it has been called BAL for many years and is
spread all over the web.

orangeudav, 2015-02-12:
I think, in some cases, we can "correct" not only first layer, but second, third etc if distortion is greater than layer thickness. But we should use (linear, for example) interpolation during upward movement along the z axis. For example, on 100% correction for 1st layer, 66% correction for 2nd layer, 33% for 3rd.
[image]

galexander1, 2015-02-12:
@orangeudav Correcting for a truly warped bed is a bad idea, and
cluttering marlin to compensate for it is also a bad idea.  The real
purpose of this feature is for geometries like delta where spherical
distortion is a nearly inevitable result from slight miscalibration.
In the case of delta, the spherical distortion is an illusion (the bed is
actually flat), and a naive Z correction can compensate effectively.

brainscan, 2015-02-12:
I think this is a great idea, not sure why some are so against it. I'd rather have this and get decent prints every time, I think some people are imagining a warped bed to be like a cereal bowl. I have a pretty warped bed and can't get a decent first layer without bed levelling but all my prints are still fine, pieces fit together and I don't have any stress getting the first layer to stick. If memory became a problem I'd rather remove other things that are cluttering marlin up.

galexander1, 2015-02-12:
@brainscan But you can get a good first layer with bed levelling?  What
bed levelling do you use to get a good first layer?

brainscan, 2015-02-12:
Roxy modified the code and added a few extra commands to it, before that I could only print in the middle of the bed or the warp from corner to corner would cause areas not to stick properly. Now I can print over the entire bed with an even and well adhered first layer every time. The smaller the nozzle the more the warp shows up. If I want ultra precise parts I'd probably have to spend a lot more on other things too not just a perfectly flat bed. As it is everything I've printed that's supposed to fit together does so just as well with bed levelling as without it but every print works now as opposed to 1 in 4.

Roxy-3DPrintBoard, 2015-03-14:
### thinkyhead commented on Jan 24
### Instead of moving the Z axis around to compensate, it would be interesting if (on the first layer) 
### the extrusion could vary, putting down extra material in the regions where the bed is lower, 
### and putting down slightly less material where the bed is higher. That would compensate for 
### an uneven bed, but not for one that is askew.

This is a very interesting idea!  I guess we should wait until the code base stabilizes with all the bug fixing first.   But this is something I would like to try!    Probably, it would make sense to be able to specify over how many layers to back off to the normal case.  I am suggesting not just doing it for the first layer.  For example, maybe the first layer you put down extra material to fill in the low spots.   But on the second layer you put down less extra material.  And maybe on the fourth layer you are not doing anything extra?

This might be worth some discussion because a lot of us have non-flat beds.  And certainly, it could be implemented as an option.

boelle, 2015-05-31:
will close this one... we should not waste time making features that correct an hardware problem.... if the bed is not flat that is not for us to thinker about... the user should get a flat bed

and if the frame is not right that is not our problem...

depperson, 2017-05-18:
does UBL cover this in recent builds?

Roxy-3D, 2017-05-18:
> does UBL cover this in recent builds?

I had to re-read the tread to be able to answer.  But the answer is "Yes!"  I didn't realize the optimizations I put into ubl_motion.cpp came from @alexborro 's thoughts.  But it is all in this post .   And his thinking forms the roots for ubl_motion.   And the Fade stuff....  That is all in ubl_motion (and fully supported in G29 F) and it came from @a0s in the above post https://github.com/MarlinFirmware/Marlin/issues/1397#issuecomment-74062400

closed 2017-05-19 by Roxy-3D