MarlinFirmware/Marlin #1398: Extruder not being driven on RAMPS 1.4 after 1 or 2 prints

KenRip, 2015-01-23:
I have a MakerFarm 8" Prusa i3v printer with the RAMPS 1.4 driver board.  I had been using the printer for many months with the original version of Marlin provided by MakerFarm (not sure what version number but the date is 3/24/2014).  This version was quite dated and didn't support the newer features so I worked to tailor the V1.0.2 Marlin FW (downloaded on 1/17/2015) and loaded this on my printer.  Everything works perfectly initially but after one or two prints the printer will no longer drive the stepper motor on the extruder (either trying to print from the SD card or when using Pronterface to drive the extruder manually).  The only way to restore the extruder to a functioning state is to use the "Restore failsafe" function from the LCD even though I have not made any changes to the settings in the EEPROM.

HackerTon, 2015-01-23:
Try to fully disable EEPROM.

CONSULitAS, 2015-01-23:
@KenRip Please remember to send M502 followed by M500 (see http://reprap.org/wiki/G-code) manually after you have updated an old firmware. It should update EEPROM settings automatically because of changings in layout, but sometimes it does not.

I had the same problem upgrading from EEPROM V07 to V14 (up to date). This was the only way that helped.

BTW: One other possible symptom I had is insane X/Y speed while traveling which gives many missed steps and could even damage not so rigid hardware. :-(

So the symptoms of misaligned EEPROM to firmware versions can be surprising and depend on many factors.

See last entries in #1209 and whole #1388 for background information.

If this helps, please don't forget to close this issue. If not, please provide further information. Thanks.

CONSULitAS, 2015-01-23:
@HackerTon This would surely help, but it would be better to solve the underlying problem @KenRip has.

It sounds a litte bit like prohibiting cars because you have no drivers license. :-)

KenRip, 2015-01-23:
I performed the "Restore failsafe" (equivalent to M502) followed by the "Store memory" (equivalent to M500) functions from the LCD and will continue testing.  So far the issue has not reoccurred.

KenRip, 2015-01-23:
From a feature request perspective...  is there anyway for the Marlin FW code to detect that it's the first time it's being executed after a FW upgrade and perform the M502 and M500 functions automatically?  Another idea would be to store the Marlin version in the EEPROM and check that the code version and version stored in EEPROM match.  If they don't perform the functions to correct it.

thinkyhead, 2015-01-23:
It's supposed to notice the version change and do the right thing, but there seems to be a possible glitch. All current EEPROM woes should be corrected pretty soon in the Development branch.

KenRip, 2015-01-25:
After reading the existing issue "Reset EEPROM when the layout changes #1388" and thinkyhead's comment above I'm closing this issue.  My problem is related to that existing issue and should be resolved with it.

closed 2015-01-25 by KenRip