MarlinFirmware/Marlin #1406: Add post unretract delay (new feature)

wgm4321, 2015-01-24:
Add an adjustable delay of 1-500ms that occurs after unretracting to provide a pause to allow the material to flow some.

boelle, 2015-04-26:
how many would find this usefull?

nophead, 2015-04-26:
You can get the same effect by slowing down the unretract.

thinkyhead, 2015-04-27:
`M208 F`

whosawhatsis, 2015-04-27:
Yeah, the whole idea is to do the unretract closer to the printing extrusion speed so that pressure can equalize before printing resumes.

nophead, 2015-04-27:
One thing I noticed recently: even if you print infill and outlines with the same flow rate, it needs more pressure to print solid infill due to the filament being constrained at one side and the last infill line is constrained on both sides.  This leads to too much flow at the start of the next outline. Very difficult to compensate for even with advanced flow control because the pressure is not simply proportional to the flow rate. It depends on the the extra back pressure caused by forcing it into its destination. You actually need it bit more retraction after infill and even more after solid infill.

wgm4321, 2015-05-01:
I'm not sure that slowing down the unretract is exactly the same.   While slowing down the unretract allows the material to come back to temp, pausing after unretract allows the pressure to equalized with the material at the right temp so the farthest part of the nozzle is primed up and at the right pressure to resume a consistent extrude.   So I think using both allows you to dial in the unretract for the best quality.

thinkyhead, 2015-05-02:
The E_JERK limit may have some noticeable effect on retract / unretract too... but it might be too small to worry about.

boelle, 2015-05-31:
closing this one as it seems only 1 have interest and that there have been provided alternative ways

closed 2015-05-31 by boelle