MarlinFirmware/Marlin #1411: Filament diameter adjustment on graphic controller

clefranc, 2015-01-24:
If I set the filament diameter using M200 D1.75, then I go to the Filament menu, I get

Fil. Dia. 1: 1.750

If I try to change the value using the graphic controller, it only adjust from 2.500 to 3.500, we should be able to go down as low as 1.250, at least 1.500.

clefranc, 2015-01-26:
The solution was simple, but no so practical, just change some values in ultralcd.cpp.

I think the source of the problem rely on the filament sensor identifier. It's used many times at many places to calculate filament size.

I rewrote some codes to correct this, can someone look at it and tell me if it worth a commit?

https://github.com/clefranc/Marlin

thinkyhead, 2015-01-26:
It looks great to me. I'm not sure what the consensus is on the default filament diameter, which has historically been 3mm, and maybe this is because historically Marlin's defaults are intended to work almost out-of-the-box on an Ultimaker 1. Anyway, your changes seem to be needed, so please go ahead and make a pull request.

whosawhatsis, 2015-01-26:
The default definitely shouldn't be 3mm. Any reputable filament supplier will sell "3mm" filament that actually has a nominal diameter closer to 2.85mm, so that would be a better default value, assuming we don't go with 1.75mm.

thinkyhead, 2015-01-26:
1.75 and 3.00 are pretty idealistic, I agree. But in fact most of the filament I have had is pretty close to 3mm, and with die swell as a factor, a slightly higher estimate of diameter might be a Good Thing™.

nophead, 2015-01-26:
Die swell doesn't alter filament volume. It simply means the filament comes out the nozzle bigger than its aperture but correspondingly shorter, when extruded into free space.

But yes my ABS filament is closer the 3mm than 2.85.

thinkyhead, 2015-01-26:
I guess it's just a s'well. All'swell that end'swell…

clefranc, 2015-01-26:
No problem with 3mm as default.

On the other hand, I just saw this issue https://github.com/MarlinFirmware/Marlin/issues/1400
about support for a 4th extruder? Do I need to add another identifier? It also means that many LCD menu need to be updated for this new extruder. A challenge to put this new information on the info screen.

Finally, is the new identifiers should be replaced by an array, instead of unique variables?

clefranc, 2015-01-27:
Another question, what should the range of the filament diameters in the LCD menu and M200 command?

1.25mm to 3.5mm (I vote for this one)
1.0mm to 4.0mm
???

I closed my PR because of this, I want to submit something that everyone will agree with.

thinkyhead, 2015-01-27:
I don't mind 1.25 to 3.5. Vendors will want to configure the range, so as long as there are #defines for that, the full range makes the best RepRap default.

wgm4321, 2015-02-15:
The filament diameter is taken from 
#define DEFAULT_NOMINAL_FILAMENT_DIA  
in configuration.h
The issue is if you hard code the range from 1.25 to 3.5 or something equivalent, the number of steps between the numbers becomes huge.
Then the menu code does:
MENU_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_SIZE_EXTRUDER_0, &filament_size[0], DEFAULT_NOMINAL_FILAMENT_DIA - .5, DEFAULT_NOMINAL_FILAMENT_DIA + .5, calculate_volumetric_multipliers);
to keep the range of numbers and steps reasonable so you don't have to spin through hundreds of steps if you have 2.85 and the default is 1.25

wgm4321, 2015-02-15:
So I think if you set DEFAULT_NOMINAL_FILAMENT_DIA to the correct value, this is not a bug/defect.

thinkyhead, 2015-02-16:
We've also got a PR pending to allow different diameters per-extruder. For now we'll assume these can be set in firmware, we won't need the full range, and small steps are good because filament shouldn't vary that much.

closed 2015-02-16 by thinkyhead

wgm4321, 2015-02-16:
One general, welcomed improvement that might help this issue would be some mechanism to increase the increment based on the speed of the knob or duration of the "spin".    Then you can have fine grain control but be able to navigate a large range.

For example, if you have used the controller to turn off the fan, you know it's inconvenient to go from 255 to 0.    I was going to implement something like that but wasn't sure how to about implementing it.
    
The controller code for input really needs to be refactored due to all the variations of xx.xxxx now required. 

Maybe it is a simple as if ten increments have be covered in the past sec, increase the increment value by 10x and possibly again to 100x.

Hmmmmmm.....

thinkyhead, 2015-02-19:
@wgm4321 The encoder code is interesting. I would like to see it change values on every "click" for fine-tuning, then if multiple "clicks" occur, it could scale in some way - only for settings that would benefit, such as the fan speed. Worth playing around with that, and maybe a useful patch can come out of it.

wgm4321, 2015-02-19:
@thinkyhead 
Actually I had code that did just that.   Went to merge upstream changes so I could submit a clean pull request and somehow lost my changes.
I'll redo tonight and submit.    
Basically, the steps per sec of the wheel was calculated and if greater than 70 (yes, sounds high but it wasn't for me), the steps moved was multiplied by 10 before being sent to the menu edit routine.   If it was 110 or higher, 100x was used.    
The multiplier code was surrounded by a check that disabled it unless the item was specially declared as benefiting from the multiplier.
I then flipped to feature on for the filament size and set the range to 1.5 - 3.5 and it worked well.
I enabled it on the fan and temp controls and was satisfied with how it work.
I'll put it in my fork once I get it up to date and people can try it out and provide feedback.

thinkyhead, 2015-02-20:
I haven't re-flashed my 3D printer in a while, but that's a change I would definitely want to incorporate as soon as possible.

wgm4321, 2015-02-20:
I have my changes in my fork:
https://github.com/wgm4321/Marlin/commit/65e1fc71be81406e58c0fa2bea324c504192afb6
In addition to the changes, I changed the filament diameter range from 1.5 - 3.25

wgm4321, 2015-02-20:
Had to fix some compile errors when  it was disabled.   See the commit above and this one:
https://github.com/wgm4321/Marlin/commit/b8e79dce897c970afb9c7721f93943046b1410b9

thinkyhead, 2015-02-20:
Messing with the roundabout ways in github to add commits to active PRs. Seems to work!