MarlinFirmware/Marlin #1412: Break apart and clean up pins.h

thinkyhead, 2015-01-25:
To make the Marlin root folder even more interesting (and to aid future development) I'm in the process of going through pins.h and breaking it up into separate files based on MOTHERBOARD name. Whether or not this produces a better final result is yet to be seen, but it's a good exercise to get a handle on the slightly messy state of pin assignments. It may end up making it easier to maintain pins and add new board support in the future, but for now it's more of an academic exercise to see whether pins files can be more consistently structured, to eliminate redundancies, and to make it much easier to find the pins you're looking for.

alexborro, 2015-01-25:
Think about that idea of a abstraction layer so we can associate a function
to a specific device.

Cheers.

Alex.

thinkyhead, 2015-01-26:
First step was to split up the pins into separate files and clean them up. The result looks pretty good. The abstraction layer should be easier to conceive given this improvement. #1416

closed 2015-01-28 by thinkyhead