MarlinFirmware/Marlin #1414: Motor of Extruder not working with Marlin development but full work with Stable version!

nahci13, 2015-01-25:
Hello all
After firmware upgrade to development version, first all motors not very good working.
MAX_FEEDRATE for Motors of Z axis need reconfiguration:

> //// MOVEMENT SETTINGS
> #define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
> #define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)

> // default settings

> / /#define DEFAULT_AXIS_STEPS_PER_UNIT   {78.7402,78.7402,200.0*8/3,760*1.1}  // default steps per unit for Ultimaker
#define DEFAULT_AXIS_STEPS_PER_UNIT   {79.5, 79.5, 4000, 838}
>#define DEFAULT_**MAX_FEEDRATE**          {500, 500, **1**, 25}    // (mm/sec)
>#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}    // X, Y, Z, E maximum  start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.

> #define DEFAULT_ACCELERATION          300    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
> #define DEFAULT_RETRACT_ACCELERATION  300   // X, Y, Z and E max acceleration in mm/s^2 for retracts

And now very good Z axis moving, same stable version.
And than for test printing motor of Extruder complete not working!??
Good luck

thinkyhead, 2015-01-26:
Hi @nahci13 would you like to help by being a tester? If so, I will add you to the list on #1209.

What kind of hardware do you have?

pixel3Design, 2015-01-26:
same here. Extruder stepper is not working.
Prusa i3 with Arduino 2560, RAMPS 1.4, StepStick Drivers.

orangeudav, 2015-01-26:
Motor not working when EEPROM_SETTINGS enabled.

CONSULitAS, 2015-01-26:
@nahci13 @pixel3Design @orangeudav 

Please remember to send M502 followed by M500 (see http://reprap.org/wiki/G-code) manually after you have updated an old firmware. It should update EEPROM settings automatically because of changings in layout, but sometimes it does not.

I had the same problem upgrading from EEPROM V07 to V14 (up to date). This was the only way that helped.

BTW: One other possible symptom I had is insane X/Y speed while traveling which gives many missed steps and could even damage not so rigid hardware. :-(

So the symptoms of misaligned EEPROM to firmware versions can be surprising and depend on many factors.

See last entries in #1209 and whole #1388 for background information.

If this helps, please don't forget to close this issue. If not, please provide further information. Thanks.

nahci13, 2015-01-26:
Thank you @thinkyhead not now!
My Hardware same @pixel3Design and with LCD ReprapDiscount.

@orangeudav Thank you! now working! :+1:

thinkyhead, 2015-01-26:
Great! We will patch ConfigurationStore soon to work around this issue in the next couple of days.

closed 2015-01-26 by thinkyhead

orangeudav, 2015-01-27:
For 8c792c7b22bd612c2e8b38ca8b23de936398a14b commands M502 -> M500 -> M501 not resolve this issue. Only EEPROM_SETTINGS disable ..

thinkyhead, 2015-01-28:
#1428 is designed to make sure EEPROM is consistent. Once it's been merged, give it a test.