MarlinFirmware/Marlin #1415: Viki LCD won't show SD card contents after removing and reinserting card

bryanboettcher, 2015-01-25:
When my machine (Makergear M2, RAMBO 1.4, Viki LCD) is powered on, the SD card works as expected.  Going to the "Print from SD card" menu displays the filesystem and printing works normally.

Upon removing the MicroSD card and transferring GCODE to it, the firmware is unaware of its reinsertion.  The display will change from "No SD card" to "Print from SD card", but the filesystem will not show up at all.  After resetting the machine, the SD card behaves properly again.

I am properly unmounting the SD card, and my workflow works on other machines with different LCD controlers (like the RepRapDiscount one with RAMPS).

thinkyhead, 2015-01-26:
I have had this issue occasionally also. I discovered in my case that it helped to tighten the connection between the LCD/SD unit and the board. Once the plugs were nice and tight the problem went away.

bryanboettcher, 2015-02-03:
This is definitely a firmware problem.  I had a printer that could read SD cards flawlessly, then stopped as soon as I changed the firmware to a newer version of Marlin.

thinkyhead, 2015-02-03:
Have you tried an earlier version of Marlin? If you can find an earlier iteration that works, we can compare and see what has changed.

bryanboettcher, 2015-02-03:
I did try an earlier version of Marlin, it works as expected.  It was a
modified version posted by MakerGear --
http://makergear.wdfiles.com/local--files/m2-lcd-panel/M2_LCD_Firmware_24Vw24Vheater.zip

I can't tell what version of Marlin it is, but it's before the switch to
.INO extension.  I tried setting the values of SDCARDDETECT to the other
values I found in pins.h (on the new Marlin), to no effect.

thinkyhead, 2015-02-03:
Hmm, well I know it's a hassle to re-flash, but if you want to keep helping to track this down, Marlin now has [two tagged versions](https://github.com/MarlinFirmware/Marlin/tags), [1.0.1](https://github.com/MarlinFirmware/Marlin/tags/1.0.1) and [1.0.2](https://github.com/MarlinFirmware/Marlin/tags/1.0.2).

If you find that one of these works for you, then it will be an easier starting-point to see what's involved, since the tagged versions are only a couple months old and have fewer changes. If the SD card still flaky in these earlier versions, I can start asking around about SPI code and see what has changed in those regions over the past year or so.

bryanboettcher, 2015-02-04:
I chased the problem down to fastio.h.  Attached are the good & bad
versions, but these are referenced from an old version of fastio.h
-- 62aab66299cc7942a9b3559474eeba7047528bcd

I go pretty cross-eyed trying to diagnose what's going on in here though.

/*
  This code contibuted by Triffid_Hunter and modified by Kliment
  why double up on these macros? see http://gcc.gnu.org/onlinedocs/cpp/Stringification.html
*/

#ifndef	_FASTIO_ARDUINO_H
#define	_FASTIO_ARDUINO_H

#include <avr/io.h>

/*
  utility functions
*/

#ifndef MASK
/// MASKING- returns \f$2^PIN\f$
#define MASK(PIN)  (1 << PIN)
#endif

/*
  magic I/O routines
  now you can simply SET_OUTPUT(STEP); WRITE(STEP, 1); WRITE(STEP, 0);
*/

/// Read a pin
#define _READ(IO) ((bool)(DIO ## IO ## _RPORT & MASK(DIO ## IO ## _PIN)))
/// write to a pin
// On some boards pins > 0x100 are used. These are not converted to atomic actions. An critical section is needed.

#define _WRITE_NC(IO, v)  do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)

#define _WRITE_C(IO, v)   do { if (v) { \
                                         CRITICAL_SECTION_START; \
                                         {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); }\
                                         CRITICAL_SECTION_END; \
                                       }\
                                       else {\
                                         CRITICAL_SECTION_START; \
                                         {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }\
                                         CRITICAL_SECTION_END; \
                                       }\
                                     }\
                                     while (0)

#define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

/// toggle a pin
#define _TOGGLE(IO)  do {DIO ##  IO ## _RPORT = MASK(DIO ## IO ## _PIN); } while (0)

/// set pin as input
#define	_SET_INPUT(IO) do {DIO ##  IO ## _DDR &= ~MASK(DIO ## IO ## _PIN); } while (0)
/// set pin as output
#define	_SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)

/// check if pin is an input
#define	_GET_INPUT(IO)  ((DIO ## IO ## _DDR & MASK(DIO ## IO ## _PIN)) == 0)
/// check if pin is an output
#define	_GET_OUTPUT(IO)  ((DIO ## IO ## _DDR & MASK(DIO ## IO ## _PIN)) != 0)

/// check if pin is an timer
#define	_GET_TIMER(IO)  ((DIO ## IO ## _PWM)

//  why double up on these macros? see http://gcc.gnu.org/onlinedocs/cpp/Stringification.html

[... 2814 lines cut ...]
#define	DIO31_DDR		DDRC
#define DIO31_PWM		NULL

#define	DIO32_PIN		PINC2
#define	DIO32_RPORT	PINC
#define	DIO32_WPORT	PORTC
#define	DIO32_DDR		DDRC
#define DIO32_PWM		NULL

#define	DIO33_PIN		PINC3
#define	DIO33_RPORT	PINC
#define	DIO33_WPORT	PORTC
#define	DIO33_DDR		DDRC
#define DIO33_PWM		NULL

#define	DIO34_PIN		PINC4
#define	DIO34_RPORT	PINC
#define	DIO34_WPORT	PORTC
#define	DIO34_DDR		DDRC
#define DIO34_PWM		NULL

#define	DIO35_PIN		PINC5
#define	DIO35_RPORT	PINC
#define	DIO35_WPORT	PORTC
#define	DIO35_DD

thinkyhead, 2015-02-05:
@bryanboettcher It looks like github has a limit on comment size, so unfortunately only the first file, (up to line 2918) was pasted into your comment. But are you saying that the commit you cite – https://github.com/MarlinFirmware/Marlin/commit/62aab66299cc7942a9b3559474eeba7047528bcd – is the one that broke SD remounting? I looked it over, and I see that it includes changes from two earlier commits, both adding pins definitions for RAMBo LCD. Commit for `pins.h`: https://github.com/MarlinFirmware/Marlin/commit/53614e51442adaa6f254f8323e2fc0449e9b63dd. Commit for `fastio.h`: https://github.com/MarlinFirmware/Marlin/commit/b42d24e31ef0060659236da1787bee87682fa68b.

So if you simply remove these pins definitions from your `pins_RAMBO.h` and `fastio.h` files, your SD card reinsertion then works ok?

boelle, 2015-03-11:
is this one still active?

for big code use something like pastebin.com

bryanboettcher, 2015-03-11:
It's still an issue, I will get the diff and post on pastebin tonight or
tomorrow night.

boelle, 2015-03-18:
forgotten us? :-D

bryanboettcher, 2015-03-18:
I haven't, but I don't have the time to go back and re-build my patch.
Sorry guys :(

boelle, 2015-03-18:
bummer :-( well i will close this one then... it can still be read by everyone and if another user should have the same issue and see this one then please open a new issue

closed 2015-03-18 by boelle