MarlinFirmware/Marlin #1422: [Feat Req] Zprobe switch on alternate ping

luismanson, 2015-01-26:
Hello, i would like to have a setting to choose other input (than mind-endstop) for z-probing, using the min switch works, but now we also depend on a working servo. 

Right now im having an issue where the head is moving under z=0, which almost breaks my bed twice. Since the servo arm is retraced (not sure why) the machine did not get noticed about this.

Repetier has Z_PROBE_PIN option.

MagoKimbra, 2015-01-27:
If you have the endstop in NC, I put them in series, one of the servo and the normal endstop of Z. If the servo is not working, the normal endstop switch and open contact if the nozzle drops too. I adjusted a little bit less than 0 ...

luismanson, 2015-01-27:
That's something I already thought, yet a bit reluctant to make this,
because my endstops are on a board with leds...(I'll have to check the
schematics)
I proposed this, since must of us, at least the ones with RAMPS only use
min or max endstops.

Another idea was to use this endstop as an alternate killswitch

avluis, 2015-02-13:
Was just thinking about this when setting up my auto leveling today.
Thinking about this, I have yet to remove the z endstop either so it does make sense to have it as a fallback.
I think it would compliment Z_SAFE_HOMING as well.

boelle, 2015-05-31:
closing this one... marlin has a pin for Z probe

  #define Z_PROBE_PIN

closed 2015-05-31 by boelle