MarlinFirmware/Marlin #1435: Stem the tide of new "boards"

thinkyhead, 2015-01-29:
Since I recently split up `pins.h` to make it clearer how to add new board support, we are starting to get an influx of new hardware support. The most recent is #1433 - the "Hephestos" printer from BQ. There are two obvious problems that I want to nip in the bud:

- Multiplication of `pins_MYBOARD.h` files cluttering up the root folder.
- Multiplication of `example_configurations` files that we have to keep in sync with the current `Configuration.h` and `Configuration_adv.h` files.

The solution to the first problem is simple enough. For machines that are equivalent to RAMPS 1.3-1.4 we should combine them in pins.h and drop (most) `pins_MYBOARD.h` files that are no more than aliases for RAMPS_13. Then, allow the `CUSTOM_MENDEL_NAME` to override any `MACHINE_NAME` (not just unnamed ones). Then only an example_configurations entry _for the specific machine_ is required.

For the second issue, perhaps we should only provide configuration overrides instead of full Configuration.h and Configuration_adv.h files, or by some other method reduce the need to sync `Configuration.h` and `Configuration_adv.h` every time they change. Perhaps we can automate the process of applying overrides to produce these files with the settings in-place.

Any other suggestions and feedback are welcome.

blubbfish, 2015-01-31:
Maybe you can make 
```
#ifndef max_temp
  #define max_temp = 100
#endif
```
in `Configuration.h` and in `/sample_conf/example/Configuration.h` you can make
```
#define max_temp = 200
```
and if you want to use a sampleconfig you only must include that file to `Configuration.h`

CONSULitAS, 2015-01-31:
@thinkyhead Hi Scott,

we should try to convince the Arduino Project to support subfolders in the sketch folder: :-)
https://github.com/arduino/Arduino/issues/1004
https://github.com/sudar/Arduino-Makefile/issues/251
https://github.com/sudar/Arduino-Makefile/issues/249

So wie would have a much more readable root folder. The mess of files in one folder has been mentioned in many discussions before.

This would not help with keeping files in sync, but with getting a more clear structure.

BTW: I think, that the "influx of new hardware support" is driven by the new dynamics in the project started by @boelle before christmas. It inspired me to port the config for the K8200 back to the current dev branch. It seems that there are many "hidden" branches of Marlin around which now merge back step by step to the main branch. For me this is a good way.

Perhaps should we change the structure of the configuration files a bit to reflect this:
* One very short init file to select the configuration: Configuration.h
  * only select board, mechanics, options
  * import file
* One configuration file to reflect the board: e.g. board_xxx.h / Configuration_board_xxx.h
  * absorbs pins - see also comment of @bjmckenz in #1416 
* One configuration file to reflect the mechanics: e.g. hardware_K8200.h / Configuration_hardware_K8200.h
  * geometry of bed and y
  * steps per mm
  * endstops
  * Display / LCD
  * ...
* One configuration file to reflect the software features options: e.g. options_stable.h / Configuration_options_stable.h
  * custom gcodes
  * firmware retract
  * experimental features
  * ...

Reasons:
* Some boards are used in more than one printer (e.g. 3Dreag and K8200)
* Mechanics are specific to the printer itself but could be changed by users (e.g. Z spindle)
* Software features are a thing of individual taste and how much the user likes the adventure of experimental features. They are mostly independent of hardware. So we could deliver a "safe and stable" and an "all in" version and make the "safe and stable" default.

Cheers Jochen

bkubicek, 2015-01-31:
> arduino/Arduino#1004 <https://github.com/arduino/Arduino/issues/1004>
> sudar/Arduino-Makefile#251
> sudar/Arduino-Makefile#249

This is no solution, it would break compatibility with older arduino
versions.

thinkyhead, 2015-02-01:
@CONSULitAS I have to agree with @bkubicek that until the Arduino project deprecates their old tools we're pretty much stuck to making ugly file structures like this one.

@blubbfish I think we'll probably end up going in that direction, or something like it. At some point we could look at making and including a simple Javascript configurator app that runs in the browser.

thinkyhead, 2015-02-01:
@CONSULitAS There has been some discussion in the past about breaking up `Configuration.h` into some bits by feature, for example, `Configuration_delta.h`, etc. But yes, organizing by machine makes some more sense. We have different categories of configuration too: Geometry, Speed, Electronics, etc. and these combine to give you a certain Machine. I'm not totally averse to having – for the time being – just the two Configuration files and lots of `example_configurations` - if we can simplify the process of managing them. If it were possible to do sub-folders it would make more sense to break up configs by board type.... A lot to mess with here!

ntoff, 2015-02-06:
Isn't the web based configurator basically what the repetier firmware has? You configure your stuff on their web page, and download the firmware.

thinkyhead, 2015-02-06:
@ntoff This is true. When you have a web server back-end at your disposal you can hide all the ugly parts of the build process. We do have some offers of web space, and while I'm crossing my fingers that we will have a server where we can run AVR Tools, I'm not counting on it yet. Anyway, while building Marlin with Arduino tools isn't a picnic, it's not too far removed from pressing a single button – once you figure out your configuration options and get all your libraries installed and....

I totally support having a web-hosted piece that does the automatic builds and makes a binary download. One of the more boring parts has always been figuring out configuration options, so it just makes sense to provide [a tool for that](https://github.com/MarlinFirmware/Marlin/pull/1467) as part of the distro. It's better than just documentation, and it provides a reference for all the legal values and so on. (Trying hard not to end up going into XML and DTDs and all that, but the configuration options can actually be expressed that way. Sticking to JSON for now....)

CONSULitAS, 2015-02-07:
@thinkyhead Thought just a bit further about it:

If we can't change the problem itself, we should perhaps change the process:

* Marlin itself just supports the boards via boards.h and pins.h - so we have only keep track of new electronics

Up to here no change.
* For the support of machines (combination of boards and peripherals like LCDs, geometry, speed and so on) we keep the example_configurations/ folder with pairs of Configuration.h and Configuration_adv.h
* The Marlin project itself would not support the examples.
* For keeping them in sync we have for exery machine (found as subfolder) a godfather / sponsor who is responsible for keeping it in sync
* The sponsor could be the hardware vendor itself or an enthusiastic with this machine, who is interested in keeping his printer up to date.
* Every machine folder should have its readme.md naming the sponsor and information of last update / last working date or version to make it easier to communicate about broken support.
* A readme.md in the examples folder top should describe the process and how to deal with problems.

What do you think about it?

* I could be the sponsor for K8200 (what I do in fact now).
* I could write the first readme.md, too and do a pull request.

btw: Even with a configruation tool there should be someone to test the default configs for the machines to work after updates and we should have a place for providing working configs for not so experienced users.

cheers Jochen

thinkyhead, 2015-02-07:
@CONSULitAS

All very good suggestions. I'd like it if the Marlin project can officially _not support_ the examples and _not keep track_ of vendor hardware, and leave this up to sponsors. It would be nice if Marlin could focus exclusively on RepRaps, but I think it's the right thing to have configurations for machines that might come with their own firmware but which can also work with Marlin. I was about to say it might be hard to find sponsors, but of course people are very enthusiastic about their personal robots, so yeah, I think we can start a topic for that and maybe hit up people on the TESTERS thread also.

thinkyhead, 2015-02-20:
Ongoing as new boards and machines come along. @CONSULitAS whenever you have a readme.md post a PR and we'll get more feedback, me-hopes.

closed 2015-02-20 by thinkyhead

CONSULitAS, 2015-05-20:
@thinkyhead

I read in an other issue that there will be example files removed, but it seems that the community likes them. See the new #2127 .

Perhaps we should devide them to supported and unsupported ones?

the supported ones
* must have a sponsor/"care taker"
* there must be a Readme.md
* they must be updated by the care taker
* if they get to outdated, they will be moved to unsupported

the unsupported ones
* will be delivered as is
* get a special folder
* there will be a Readme.md explaining their state in this folder

I would write the readme files if this is common sense.

For a first step I gave @Wackerbarth hints in his PR, hoping he will be the caretaker for this in future. :smile:

Wackerbarth, 2015-05-20:
I feel that it is important to have “out-of-the-box” configurations available for as many “active” printers as possible.
Otherwise, we devolve into a system of “Everyone has their own fork” of the firmware and there is no longer a unity of design and functionality in the “Marlin” community. Rather it becomes a fragmented community of “Marlin-variants”.

However, if we are going to have a folder for each of the variants, supported, or not, the “default” configuration (whatever that is) should be treated in the same manner.

Then the Configuration.h file in the main tree needs to be reduced to a single “#include” of the appropriate variant.
However, this can, and IMHO should, then be morphed into an include controlled by a preprocessor definition.
That way, all of the supported variants can actually be built from the same source tree.

BTW, I also suggest that the variant configurations each be stripped of the sections that do not apply. Having #defines that apply to deltabots in a configuration file for a cartesian printer just clutters the presentation.

Richard

Wackerbarth, 2015-05-20:
I have no objection to the idea of “supported/non-supported”.

Actually, the “non-supported” configurations can just be dropped from the source tree.
If someone wants to return one of them to the supported category, it is an easy matter to retrieve the files from the git archive, correct them, and re-add them to the current tree.

As for all of the comments that have been requested in the configuration files, I feel that these are redundant. You are requiring a Readme file in the folder containing the files. That file provides a place for the documentation of all of the files within that configuration folder.

Wackerbarth, 2015-05-20:
Speaking of "non-supported", why are we "supporting" old versions of the Arduino IDE?
Support life would be much easier if everyone used the current IDE (preferably 1.6.2 or later), but at least 1.5.7

CONSULitAS, 2015-05-20:
@Wackerbarth :+1: 

> I feel that it is important to have “out-of-the-box” configurations available for as many “active” printers as possible.
I think the same. Even more: There are printers which have no appropriate support by their producers. So this is the only way for the average user to get an update.

> Then the Configuration.h file in the main tree needs to be reduced to a single “#include” of the appropriate variant.
This would be really nice, but this seems not to be easy with the arduino IDE.

> BTW, I also suggest that the variant configurations each be stripped of the sections that do not apply. 
I would like this, too. But it would be even better to have a really short config file which only sets the diffs for the printer based on the standard configuration. This together with really short comments explaining the needs.

> As for all of the comments that have been requested in the configuration files, I feel that these are redundant. You are requiring a Readme file in the folder containing the files. That file provides a place for the documentation of all of the files within that configuration folder.
OK. But this helps if you work with the configuration files.

E.g. if you add a new option to Configuration.h you will need to do this to every example file. You can get confused rather quick, if you dont see, which file you are editing (they are all named the same).

It happend to me, that I had my K8200 config files in Marlin main folder (for testing) while making a pull request. I didnt even see this in the diffs for my PR.

> Speaking of "non-supported", why are we "supporting" old versions of the Arduino IDE?
Support life would be much easier if everyone used the current IDE (preferably 1.6.2 or later), but at least 1.5.7
:+1: from me. Particularly because it is open source, so no one has a real reason to stick to 1.0 IMHO. But you should address this to the more active coders like @thinkyhead @Wurstnase or @AnHardt

Wackerbarth, 2015-05-20:
> On May 20, 2015, at 12:09 PM, CONSULitAS <notifications@github.com> wrote:
> 
> @Wackerbarth <https://github.com/Wackerbarth> 
> 
> I feel that it is important to have “out-of-the-box” configurations available for as many “active” printers as possible.
> I think the same. Even more: There are printers which have no appropriate support by their producers. So this is the only way for the average user to get an update.
> 
> Then the Configuration.h file in the main tree needs to be reduced to a single “#include” of the appropriate variant.
> This would be really nice, but this seems not to be easy with the arduino IDE.
> 
> BTW, I also suggest that the variant configurations each be stripped of the sections that do not apply. 
> I would like this, too. But it would be even better to have a really short config file which only sets the diffs for the printer based on the standard configuration. This together with really short comments explaining the needs.
> 
> As for all of the comments that have been requested in the configuration files, I feel that these are redundant. You are requiring a Readme file in the folder containing the files. That file provides a place for the documentation of all of the files within that configuration folder.
> OK. But this helps if you work with the configuration files.
> 
> E.g. if you add a new option to Configuration.h you will need to do this to every example file. You can get confused rather quick, if you dont see, which file you are editing (they are all named the same).
> 
Two comments on this:

1) Although the editor that I use does a reasonable job in advising me of the folder/filename of each file, I have no objection to the idea of placing a portion of the file path in a standardized header for each file in the system. However, this would not include “documentation” about the building and testing of the configuration, only the file identification.

2) “add new option to every example file” — I agree that maintaining the current methodology is a PITA. To me, this appears to be a deficiency in the structure of the configuration files. IMHO, configuration should follow an “object-oriented” approach with, when necessary, particular variants overriding particular values of their base definition.  New options would be added to the common base and automatically inherited.

> It happened to me, that I had my K8200 config files in Marlin main folder (for testing) while making a pull request. I didnt even see this in the diffs for my PR.
> 
> Speaking of "non-supported", why are we "supporting" old versions of the Arduino IDE?
> Support life would be much easier if everyone used the current IDE (preferably 1.6.2 or later), but at least 1.5.7
>  from me. Particularly because it is open source, so no one has a real reason to stick to 1.0 IMHO. But you should address this to the more active coders like @thinkyhead <https://github.com/thinkyhead> @Wurstnase <https://github.com/Wurstnase> or @AnHardt <https://github.com/AnHardt>This is particularly true because the IDE and the Arduino “core” are actually two different considerations. Because I have yet to complete the conversion, on some AVR processor devices, I’m actually building firmware that uses the old 1.0 core, but I do ALL of my builds using a 1.6.5 IDE and the compilers packaged with it.
As I manage to upgrade the builds, I will drop the older core files.

thinkyhead, 2015-05-21:
> I read in an other issue that there will be example files removed, but it seems that the community likes them.

I have no objection to keeping lots of `Configuration.h` / `Configuration_adv.h` files, and even updating them with the latest additions. But I do agree that the ultimate responsibility for the actual settings values should be handled by "custodians" (who will sometimes be vendors).

We have discussed a concept of making all configuration values overridable (in advance), but just doing that makes the configs even bigger and uglier. But, we could potentially add an additional `#include "Configuration_override.h"` (for example) to the end of the configurations, before the `SanityCheck.h` and `Conditionals.h` includes, which would allow you to just override the few options that need to be overridden. Then the examples would only need to include the overrides, which might be only 10 or 20 lines long.

It's just an idea at this point, and it has complications.

Wackerbarth, 2015-05-21:
> On May 20, 2015, at 10:24 PM, Scott Lahteine <notifications@github.com> wrote:
> I have no objection to keeping lots of Configuration.h / Configuration_adv.h files, and even updating them with the latest additions. But I do agree that the ultimate responsibility for the actual settings values should be handled by "custodians" (who will sometimes be vendors).
> 
I agree that you need to have “custodians” for the various Configurations. However, you will need to consider configuration changes in the context of there being many configurations and avoid introducing changes to the configurations whenever possible. This is particularly important when it comes to optional features

> We have discussed a concept … which would allow you to just override the few options that need to be overridden. Then the examples would only need to include the overrides, which might be only 10 or 20 lines long.
> 
> It's just an idea at this point
> 

Yes. And ideas mature over time. I think that it has potential.

thinkyhead, 2015-05-21:
> This is particularly important when it comes to optional features

The policy is to always make features disabled by default to avoid just this kind of problem.

Wackerbarth, 2015-05-21:
Actually, I had in mind the configuration parameters for the option itself. (not the “option_present” flag which, as you mention, properly defaults to “not_present”)

> On May 21, 2015, at 12:28 AM, Scott Lahteine <notifications@github.com> wrote:
> This is particularly important when it comes to optional features
> 
> The policy is to always make features disabled by default to avoid just this kind of problem.
>

thinkyhead, 2015-05-21:
> the configuration parameters for the option

Yes, good point. Best to mediate against problems in the configs instead of having to warn users or write documentation that few will read.