MarlinFirmware/Marlin #1437: error ultralcd when FULL_GRAPHIC_SMART_CONTROLLER

MPJ987, 2015-01-29:
Hello,

I have a Prusa i3 hephestos with ramps 1.4 and Arduino mega 2560.

I also have a RepRapDiscount Full Graphic Smart Controller that I want to install on my printer.

every time I: #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
And (//) undefine all others in the // LCD and SD support
Added u8glib to the library.

Now I get the following error when verificer changes:

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
Arduino: 1.0.6 (Windows NT (unknown)), Board: "Arduino Mega 2560 or Mega ADK"
ultralcd.cpp: In function 'void config_lcd_level_bed()':
ultralcd.cpp:758: error: 'lcd' was not declared in this scope
ultralcd.cpp: In function 'void lcd_level_bed_cooling()':
ultralcd.cpp:769: error: 'lcd' was not declared in this scope
ultralcd.cpp:791: error: 'lcd' was not declared in this scope
ultralcd.cpp: In function 'void lcd_level_bed()':
ultralcd.cpp:800: error: 'lcd' was not declared in this scope
ultralcd.cpp: In function 'void lcd_load_material_extrud_1()':
ultralcd.cpp:944: error: 'lcd' was not declared in this scope
ultralcd.cpp: In function 'void lcd_unload_material_extrud_1()':
ultralcd.cpp:990: error: 'lcd' was not declared in this scope
ultralcd.cpp: In function 'void lcd_init()':
ultralcd.cpp:1418: error: 'lcd' was not declared in this scope

I have tried the same with both new and old versions of Marlin and Arduino.
http://reprap.org/wiki/File:Marlin_Hephestos.zip

Hope you can help me.

Best Regards
Mark from Denmark

thinkyhead, 2015-01-30:
Did [this answer](http://forum.arduino.cc/index.php?topic=294345.msg2066333#msg2066333) help?

thinkyhead, 2015-01-30:
I just tested the latest Development branch with Arduino 1.5.2 and `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` worked. Can you confirm that you have the `LiquidCrystal_I2C.h` library installed?

MPJ987, 2015-01-30:
Hephestos come with a RepRapDiscount Smartcontroller out of the box.

I also tried to downloaded and added to the library:
https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library

Downloaded the last version of U8glib for Arduino: 1.17
https://code.google.com/p/u8glib/

[image]

victorandueza, 2015-01-30:
Hello @MPJ987,

As I can see in your report, you are using an old verion of the bq Hephestos firmware. In this moment I am waiting for the pull request in the Marlin repo, where I have added the configuration for this printer.
https://github.com/MarlinFirmware/Marlin/pull/1433

As soon as it is merged, you will be able to use the RepRap Discount Full Graphic Smart Controller, just adding the u8glib to the library and discommenting the correct #define in the Configuration.h file.

Thank you for your report and I am glad to support you in any issues that you can have.

Regards.

MPJ987, 2015-01-30:
Is there a newer version than the one I use: 1.3.1?

victorandueza, 2015-01-30:
For the moment the Marlin FW has a better support of the LCD configuration. I recommended you to install the Marlin firmware as soon my pull request it is merged. Meanwhile, you can see the changes that I have implemented in the marlin code just looking the 'Files Changed' of the pull request where you can find all the Hephestos configuration changes.

MPJ987, 2015-01-30:
Thanks for the help, I am waiting for the new version is available.
How long time do you think that goes?

thinkyhead, 2015-01-31:
Soon. #1431 is the pending update.

closed 2015-02-20 by thinkyhead