MarlinFirmware/Marlin #1442: Compiling error when trying to setup Viki 2 LCD

Natealus, 2015-01-31:
Good evening gents. I have a Rostock Max V2 that I upgraded with a Kraken 4 nozzle and Azteeg X3 Pro. The original LCD that came with it I couldn't find pinouts for so I ordered and just received a new Viki 2 LCD. Nice looking unit but I'm having a problem with implementing it into the Marlin firmware. I added the LiquidCrystal library, SD, U8Glib, and SPI libraries. And all of the code additions for adding Viki 2 support from a fork on Panucatt's website. Link to it is https://github.com/royco/Marlin/commit/4bfc47501a6436fd4aa2aac48a5a7eb70d9c0c75

But when I go to compile and upload it...I get:
  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
Arduino: 1.0.6 (Windows NT (unknown)), Board: "Arduino Mega 2560 or Mega ADK"
ultralcd.cpp:40: error: initializer-string for array of chars is too long
In file included from ultralcd.cpp:43:
/dogm_lcd_implementation.h: In function 'void lcd_implementation_init()':
dogm_lcd_implementation.h:124: error: 'STRING_SPLASH' was not declared in this scope

I know enough about scripting to edit oonfigs and such but I am not yet a Jedi Coder...so any assistance would be greatly appreciated. And if we can merge Viki 2 pins and support into our development that would be awesome. :)

thinkyhead, 2015-01-31:
STRING_SPLASH is defined in Configuration.h in more recent updates to Marlin. Compare your configuration files to the most recent in this repo.

Natealus, 2015-01-31:
I just downloaded and re-configured the whole firmware a few days ago to add in the changes for the 4th extruder. Is it a variable that has to be defined or do you think the line is missing?

Natealus, 2015-01-31:
Ok so update I compared my configuration.h and the one here on GitHub and I do see a difference in that first string version section of the file. Only had 2 of the 5 lines there so I fixed that and went to verify it. Unfortunately we have a:
ultralcd.cpp:40: error: initializer-string for array of chars is too long

I know its probably something silly just like the few missing lines there but I'm not quite there on coding yet to identify what specifically needs to be changed here.

Natealus, 2015-01-31:
It highlighted this section in ultralcd.cpp:
uint8_t lcd_status_message_level;
char lcd_status_message[LCD_WIDTH+1] = WELCOME_MSG;

Just the second line though.

thinkyhead, 2015-02-01:
Try shortening your `MACHINE_NAME`.

Natealus, 2015-02-01:
Sorry I was out last night from the code work. But I redownloaded the dev branch last night to get all the recent updates and installed GitHub Windows so I can try to make a pull request for the Viki 2 additions. Gotta readd all that stuff on the newly changed files.

Natealus, 2015-02-02:
Well there's good news and bad news. Good news is it works...bad news is I've experienced the exact same problem as somebody else that has a delta and graphics lcd working at the same time. I seem to have a stall and the movement motors just stop for a quarter to half second every time the LCD refreshes. I'm kind of wondering the same thing as the other guy though...if it works fine with both on Repetier...what's stopping Marlin from handling it properly? I even tuned down the Delta segments so it would reduce processor load and it didn't seem to have any effect. That makes me think it isn't as much a processing problem as a stall in the code.

Please understand I'm not placing fault on anyone here...I think all of the coders are doing really good work on improving Marlin. I just think we need to investigate it more thoroughly.

galexander1, 2015-02-02:
I haven't looked at the graphics LCD driver, but my guess is that it is,
as you say, a blocking problem.  It's possible the LCD driver code is
slow for no good reason, in which case maybe we can simply increase its
speed.  Otherwise, we could force the LCD refresh routine to only do
1/10th of the screen at once (or something), so that the other tasks get
to have a chance to run even while it's going.

It might be worth having a look at the Repetier code. :)

FWIW, I don't have an LCD.

thinkyhead, 2015-02-03:
@Natealus For now, reducing `DELTA_SEGMENTS_PER_SECOND` from 200 down to 120 is the best known workaround.

Natealus, 2015-02-05:
Yeah I've messed around with it a bit. Going around in circular radiuses are the worst because they're causing many more calculations but if I do reduce it down to 100 segments I can see it being acceptable to work at a trade show. I hesitate to leave it at that for all of my materials and quality tests though and have just been disabling the LCD for the time being. There is still a noticable stutter even though its reduced dramatically from what happens at 200 or 225. But I noticed I was getting a bit of stutter when disabling the LCD and putting back to 200...turns out the thermal runaway protection was causing a small draw on the processor that it couldn't afford so I turned it off cause I'm confident my setup won't have that sort of issue. Seemed to have smoothed back to normal.

Have you seen anything about I2C library improvements? And @thinkyhead I like your idea in another issue about confining the refreshes to just important things like temp, perhaps Z height, and progress. Most of the screen doesn't change so if its refreshing the full frame every refresh I can see that causing some extra time.

thinkyhead, 2015-02-05:
@Natealus I'm trying to think more "electronically" – the fewer wavy signals we need to send down those serial lines the less "noisy" the whole thing will be.

Natealus, 2015-02-08:
Yeah definitely a solid goal there. I've noticed talk in a couple of other issues involving the libraries. Have any of the contributors found info on any new/updated libraries to try?

thinkyhead, 2015-02-09:
I've not heard anything on that front, but I would be interested to see some comparisons between LCD libraries. It's possible they are trying to be too general-purpose, and perhaps we can make a stripped-down version of the library that does less.

Natealus, 2015-02-10:
Yeah because we're asking for just a little bit of info...not that the LCD is a main priority element in the scheme of a 3D printer. So yeah if it could be slimmed up or have it's priority secondary to print moves and calculations...that would be ideal.

galexander1, 2015-02-10:
@Natealus there are a bunch of LCD modules.  If you submit a patch
appropriate to your LCD module, the support for that specific module will
improve.

Natealus, 2015-02-10:
@galexander1 Well I thought we found a delay having to do with delta kinematics calculations and graphics LCDs in general on Arduino processors. As it is I have to turn Delta Segments down to 80 per sec to get smooth"ish" circle printing with LCD enabled. vs 175 I currently run at over USB. I wish the Azteeg X3 Pro came with ARM instead. But it seems all graphics LCDs are having trouble on arduinos with deltas. Even some carriage styles have that issue too. I'm just adding in the defines necessary for the newer Viki 2. It's still using the LiquidCrystal library and SD library and U8glib and SPI. I have a sneaking suspicion that SPI library could use some tweaks cause I've seen many many articles online about problems with SPI and multiple devices trying to access it.

I believe Thinkyhead is looking into ways to slim down the code and make it more efficient. There's also been talk about some of the libraries being outdated or possible fixes for interrupt delays which Arduinos aren't so good at working with if they aren't coded in a particular way.

nophead, 2015-02-10:
You don't have to print with USB if you don't use an LCD. You can start an
SD print over USB was well, or even completely stand alone with no user
interface at all.

Natealus, 2015-02-10:
So its possible to just start a print by just inserting the SD with all the graphics lcd stuff disabled? I'm guessing you only have one file on there at a time with that method right? Although thats far less ideal with trying different slice settings and such but could be useful for trade shows. So what programs do you like @nophead ?

nophead, 2015-02-10:
If the file is called AUTO0.G is will run when the board is reset.

Without the LCD enable you can also start a specified file over USB and
then disconnect it.

I like Octoprint on RPI for my host and Skeinforge to slice.

Natealus, 2015-02-10:
Is that Repetier host? Or another one I'm not familiar with?

thinkyhead, 2015-02-11:
@Natealus I haven't delved into the LCD code as yet. What I'm thinking, ahead of that, is to look at the lower-level functions that the higher-level functions are using and seeing if they are exposed. If so, we can see about using those directly.

thinkyhead, 2015-02-11:
I'm :100:% willing to accept donations of a graphical LCD, by the way!

nophead, 2015-02-11:
No it isn't Repetier, it's a web based host: http://octoprint.org/

Natealus, 2015-02-11:
@thinkyhead Ah I see so you can cause less code to run if you can more directly access certain parts. 

@nophead I'll have to look at it and give it a try cause it looks like this might just be a sad reality of MatterControl. I can't seem to find any place to disable that last M84 command. It runs without fail no matter what I've tried. Its a shame cause its put together pretty well otherwise. Maybe if I export the gcode I can always take that line out.

nophead, 2015-02-11:
You could hack Marlin so M84 doesn't disable XYZ on a Delta.

Natealus, 2015-02-11:
I think that's kind of what Thinkyhead was thinking. An option to disable the firmware's reaction to the M84 code in this kind of situation.

galexander1, 2015-02-11:
If you send M84, it should disable the steppers.  Don't be wrong.

We have nothing to do with your slicer, and we are not generic 3d printer
technical support.

Seriously.

thinkyhead, 2015-02-12:
@nophead I think you meant to say https://github.com/foosel/OctoPrint ;-)

Natealus, 2015-02-12:
Yeah I've heard several people mentioning Octoprint. I'm probably going to give that one a try first. Maybe take a stab at modifying MatterControl since its an open source project. 

But I take it from some people's responses that most people have a slicer and either export to SD card or have a seperate host program like Octoprint or Repetier host? Is MatterControl is somewhat unique being both?

nophead, 2015-02-12:
Both Repetier and OctoPrint to a lesser extent can integrate an wrap Slicers. Some hosts are also slicers or vice versa. The line between them is getting more and more blurred as people try to dumb down 3D printing to appeal to the masses. I am old fashioned and still run Skeinforge from the command line. I believe in separation of concerns and like to keep the geometric aspect of slicing separate from machine control and user interface separate from real time motor control, etc.

@thinkyhead, not sure why you think I didn't mean the page I linked. That is the main front page for OctoPrint. It links to the page you mentioned in the setup instructions.

Natealus, 2015-02-12:
@nophead Yeah Skeinforge has been around for a while from what I understand alot of people like it. And that's kind of what my issue with MatterControl is now is that Geo slicing and machine control and UI are all melded together in MatterControl...from a machine control standpoint for manual operations its pretty nice...but it ghosts an M84 in my ending G-code. So yeah I might want to explore the separated approach if I can't find how M84 keeps sneaking in my epilogue.

thinkyhead, 2015-02-13:
@nophead The joke of my post was that as githubbies we should go straight to the source code and solve the problem there.

thinkyhead, 2015-02-13:
I'll do y'all one better... https://github.com/foosel/OctoPrint/blob/b7badcbf0274f3f0e2dcacfa8548bf7ae4522212/src/octoprint/printer.py#L308

Natealus, 2015-02-13:
+1 @thinkyhead Yeah I think MatterControl's problem is in a similar place. I think its the host directly doing it no matter what cause I looked at the G-Code history in the appdata folder and it doesn't appear to be in there.

nophead, 2015-02-13:
Possibly a bodge but it does seem to me that the firmware is the gatekeeper
for other critical limits like motor speed, soft endstops, etc, so that
hosts and slicers can't break the machine. So perhaps it makes sense on
machines where the motors should never be disabled that can be enforced in
the firmware as well.

galexander1, 2015-02-13:
@nophead We don't enforce endstops if the host sends malicious G-code.
If you send a G92, you can pretty easily convice the robot to ignore any
endstops.  We don't consider that a bug.

If the host sends an M84, we turn off the stepper motors.  That's what
M84 is for.  If you don't want that, don't send an M84.

On @Natealaus' robot, if he wants to manually move the head around with
his fingers for some reason, he can manually send M18 (or M84), push it
around, and then send M17.  M18/M84 don't break his robot, and provide
exactly the functionality it was designed for.

nophead, 2015-02-13:
Well slicers commonly tell the Z axis to go at crazy speeds and rely on the firmware capping it. By your logic it should do what the slicer says and cause the Z axis to skip steps. Then we should force all slicer writers to have a Z speed in their config. 

Similarly if the slicer says heat to 700C we should do that and melt the hot end and burn down the users house.

galexander1, 2015-02-13:
Movement and heating are both complicated functions that accept a wide range of inputs.  We do provide rudimentary range checking, and I think it is a good idea.

M84 is not complicated.  M84 has one purpose.  If you don't want that purpose, don't send that code.

Natealus, 2015-02-13:
Yeah there is already a thermal runaway protection that can be enabled. And min and max temps incase the temp sensor dies. I do think its the host now cause I investigated the generated g-code files and the M84 is absent from there. Something that I gotta look through MatterControl's code cause I think it's hard coded.

But anyways Thinky and I figured out the compiling issues in this post and pushed some commits to fix up some things with 4 extruders and Viki 2 implementation. The compilation issue is fixed.

closed 2015-02-13 by Natealus

clefranc, 2015-02-14:
@thinkyhead 
"I'm :100:% willing to accept donations of a graphical LCD, by the way!"

If you promise to fix the stuttering with delta, I'll send you one...

Natealus, 2015-02-14:
*double thumbs up* :+1:  :+1:  :) Yeah I just turned my Viki on and I had to drop segments from 165 to 80 for a nice smooth motion even when manually moving. But there's some ups to calibrating with the lcd :)

thinkyhead, 2015-02-16:
@clefranc Ah, then I will be forced to finish building the delta on my workbench! I may even be able to get hold of a delta to work on the issue. (Being in Seattle, I have Metrix to draw from.) I def. want to fix the issue by some magick. But truly, I should be able to do much simply by examining and understanding the code. I've just got 4 other code-related things going that are keeping me from that task atm.

Natealus, 2015-02-17:
Yup building a printer is a time consuming task. Takes a good plot of hours. So does coding lol. It feels good to be productive. :)