MarlinFirmware/Marlin #1443: [question] Minimal speed stepper.ccp

Wurstnase, 2015-01-31:
I've talked some days ago with alex and thought the ISR-Part is not solvable for me. But I want to give it a try. I need more speed on my Due/Radds and start to modify bobc/Marlin. 

Just start to stock in this line. How is the Mega limited for minimal speed? Why /500000? 

 if(step_rate < (F_CPU/500000)) step_rate = (F_CPU/500000);
  step_rate -= (F_CPU/500000); // Correct for minimal speed

https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/stepper.cpp#L259

Greetings 
Nico

alexborro, 2015-01-31:
Nico, I cannot tell you about the min speed, but I can shed some light on
max speed.

The stepper ISR calculates the interval between interruption in order to
pulse the stepper in the proper time. Higher the speed, shorter the time.

This is done in the last part of ISR:
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/stepper.cpp#L690
It calls that calc_timer that uses a speed_lookuptable to find a value to
fill OCR1A (Output Compare Register 1A)..
Each unit means 0.5us, I mean, if you wanna pulse the stepper at 1Khz, just
set OCR1A with 2000.

The max frequency for this interruption is 10Khz (OCR1A = 200), because if
we go lower than that there will be no available time between interruptions
to process other stuffs.

Marlin can pulse stepper up to 40Khz because we can pulse it up to 4 times
in the same interruption for really fast moves (look for step_loops
variable).

I really don't understand the meaning of that speed_lookuptable.. But there
is a python script to build it.. I think if you plan to use faster CPU, you
need to create a proper table for the new clock.

If someone over there understand the reason of the speed_lookuptable,
please, let me know.

Cheers.

Alex.

Wurstnase, 2015-01-31:
Thanks for explanation. Now I know where the 2000 comes from.

The timer-calculation at bobc/Marlin is a bit different and doesn't use the lookup-table. There it is just the ISR-speed (At the moment 84MHz/32 clock3) divided by the step_rate.

Wurstnase, 2015-02-01:
Ok found something about the lookuptable. This is close the same as 2,000,000/step_rate, just slightly faster. (16MHz/8Prescaler = 2M...)

I just startet a new old Version of Marlin from bobc with alot of modifications. If someone has a RADDS or RAMPS-FD for testing and helping with programming, you are welcome!

Wurstnase, 2015-02-05:
And finally the solution for the min-speed.

The Mega has a 16bit-counter.
16MHz / 500k = 32 as min_speed
16M / 8 Prescaler = 2MHz clock
2MHz / 32 = 62500 which is < than 16bit (65535)

so we are save that the counter never overflow.

alexborro, 2015-02-05:
Now I ask you a question.. what is the "500k" ???

Wurstnase, 2015-02-05:
500k < 8*2^16. So this works also with 20MHz.

20M / 500k = 40
20M / 8 = 2.5MHz
2.5 / 40 < 2^16

or

2^16 > 2MHz/x
x < 2MHz/(2^16)
x < 16MHz/ (8 * 2^16)

closed 2015-02-20 by thinkyhead