MarlinFirmware/Marlin #1444: Stick a folder in the top of LCD File List

alexborro, 2015-01-31:
@thinkyhead , I'm thinking in a feature and I guess you are the right one to help me  ;-)

I usually have a folder in my SD Card named "Setup" that contains files to help machine maintenance, like "Load Filament", "Eject Filament" and so one..

I like to reverse the order listing of files to have the last added in the top, saving me some time. The problem is my Setup folder is in the bottom of the list.

Is there a easy way to keep folder/files starting with "_" in the top of the list, no matter when it was added?
So one can rename the folder having useful files and have it always in the top, like my "_Setup" ??
There is no requirement for sorting more than one file starting with "_"; they just need to be in the top.

Cheers.

Alex.

thinkyhead, 2015-01-31:
You should use my SD file sorting patch. I will make a branch in my repo based on Development which includes the feature. Then you can use it to sort exactly the way you wish! I'll let you know when the patch is ready.

alexborro, 2015-01-31:
Great.. but this patch is that polemic one, right?? Is there a way to do what I want in a easy way we can merge in official branch??

Cheers.

Alex.

thinkyhead, 2015-02-01:
Well, actually it was designed to be merged into the main branch – that was my naive hope. But one collaborator had strong (and reasonable) reservations, because it (necessarily) uses malloc, but in a fairly well-bounded way, and it frees all allocations before printing begins.

The dynamic allocation part is more-or-less okay in itself because it frees completely. But if you try to read too many files within a single folder (and for the name caching option, if you have too many long filenames in a single folder), then you could potentially run out of SRAM.

There's only 3K of extra space on a Mega2560 (the most common card), but newer cards – DUE, Teensy, etc. – have a lot more SRAM. So, while this feature is more supportable on bigger cards, before it will be acceptable to the main branch it has to be tested to demonstrate its predictability, and it must be able to fail elegantly if it can't fit the file index (or, optionally, filenames) in SRAM.

Anyway, if no one minds our adding an "Experimental" branch, then I am happy to do so, and get some help to make this feature more robust. Let me roll my patch against Development now, then you can at least grab a copy and see how it works, and you'll see where it needs work.

thinkyhead, 2015-02-01:
Here's my branch, rebased on the latest Development.... https://github.com/thinkyhead/Marlin/compare/sd_sorting

AnHardt, 2015-02-01:
There isn't anything to compare. :-(

thinkyhead, 2015-02-01:
Sorry, somehow it got unpublished – glitches in my Github desktop app. I have restored it at the same URL.

thinkyhead, 2015-02-24:
@alexborro Thinking about a less-polemic, but similar solution, which is to pre-scan new folders on entry and gather info about which items are folders. You could use a fixed-size buffer and set a single bit per file-entry, up to some limit, as in `bool bitFlag(n) { return 0 != (bitArray[n>>3] & (1<< (n&7))); }`

alexborro, 2015-02-24:
That's is nice.. The idea is to first show folders starting with "_", like
a "_Config" folder. With one byte we can have 8 starting folders and I
think it is more than enough.
To be honest, I'm not used to program file reading routines; do you mind
implementing it?? I can help testing.

Thanks.

Alex.

thinkyhead, 2015-02-24:
@alexborro I'll muck around with over the next 24-48 hours.

thinkyhead, 2015-02-27:
See if this works. Actually all it does is to sort folders to the top (or bottom) with no sorting of names. https://github.com/MarlinFirmware/Marlin/compare/Development...thinkyhead:sd_folder_sorting

alexborro, 2015-02-27:
> Development...thinkyhead:sd_folder_sorting

It is almost working  ;-)

It is placing the folders on top, but hiding folders starting with "_"..
Such folders are not showed at all.

thinkyhead, 2015-02-28:
@alexborro The hiding of files/folders beginning with _ is probably a remnant old code. I'll see about stripping out that filter.

thinkyhead, 2015-02-28:
Just alter `cardreader.cpp` lines 97 and 99 and you should be good.

ZetaPhoenix, 2015-02-28:
Hiding files starting with _ is useful for Mac users as Mac creates files like _DS_Store for indexing iirc. Maybe you can make it show folders but not files. 

-Jon

CONSULitAS, 2015-02-28:
@ZetaPhoenix Sorry, but you are wrong. The files are named .DS_Store (beginning with a dot, see http://en.wikipedia.org/wiki/.DS_Store). But there are even more like .Trashes .Spotlight-V100 .fseventsd and so on.

But on Mac OS X all this files start with a dot - like on every unix flavor for hidden files.

@alexborro So please change the behavior to hiding files starting with dots instead of underscore. This would be really helpful!

thinkyhead, 2015-02-28:
@ZetaPhoenix No. The files Mac OS X creates are .DS_Store files. They begin with a dot. There are no (standard) hidden files on OS X that rely on file attributes. They all begin with a dot. It's a well-behaved UNIX.

ZetaPhoenix, 2015-02-28:
Like I said, IIRC. Apparently I remembered wrong. 😱

thinkyhead, 2015-02-28:
Yeah, I always wondered why filenames with underscores have been filtered up to now. If it's due to some common misconception about OS X then it's good to clear that up. I have a patch about ready, cleaning up `cardreader.cpp`, so I can make the change there as well.

boelle, 2015-05-31:
will close this one as there are no activity.... and @thinkyhead provided a fix

closed 2015-05-31 by boelle