MarlinFirmware/Marlin #1447: Editable menu-items with longer descriptions do not fit on screen while editing - on graphical displays.

AnHardt, 2015-02-01:
While editing the values on a DOGM-display a larger font is used, Due to that, for longer descriptions, text and value overlap.

Possible fixes in lcd_implementation_drawedit() in dogm_lcd_implementation.h:
a) Use the normal font. (will save 3012 bytes)
b) Split screen in 2 lines. ( will still allow only 13 characters, may use some additional bytes ) (but i don't need my glasses)

Any opinions?

thinkyhead, 2015-02-01:
I would like to see the difference, if you have a chance to post pictures. Otherwise, the Normal font _sounds_ like a better option!

AnHardt, 2015-02-01:
Now:
[image]
[image]

AnHardt, 2015-02-01:
Fix type a):
[image]

AnHardt, 2015-02-01:
Fix type b):
[image]
[image]

thinkyhead, 2015-02-02:
Is it possible to detect if the line will (potentially) overlap, and then use the smaller font only in that case?

AnHardt, 2015-02-02:
When i thought about centering i discovered that a simple strlen() does not work. Guess that the special characters are represented by two bytes. Have to investigate this.

thinkyhead, 2015-02-02:
@AnHardt 

    int my_strlen_utf8_c(char *s) {
       int i = 0, j = 0;
       while (s[i]) {
         if ((s[i] & 0xc0) != 0x80) j++;
         i++;
       }
       return j;
    }

AnHardt, 2015-02-03:
@thinkyhead 
Discovered the problem - me. Should have used strlen_P(). Results have been a bit random :-).
Autoadaption is on its way.

AnHardt, 2015-02-03:
Impressions from the test-branch.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
17 is the limit for a 3 digit value
[image]
As before. No chance for the thermometer. But picture at least proves - working with different length of values.
[image]

AnHardt, 2015-02-03:
Thermometer workaround wit some spaces.
[image]

thinkyhead, 2015-02-03:
Wow, yeah, that looks about perfect!

PR #1462 (2015-02-03): Fixes issue #1447 - Overlapping of long descriptions with values.
Fixes #1447 
Edit screen locks exactly like the last preview.
closed 2015-02-03 by thinkyhead